toppers/hakoniwa-px4sim #41: 一箇所間違いっぽい。

kenjihiranabe, 2023-12-08:
https://github.com/toppers/hakoniwa-px4sim/blob/c1d1faf51ad90d2d1f7a9ddff0fce92759e47217/hakoniwa/src/assets/drone/physics/ground_frame/drone_dynamics_ground_frame.cpp#L30C149-L30C150

s_psy は c_psi の間違いっぽい。。。確認ください。

tmori, 2023-12-08:
> https://github.com/toppers/hakoniwa-px4sim/blob/c1d1faf51ad90d2d1f7a9ddff0fce92759e47217/hakoniwa/src/assets/drone/physics/ground_frame/drone_dynamics_ground_frame.cpp#L30C149-L30C150
> 
> s_psy は c_psi の間違いっぽい。。。確認ください。

ご確認いただき、ありがとうございます！確認しました。

バグ：
> - (thrust.data /  this->param_mass ) * ( this->cache.cos_phi * this->cache.sin_theta * this->cache.sin_psi - this->cache.sin_phi * this->cache.sin_psi )

正しい：
>  - (thrust.data /  this->param_mass ) * ( this->cache.cos_phi * this->cache.sin_theta * this->cache.sin_psi - this->cache.sin_phi * this->cache.cos_psi )

これが一致しない原因かもですね。確認してみます！！

tmori, 2023-12-08:
以下で修正しました。

https://github.com/toppers/hakoniwa-px4sim/pull/42

commit 1f10bd32 (2023-12-08): fix #41
commit 01ad1ce2 (2023-12-08): Merge pull request #42 from toppers/fix-no41

fix #41
closed 2023-12-08 by tmori