toppers/hakoniwa-px4sim #82: 右前方へ移動して目標値で停止するテストを実施する

tmori, 2023-12-18:
# テスト環境

* Windows 環境
* 機体の移動にはQGCを利用

# テスト目的

右前方へ移動して目標値で停止する機体パラメータを探索する

現時点で安定してホバリング＆前進して停止するパラメータは以下：

```
{
    "simulation": {
      "lockstep": true,
      "timeStep": 0.003,
      "logOutputDirectory": "./",
      "logOutput": {
        "sensors": {
            "acc": true,
            "gyro": true,
            "mag": true,
            "baro": true,
            "gps": true    
        },
        "mavlink": {
            "hil_sensor": true,
            "hil_gps": true,
            "hil_actuator_controls": true
        }
      },
      "mavlink_tx_period_msec": {
        "hil_sensor": 3,
        "hil_gps": 3
      },
      "location": {
        "latitude": 47.641468,
        "longitude": -122.140165,
        "altitude": 121.321,
        "magneticField": {
          "intensity_nT": 53045.1,
          "declination_deg": 15.306,
          "inclination_deg": 68.984
        }  
      }
    },
    "components": {
        "droneDynamics": {
            "physicsEquation": "BodyFrame",
            "airFrictionCoefficient": 0.2,
            "inertia": [ 0.05, 0.05, 0.05 ]
          },
          "rotor": {
            "tau": 0.001,
            "k": 1000.0,
            "rpmMax": 2200
          },
          "thruster": {
            "rotorPositions": [
              {"position": [  0.3,  0.3, 0 ], "rotationDirection":  1.0e-3 },
              {"position": [ -0.3, -0.3, 0 ], "rotationDirection":  1.0e-3 },
              {"position": [  0.3, -0.3, 0 ], "rotationDirection": -1.0e-3 },
              {"position": [ -0.3,  0.3, 0 ], "rotationDirection": -1.0e-3 }
              ],
            "parameterA": 2.18e-06,
            "parameterB": 6.94444e-09,
            "parameterJr": 1.0e-03
          },
          "sensors": {
            "acc": {
              "sampleCount": 4,
              "noise": 0
            },
            "gyro": {
              "sampleCount": 4,
              "noise": 0
            },
            "mag": {
              "sampleCount": 4,
              "noise": 0.0001
            },
            "baro": {
              "sampleCount": 4,
              "noise": 0
            },
            "gps": {
              "sampleCount": 4,
              "noise": 0
            }
          }
    }
  }
  
```

# テスト内容

機体をホバリングした状態になったら、QGCで機体の右前方を目標値として設定して移動させる

期待値：目標値で機体が停止する

テストエビデンス：
機体のロギングデータ（安定度を見たいため）

tmori, 2023-12-23:
# テスト結果

以下のコンフィグで安定して移動することを確認した。

```
{
    "simulation": {
      "lockstep": true,
      "timeStep": 0.003,
      "logOutputDirectory": "./",
      "logOutput": {
        "sensors": {
            "acc": true,
            "gyro": true,
            "mag": true,
            "baro": true,
            "gps": true    
        },
        "mavlink": {
            "hil_sensor": true,
            "hil_gps": true,
            "hil_actuator_controls": true
        }
      },
      "mavlink_tx_period_msec": {
        "hil_sensor": 3,
        "hil_gps": 3
      },
      "location": {
        "latitude": 47.641468,
        "longitude": -122.140165,
        "altitude": 121.321,
        "magneticField": {
          "intensity_nT": 53045.1,
          "declination_deg": 15.306,
          "inclination_deg": 68.984
        }  
      }
    },
    "components": {
        "droneDynamics": {
            "physicsEquation": "BodyFrame",
            "airFrictionCoefficient": 0.1,
            "inertia": [ 0.01, 0.01, 0.02 ],
            "mass_kg": 1,
            "position_meter": [0, 0, 0 ],
            "angle_degree": [0, 0, 0]
          },
          "rotor": {
            "Tr": 0.1,
            "Kr": 8000,
            "rpmMax": 8000
          },
          "thruster": {
            "rotorPositions": [
              {"position": [  0.3,  0.3, 0 ], "rotationDirection":  1.0 },
              {"position": [ -0.3, -0.3, 0 ], "rotationDirection":  1.0 },
              {"position": [  0.3, -0.3, 0 ], "rotationDirection": -1.0 },
              {"position": [ -0.3,  0.3, 0 ], "rotationDirection": -1.0 }
              ],
            "HoveringRpm": 4000,
            "parameterB":  5.0e-09,
            "parameterJr": 1.0e-08
          },
          "sensors": {
            "acc": {
              "sampleCount": 1,
              "noise": 0
            },
            "gyro": {
              "sampleCount": 1,
              "noise": 0
            },
            "mag": {
              "sampleCount": 1,
              "noise": 0
            },
            "baro": {
              "sampleCount": 1,
              "noise": 0
            },
            "gps": {
              "sampleCount": 1,
              "noise": 0
            }
          }
    }
  }
  
```

ログ：

[log_rf.zip](https://github.com/toppers/hakoniwa-px4sim/files/13757461/log_rf.zip)

グラフ：

[image]

[image]

closed 2023-12-23 by tmori

kenjihiranabe, 2023-12-23:
いい感じやねー

commit 26114dbb (2023-12-23): fix #82