toppers/hakoniwa-px4sim #83: 右方向へ移動して目標値で停止するテストを実施する

tmori, 2023-12-18:
# テスト環境

* Windows 環境
* 機体の移動にはQGCを利用

# テスト目的

右へ移動して目標値で停止する機体パラメータを探索する

現時点で安定してホバリング＆前進して停止するパラメータは以下：

```
{
    "simulation": {
      "lockstep": true,
      "timeStep": 0.003,
      "logOutputDirectory": "./",
      "logOutput": {
        "sensors": {
            "acc": true,
            "gyro": true,
            "mag": true,
            "baro": true,
            "gps": true    
        },
        "mavlink": {
            "hil_sensor": true,
            "hil_gps": true,
            "hil_actuator_controls": true
        }
      },
      "mavlink_tx_period_msec": {
        "hil_sensor": 3,
        "hil_gps": 3
      },
      "location": {
        "latitude": 47.641468,
        "longitude": -122.140165,
        "altitude": 121.321,
        "magneticField": {
          "intensity_nT": 53045.1,
          "declination_deg": 15.306,
          "inclination_deg": 68.984
        }  
      }
    },
    "components": {
        "droneDynamics": {
            "physicsEquation": "BodyFrame",
            "airFrictionCoefficient": 0.2,
            "inertia": [ 0.05, 0.05, 0.05 ]
          },
          "rotor": {
            "tau": 0.001,
            "k": 1000.0,
            "rpmMax": 2200
          },
          "thruster": {
            "rotorPositions": [
              {"position": [  0.3,  0.3, 0 ], "rotationDirection":  1.0e-3 },
              {"position": [ -0.3, -0.3, 0 ], "rotationDirection":  1.0e-3 },
              {"position": [  0.3, -0.3, 0 ], "rotationDirection": -1.0e-3 },
              {"position": [ -0.3,  0.3, 0 ], "rotationDirection": -1.0e-3 }
              ],
            "parameterA": 2.18e-06,
            "parameterB": 6.94444e-09,
            "parameterJr": 1.0e-03
          },
          "sensors": {
            "acc": {
              "sampleCount": 4,
              "noise": 0
            },
            "gyro": {
              "sampleCount": 4,
              "noise": 0
            },
            "mag": {
              "sampleCount": 4,
              "noise": 0.0001
            },
            "baro": {
              "sampleCount": 4,
              "noise": 0
            },
            "gps": {
              "sampleCount": 4,
              "noise": 0
            }
          }
    }
  }
  
```

# テスト内容

機体をホバリングした状態になったら、QGCで機体の右方向を目標値として設定して移動させる

期待値：目標値で機体が停止する

テストエビデンス：
機体のロギングデータ（安定度を見たいため）

tmori, 2023-12-19:
# テスト結果（NG）

右へ水平移動することを確認したが、目標値へ到達して、しばらくすると、PX4が以下の警告を出力し、
機体が不安定になった。

```
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
WARN  [health_and_arming_checks] Compass needs calibration - Land now!	
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
```

kenjihiranabe, 2023-12-19:
いくつか僕の解釈を書いておくので、認識があっているか確認お願い。
[image]
仮説：駆動側からみて、すごく「軽い」ものを制御しているのではないか？慣性側（m, I) の値と駆動側(Thrust, Torque)のバランスが間違っていないか？
質問：ASIMのログでもこのくらい細かい振動していますか？

tmori, 2023-12-19:
> いくつか僕の解釈を書いておくので、認識があっているか確認お願い。 

すごく細かく制御されているのは同じ人市k知恵す。
あと、初動の動きも、その通りだと思います・

> 質問：ASIMのログでもこのくらい細かい振動していますか？

AirSimの方は、内部データが取れないので、ちょっとわからないですが、
QGCで姿勢角の動きを見ていた感じからすると、すごく安定していたと思います。

> 仮説：駆動側からみて、すごく「軽い」ものを制御しているのではないか？慣性側（m, I) の値と駆動側(Thrust, Torque)のバランスが間違っていないか？ 

なるほど。

> "inertia": [ 0.05, 0.05, 0.05 ]

イナーシャが小さすぎるかもしれないですね。

tmori, 2023-12-19:
AirSimの右方向移動のログ：

[log.zip](https://github.com/toppers/hakoniwa-px4sim/files/13712866/log.zip)

tmori, 2023-12-19:
AirSim:
[image]

QGC:
[image]

tmori, 2023-12-21:
# テスト内容と結果２（NG）

## テスト内容

機体をホバリングした状態になったら、QGCで機体の右方向を目標値として設定して移動させる

期待値：目標値で機体が停止する

## コンフィグパラメータ

```
{
    "simulation": {
      "lockstep": true,
      "timeStep": 0.003,
      "logOutputDirectory": "./",
      "logOutput": {
        "sensors": {
            "acc": true,
            "gyro": true,
            "mag": true,
            "baro": true,
            "gps": true    
        },
        "mavlink": {
            "hil_sensor": true,
            "hil_gps": true,
            "hil_actuator_controls": true
        }
      },
      "mavlink_tx_period_msec": {
        "hil_sensor": 3,
        "hil_gps": 3
      },
      "location": {
        "latitude": 47.641468,
        "longitude": -122.140165,
        "altitude": 121.321,
        "magneticField": {
          "intensity_nT": 53045.1,
          "declination_deg": 15.306,
          "inclination_deg": 68.984
        }  
      }
    },
    "components": {
        "droneDynamics": {
            "physicsEquation": "BodyFrame",
            "airFrictionCoefficient": 0.324,
            "inertia": [ 0.03, 0.03, 0.06 ]
          },
          "rotor": {
            "Tr": 0.1,
            "Kr": 8000,
            "rpmMax": 8000
          },
          "thruster": {
            "rotorPositions": [
              {"position": [  0.3,  0.3, 0 ], "rotationDirection":  1.0 },
              {"position": [ -0.3, -0.3, 0 ], "rotationDirection":  1.0 },
              {"position": [  0.3, -0.3, 0 ], "rotationDirection": -1.0 },
              {"position": [ -0.3,  0.3, 0 ], "rotationDirection": -1.0 }
              ],
            "HoveringRpm": 4000,
            "parameterB":  0.5e-07,
            "parameterJr": 1.0e-08
          },
          "sensors": {
            "acc": {
              "sampleCount": 1,
              "noise": 0
            },
            "gyro": {
              "sampleCount": 1,
              "noise": 0
            },
            "mag": {
              "sampleCount": 1,
              "noise": 0.0001
            },
            "baro": {
              "sampleCount": 1,
              "noise": 0
            },
            "gps": {
              "sampleCount": 1,
              "noise": 0
            }
          }
    }
  }

```

## テスト結果

機体は右方向へ移動したが、しばらくすると落ちてしまう。

また、PX4からは相変わらず、以下の警告メッセージが出力される

```
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
```

kenjihiranabe, 2023-12-21:
分析しました。Bを1/3 程度にすることで、初動の揺れと最後の揺れも解決できるかも。
[image]

tmori, 2023-12-21:
落ちる原因はよくわからないのですが、以下のパラメータを調整したところ、落ちなくなりました。

修正前：
　"airFrictionCoefficient": 0.324
　"inertia": [ 0.03, 0.03, 0.06 ]
    "parameterB":  0.5e-07,
修正後：
　"airFrictionCoefficient": 0.1
　"inertia": [ 0.01, 0.01, 0.02 ]
    "parameterB":  2.0e-08,

修正後の結果：

警告はまだ出ます。

```
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
```

ログ一式：

[test_log3.zip](https://github.com/toppers/hakoniwa-px4sim/files/13736244/test_log3.zip)

グラフ（代表的なもの）

[image]

加速度：

[image]

[image]

ジャイロ：

[image]

Thrust：

[image]

トルク：

[image]

位置：

[image]

kenjihiranabe, 2023-12-21:
32秒から急に右に動き出し
3秒後、35秒で一旦目標値を通り越して
5秒後、40秒でバックして戻っている
40秒から60秒の、20秒間は目標値を行きすぎたり、戻ったりしている。
んですね。つまり、3秒で到達して、20秒、位置合わせしている。

tmori, 2023-12-23:
#89 の修正を適用した結果

# controls

[image]

# pos

[image]

# rot

[image]

# acc(x, y)

[image]

# acc(z)

[image]

# gyro

[image]

# thrust

[image]

# torque

[image]

tmori, 2023-12-23:
安定して右方向いどうするようになったのでクローズします。

closed 2023-12-23 by tmori