toppers/hakoniwa-px4sim #88: 機体の初期姿勢が前方を向いていないと、そもそもホバリングしない

tmori, 2023-12-22:
# ヨー角：90度の場合

現象：プロペラは回転するが、浮上しない。そして、どっかへ飛んでいってしまう。

[image]

ログ：

[log_90deg.zip](https://github.com/toppers/hakoniwa-px4sim/files/13756760/log_90deg.zip)

# ヨー角：180度の場合

現象：シミュレーション開始と同時に、浮上し始める。。離陸指示もしていないのに。

[image]

ログ：

[log_180deg.zip](https://github.com/toppers/hakoniwa-px4sim/files/13756769/log_180deg.zip)

# ヨー角：270度の場合

現象：離陸指示を出すとQGCが「離陸することができません。機体位置が不明です。」とエラーメッセージを出す。
さらに、PX4からは以下の警告が出力される。

```
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
WARN  [health_and_arming_checks] Preflight Fail: vertical velocity unstable
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
```

ログ：

[log_270deg.zip](https://github.com/toppers/hakoniwa-px4sim/files/13756780/log_270deg.zip)

kenjihiranabe, 2023-12-22:
やったじゃん！これが原因だね。
ここまでわかれば後は行けるでしょう。

tmori, 2023-12-22:
ヨー角：180度の場合の原因は、これでした。

https://github.com/toppers/hakoniwa-px4sim/issues/89

暫定修正して、ホバリングすることを確認済みです。

kenjihiranabe, 2023-12-23:
教訓：今回なぜ単体テストをすり抜けたか。
全方向に body を回してテストいるが、最初のベクトルが、v = {1, 0, 0 } だった。
この y=z=0 ベクトルはいくらどの方向に回しても、このバグはキャンセルされてでない。
```
   z_e =   (- s_theta)                                 * x
           + (s_psi * c_theta)                         * y
           + (c_psi * c_theta)                         * z;
```
なので。テストを追加しました。

closed 2023-12-23 by kenjihiranabe