toppers/hakoniwa-px4sim #90: QGCで見ると、機体の位置が突然違うところにいく現象が発生する

tmori, 2023-12-22:


tmori, 2023-12-22:
以下の修正をしたら改善したようなきがする。

位置が突然飛ぶ現象は相変わらず発生する。

[image]

tmori, 2023-12-22:
位置が飛ぶ現象は、おそらく、PX4内のカルマンフィルタの推測値が突然変わったのではないかと推測している。

tmori, 2023-12-23:
南北の移動の場合は、位置飛びが発生しない！

[image]

tmori, 2023-12-23:
こんなに遠くまで行って、帰還できるようになりました。

[image]

tmori, 2023-12-23:
AirSimの右方向移動時のデータ

磁気センサ：

[image]

GPSセンサ：

[image]

[image]

tmori, 2023-12-23:
箱庭の方

磁気センサ：

[image]

GPSセンサ：

[image]

[image]

tmori, 2023-12-23:
# 真北移動時のグラフ

[image]

[image]

[image]

# 考察

移動速度が５m/secを超えると、PX4が減速をかけ、少しずつ目標値へ近づいている。

tmori, 2023-12-23:
# 真東移動時のグラフ

[image]

[image]

[image]

# 考察

移動速が５m/secを超えると、PX4が減速をかけるが、vnとの連動がなんか変で、動きがおかしい。

tmori, 2023-12-23:
jmavsimの磁気センサの実装：

https://github.com/PX4/jMAVSim/blob/66b764ada522893c05224950aa6268c809f8e48a/src/me/drton/jmavsim/SimpleSensors.java#L135

tmori, 2023-12-23:
真東に移動する時の軌跡を、drone_dynamics.csvでグラフ化したら、実際の動きとして変な動きをしているようだ。

[image]

tmori, 2023-12-23:
[image]

tmori, 2023-12-23:
真東に移動しているのだけど、速度が上限（5m/s）を超えたので、ピッチを上げてブレーキをかけようとしたときに、ロール側に力が働いてしまい、変な軌跡になっていると思われる。

kenjihiranabe, 2023-12-23:
なぜロールする、、、北だと起こらないのだよね。
mag かなあ。確かに東向きに進むときはmagをずっと左に見てるから、北とは違うよね。

tmori, 2023-12-23:
動画とれた！

https://github.com/toppers/hakoniwa-px4sim/assets/164193/a01621a2-d656-4dd4-a14b-01d1e1597eeb

tmori, 2023-12-24:
リテイクしました。

https://github.com/toppers/hakoniwa-px4sim/assets/164193/c61bbbd1-91a6-4fe4-9736-4c96eb4c8e0c

tmori, 2023-12-24:
https://github.com/toppers/hakoniwa-px4sim/assets/164193/54220569-cdf4-4e17-afb0-95e6194f7b1a

リテイクしたところ、明らかにPX4側がドローンの位置を誤解していますね。

tmori, 2023-12-24:
この問題は時間かかりそうですね。長期戦で望む方が良さそう。
カルマンフィルタの勉強しよう。

tmori, 2023-12-28:
本件は、機体の水平速度の上限設定することで発生しなくなったのでクローズします。

https://github.com/toppers/hakoniwa-px4sim/issues/94#issuecomment-1869296134

tmori, 2025-03-05:
復活させます

tmori, 2025-03-06:
原因わかったかも。

GPSセンサの経度、緯度の補正を考慮にして修正したら発生しなくなった。

```
double lon = this->ref_lon + (p.data.y / (111000.0 * cos(this->ref_lat * M_PI / 180.0)));
```