toppers/hakoniwa-px4sim #109: MATLAB の実装に渡す情報よために、Cインターフェースを作る。

kenjihiranabe, 2023-12-28:
インターフェース設計のメモ。

std::tuple は、struct リターンするかな？ポインタで値返すと引数多くなっていやだなあ。

他は、extern "C" で、関数シンボルを名前変えないでリンカーンに渡すようにした明示的にインターフェース関数つくり、.h にC用の#ifndef __colusplus で入れて、両言語でコンパイル可能にするか。

C言語用は、名前空間大域に出してもいいかな。

If your C++ library only has functions and std::tuple return values, and doesn't have classes, you can make a header file to C lang by following these steps: https://www.geeksforgeeks.org/header-files-in-c-c-with-examples/ https://learn.microsoft.com/en-us/cpp/cpp/header-files-cpp?view=msvc-170

•  Declare the C++ functions with extern "C" to avoid name mangling. This tells the compiler to use the C calling convention and linkage for the functions. For example:

// foo.hpp
extern "C" {
int foo(int x, int y); // a C++ function that returns an int
void bar(int* x, int* y); // a C++ function that returns a std::tuple<int, int>
}

•  Define the C++ functions in a source file with the same name as the header file. Use the std::tie function to unpack the std::tuple return values into pointers. For example:

// foo.cpp
#include "foo.hpp"
#include <tuple>

int foo(int x, int y) {
// some C++ code that returns an int
return x + y;
}

void bar(int* x, int* y) {
// some C++ code that returns a std::tuple<int, int>
std::tuple<int, int> result = std::make_tuple(x * y, x - y);
// unpack the tuple into pointers
std::tie(*x, *y) = result;
}

•  Include the header file in your C code and call the C++ functions as usual. For example:

// main.c
#include "foo.h"
#include <stdio.h>

int main() {
// call the C++ function that returns an int
int result = foo(3, 4);
printf("result = %d\n", result);

// call the C++ function that returns a std::tuple<int, int>
int x = 5;
int y = 6;
bar(&x, &y);
printf("x = %d, y = %d\n", x, y);

return 0;
}

This way, you can make a header file to C lang for your C++ library with functions and std::tuple return values. However, you need to be careful about some issues, such as exception handling, memory management, and data types compatibility. https://www.geeksforgeeks.org/header-files-in-c-c-with-examples/ https://learn.microsoft.com/en-us/cpp/cpp/header-files-cpp?view=msvc-170

If you want to learn more about how to make a header file to C lang for your C++ library, you can check out this site https://www.geeksforgeeks.org/header-files-in-c-c-with-examples/ or this site https://learn.microsoft.com/en-us/cpp/cpp/header-files-cpp?view=msvc-170 for some tutorials and examples. blush

tmori, 2023-12-30:
相乗りして、箱庭のMATLAB／Simulink対応の設計メモです。

以下を決めていく必要がありますね。

* 物理式の関数インタフェース
* 入力データ型
* 出力データ型

[image]

kenjihiranabe, 2024-01-16:
明日その方向で考えます。

検討点
- うまくやると utest.cpp にあるユニットテストがそのまま使えて、全体に組み込む前に単体テスト可能。
- 配置ディレクトリは、もしかしたら、hakoniwa-px4sim/matlab-if としてphysics から切り離しておいた方が、いろんなモジュールで matlab 実装が膨らむことができてよいかも
-

kenjihiranabe, 2024-03-28:
できたのでクローズ

closed 2024-03-28 by kenjihiranabe