toppers/hakoniwa-px4sim #117: 機体のサイズを変更してみる

tmori, 2024-01-04:


tmori, 2024-01-04:
# 小型ドローン向けの機体パラメータ

本体重量：0.1kg(100グラム)
幅：0.1m
奥行：0.1m

上記をベースにした、箱庭機体パラメータを以下とする。
なお、inertia, parameterB, parameterJrはパラメータ探索で調査する。
```
  "components": {
      "droneDynamics": {
          "physicsEquation": "BodyFrame",
          "airFrictionCoefficient": 0.324,
          "inertia": [ 0.02, 0.02, 0.04 ],
          "mass_kg": 0.1,
          "position_meter": [ 0, 0, 0 ],
          "angle_degree": [ 0, 0, 90 ]
        },
        "rotor": {
          "Tr": 0.05,
          "Kr": 8000,
          "rpmMax": 8000
        },
        "thruster": {
          "rotorPositions": [
            {"position": [  0.05,  0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [ -0.05, -0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [  0.05, -0.05, 0 ], "rotationDirection": -1.0 },
            {"position": [ -0.05,  0.05, 0 ], "rotationDirection": -1.0 }
            ],
          "HoveringRpm": 4000,
          "parameterB":  60.0e-09,
          "parameterJr": 5.0e-08
        },
```

tmori, 2024-01-04:
# 安定飛行したコンフィグパラメータ

やっていて気づいたこと。

1. inertiaは計算結果通りだとかなりふらつくので 1/10にした。
2. 機体の回転速度が速いと `WARN  [health_and_arming_checks] Compass needs calibration - Land now!`が出やすいのでは？と思った。

```
{
  "simulation": {
    "lockstep": true,
    "timeStep": 0.003,
    "logOutputDirectory": "./",
    "logOutput": {
      "sensors": {
          "acc": true,
          "gyro": true,
          "mag": true,
          "baro": true,
          "gps": true    
      },
      "mavlink": {
          "hil_sensor": true,
          "hil_gps": true,
          "hil_actuator_controls": true
      }
    },
    "mavlink_tx_period_msec": {
      "hil_sensor": 3,
      "hil_gps": 3
    },
    "location": {
      "latitude": 47.641468,
      "longitude": -122.140165,
      "altitude": 121.321,
      "magneticField": {
        "intensity_nT": 53045.1,
        "declination_deg": 15.306,
        "inclination_deg": 68.984
      }  
    }
  },
  "components": {
      "droneDynamics": {
          "physicsEquation": "BodyFrame",
          "airFrictionCoefficient": 0.0001,
          "inertia": [ 0.0004, 0.0004, 0.0008 ],
          "mass_kg": 0.1,
          "position_meter": [ 0, 0, 0 ],
          "angle_degree": [ 0, 0, 90 ]
        },
        "rotor": {
          "Tr": 0.1,
          "Kr": 8000,
          "rpmMax": 8000
        },
        "thruster": {
          "rotorPositions": [
            {"position": [  0.05,  0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [ -0.05, -0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [  0.05, -0.05, 0 ], "rotationDirection": -1.0 },
            {"position": [ -0.05,  0.05, 0 ], "rotationDirection": -1.0 }
            ],
          "HoveringRpm": 4000,
          "parameterB":  2.0e-09,
          "parameterJr": 1.0e-08
        },
        "sensors": {
          "acc": {
            "sampleCount": 1,
            "noise": 0.03
          },
          "gyro": {
            "sampleCount": 1,
            "noise": 0.01
          },
          "mag": {
            "sampleCount": 1,
            "noise": 0.03
          },
          "baro": {
            "sampleCount": 1,
            "noise": 0.005
          },
          "gps": {
            "sampleCount": 1,
            "noise": 0
          }
        }
  }
}
```

tmori, 2024-01-05:
# 気づいたこと

長距離移動時の左展開移動は安定している。（約５０ｍ）

[image]

しかし

kenjihiranabe, 2024-01-05:
昨日，僕が勘違いしていた点．質量が端っこにあると思っていたので，腕の長さが大きく効くと思っていたが，むしろ中心に近づいているのだね（腕の長さはトルクには効くがイナーシャには効かない）．

全体が，半径 0.025[m] の球だとしてイナーシャを計算すると，

Ixx = Iyy = Izz = (2/5)MR^2 = (2/5)(0.1)(0.025)^2 = 2.5 x 10^{-5}

0.03 [m] の球だと，7.2 x 10^{-4} =3.6 x 10^{-5}

くらいなので，もっと小さいかもしれない．（ @tmori さんの計算で，m は (1/4)m の間違いなのと，端に重さがある仮定が間違っていて，かなり大きめのイナーシャが森さんの計算では出ている模様．

tmori, 2024-01-05:
> くらいなので，もっと小さいかもしれない．（ @tmori さんの計算で，m は (1/4)m の間違いなのと，端に重さがある仮定が間違っていて，かなり大きめのイナーシャが森さんの計算では出ている模様．
計算間違いのご指摘ありがとうございます。

たしかに 1/4 しないとだめでした。

再度計算しなおしします。

全体質量(4m)=0.1kg
ローター質量：0.025kg
半径(a)：0.05m

Iz = 8 x 0.025kg x (0.05m)^2 = 0.0005 kg mg^2
Ix = Iy = 0.00025kg mg^2

kenjihiranabe, 2024-01-05:
数値はいい線な感じがします．
ただ，「全体質量(4m)=0.1kg，ローター質量：0.025kg」ですが，ロータよりも中心のボディ部分が重いのでは？

tmori, 2024-01-05:
> 数値はいい線な感じがします． ただ，「全体質量(4m)=0.1kg，ローター質量：0.025kg」ですが，ロータよりも中心のボディ部分が重いのでは？

あー、その観点はなかったですね。確かに。
中心部の重量を含めてみます。

kenjihiranabe, 2024-01-05:
ん？全体質量には，ロータとボディが含まれていて，合わせて0.1kg ですよね？ロータ単体の質量も仕様にあったりするの？

tmori, 2024-01-05:
> 全体質量には，ロータとボディが含まれていて，合わせて0.1kg ですよね？
「約80 g（プロペラとバッテリー含む）」と書いてあるので、全体の質量だと思います。

ひとまず、以下くらいで配分してみようかと。
全体質量=0.1kg
中心質量=0.05kg
全ローター質量(4m)：0.05kg
半径(a)：0.05m

kenjihiranabe, 2024-01-05:
上の写真みると，バッテリーが中央にあり，もっと中心が重そう．．．だけど，それでも上記の計算やってみると多分イナーシャ一桁下がると思う．

tmori, 2024-01-05:
中心質量を3/4 にしてみました。

全体質量 | 0.1 | kg
-- | -- | --
中心質量 | 0.075 | kg
全ローター質量 | 0.025 | kg
ローター質量 | 0.00625 | kg
半径 | 0.05 | m
Iz | 0.000125 | kg m^2
Ix=Iy | 0.0000625 | kg m^2

kenjihiranabe, 2024-01-05:
[image]

おお，ありがとう．半径ですが，ガード部分が入っている（？）のでさらに短そう．
中心部分のイナーシャはどうやって求めました？
（円筒にするのが良さそう．Iy は Ix の2倍くらいありそうですね）

tmori, 2024-01-05:
> ![image](https://private-user-images.githubusercontent.com/164193/294297252-8cec4f69-623a-4d5b-8326-76db096118f1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3Lm [...]

以下の計算式で求めましたが、
Iy=Iz=Ix/2
くらいでしょうか？

[image]

kenjihiranabe, 2024-01-05:
上記のスクショのすぐ下部分に，円筒の計算があって，中心部（全質量の75%）は円筒計算でやるのだと思います．（この絵のzをxにとって）
[image]

tmori, 2024-01-05:
なるほどです。
中心部の質量：c_m
ローターの質量：m
半径：a
とした場合、

Iz = c_m * a^2 / 2
Ix = 4 * m * a^2
Iy = Ix / 2

ですかね？

tmori, 2024-01-05:
計算結果：

項目 | 値 | 単位
-- | -- | --
全体質量 | 0.1 | kg
中心質量 | 0.075 | kg
全ローター質量 | 0.025 | kg
ローター質量 | 0.00625 | kg
半径 | 0.05 | m
Iz | 0.00009375 | kg m^2
Ix | 0.0000625 | kg m^2
Ix | 0.00003125 | kg m^2

tmori, 2024-01-05:
このパラメータセットでひとまずリグレッションテスト合格しました。

[image]

tmori, 2024-01-05:
# 第二関門通過

[image]

tmori, 2024-01-05:
# パラメータ探索を始めます。

ベースパラメータ：実際の機体をベースにパラメータチューニングしたものをベースにする

```
{
  "simulation": {
    "lockstep": true,
    "timeStep": 0.003,
    "logOutputDirectory": "/Users/tmori/project/oss/hakoniwa-px4sim/px4/auto-test/test_resource/logs/",
    "logOutput": {
      "sensors": {
          "acc": true,
          "gyro": true,
          "mag": true,
          "baro": true,
          "gps": true    
      },
      "mavlink": {
          "hil_sensor": true,
          "hil_gps": true,
          "hil_actuator_controls": true
      }
    },
    "mavlink_tx_period_msec": {
      "hil_sensor": 3,
      "hil_gps": 3
    },
    "location": {
      "latitude": 47.641468,
      "longitude": -122.140165,
      "altitude": 121.321,
      "magneticField": {
        "intensity_nT": 53045.1,
        "declination_deg": 15.306,
        "inclination_deg": 68.984
      }  
    }
  },
  "components": {
      "droneDynamics": {
          "physicsEquation": "BodyFrame",
          "airFrictionCoefficient": 0.0001,
          "inertia": [ 0.0000625, 0.00003125, 0.00009375 ],
          "mass_kg": 0.1,
          "position_meter": [ 0, 0, 0 ],
          "angle_degree": [ 0, 0, 90 ]
        },
        "rotor": {
          "Tr": 0.14,
          "Kr": 8000,
          "rpmMax": 8000
        },
        "thruster": {
          "rotorPositions": [
            {"position": [  0.05,  0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [ -0.05, -0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [  0.05, -0.05, 0 ], "rotationDirection": -1.0 },
            {"position": [ -0.05,  0.05, 0 ], "rotationDirection": -1.0 }
            ],
          "HoveringRpm": 4000,
          "parameterB":  3.0e-11,
          "parameterJr": 1.0e-10
        },
        "sensors": {
          "acc": {
            "sampleCount": 1,
            "noise": 0.03
          },
          "gyro": {
            "sampleCount": 1,
            "noise": 0.01
          },
          "mag": {
            "sampleCount": 1,
            "noise": 0.03
          },
          "baro": {
            "sampleCount": 1,
            "noise": 0.01
          },
          "gps": {
            "sampleCount": 1,
            "noise": 0
          }
        }
  }
}

```

パラメータ探索範囲：
B, Jr を以下の範囲で回します

B：1.0 2.0 3.0 4.0. 5.0 6.0 7.0 8.0 9.0 10.0　X（e-11)
Jr：1.0 2.0 4.0 8.0 16.0　　　X（e-10）

tmori, 2024-01-05:
# 最強のパラメータに出会ってしまいました

B=3.0e-10, Jr=6.0e-10

移動距離のスコア：

```
param1=3.0 param2=6.0 d=215.96753540135455 score=84.12552504601457
```
[image]

警告メッセージ：なし
```
INFO: START TEST PARAM=3.0-6.0
INFO: Activating hakoniwa
HAKO_CAPTURE_SAVE_FILEPATH : ./capture.bin
HAKO_BYPASS_IPADDR : 127.0.0.1
HAKO_CUSTOM_JSON_PATH : ../config/custom.json
DRONE_CONFIG_PATH : /Users/tmori/project/oss/hakoniwa-px4sim/px4/auto-test/test_resource/work/drone_config.json
HAKO_BYPASS_PORTNO : 54001
INFO: shmget() key=255 size=1129352 
INFO: hako_master_init() success
Robot: DroneAvator, PduWriter: DroneAvator_drone_motor
channel_id: 0 pdu_size: 88
INFO: DroneAvator create_lchannel: logical_id=0 real_id=0 size=88
Robot: DroneAvator, PduWriter: DroneAvator_drone_pos
channel_id: 1 pdu_size: 48
INFO: DroneAvator create_lchannel: logical_id=1 real_id=1 size=48
WAIT START
INFO: Activating PX4
ps: stdin: not a terminal
[0/1] launching px4 none_iris (SYS_AUTOSTART=10016)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
env SYS_AUTOSTART: 10016
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 971 bytes, decoded 971 bytes (INT32:15, FLOAT:29)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7868392 bytes
INFO  [init] PX4_SIM_HOSTNAME: localhost
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator_mavlink] Simulator connected on TCP port 4560.
INFO: px4 reciver start
INFO: COMMAND_LONG ack sended
ERROR [simulator_mavlink] poll timeout 0, 22
ERROR [simulator_mavlink] poll timeout 0, 22
INFO: start simulation
start
WAIT RUNNING
PDU CREATED
START CREATE PDU DATA: total_size= 136
INFO: shmget() key=256 size=136 
PDU DATA CREATED
CREATED ADDR=0x104cd400c
INFO: setup start
param_A: 1.53281e-08
param_B: 3e-10
param_Jr: 6e-10
INFO: setup done
INFO: start simulation
[... 181 lines cut ...]
mean_score: 88.84986630897605
upper_score: 89.89599151920294
lower_score: 87.7623571531925
total_score: 88.83607166045716
Score: 88.83607166045716
relative_start_time: 209.278
relative_end_time: 214.278
actual_mean: 39.98044079244151
actual_stdev: 0.01815650903326014
mean_score: 98.10536517694351
upper_score: 99.99016266813388
lower_score: 93.13466179033084
total_score: 97.07672987846941
Score: 97.07672987846941
relative_start_time: 209.278
relative_end_time: 214.278
actual_mean: -0.09863105338932214
actual_stdev: 0.07480280232854988
mean_score: 61.48332201911916
upper_score: 97.20099116229267
lower_score: 22.22471750270389
total_score: 60.30301022803857
Score: 60.30301022803857
INFO: END TEST
```

tmori, 2024-01-05:
# B＝3.0までの評価結果

3.0になると振動することで、安定飛行が達成されている模様。

[image]

tmori, 2024-01-05:
# 次の探索範囲

B＝1.5, Jr=10 の場合をやってみたら、警告は多少出るけれど、いい感じの動きだった。
[image]

tmori, 2024-01-05:
# 探索結果

[image]
[distance_result.csv](https://github.com/toppers/hakoniwa-px4sim/files/13840758/distance_result.csv)

kenjihiranabe, 2024-01-05:
楽しみ

tmori, 2024-01-09:
# Bを大きくした時の調査結果（色々いじった結果）

* Bを大きくすると、振動がすごく大きくなる
* Jrを大きくすることで解消できる
* B=30の場合、Jr=2500-4000 くらい
* 縦揺れが出る場合、Krを小さくすると解消される場合がある

tmori, 2024-01-09:
# 次のパラメータ探索範囲

方針１：Bは大きくする（今までは1-3の間だったけれど、これを30にしてみる）
方針２：Jrは、固定値とする（Jr=2500)
方針３：TrとKrを変えてみる

調査範囲
Kr=5000, 6000, 7000, 8000
Tr=0.05, 1.0, 1.5, 2.0

tmori, 2024-01-09:
# 探索結果

[image]
[distance_result.csv](https://github.com/toppers/hakoniwa-px4sim/files/13868495/distance_result.csv)

tmori, 2024-01-09:
# 次は、Kr=6000にして、Trを0.08 0.09 0.1 0.11 0.12の範囲で調べる

tmori, 2024-02-17:
機体サイズの変更としてのパラメータ探索は一旦終了します。
パラメータ探索のissueで引き続き検討する。

closed 2024-02-17 by tmori