toppers/hakoniwa-px4sim #118: シンプル化のリファクタリング <tuple> をやめて，シンプルな struct に変更

kenjihiranabe, 2024-01-05:
作戦

- VectorType { double x, y, z; } AngularType{ double phi, theta, psi } としてタイプセーフに（速度/加速度と角速度/角加速度に分け，角速度グループと速度グループの中は区別しない）
- 内部を変更し，テストも書き直す
- @tmori さん部分のコンバーターコードを変更

kenjihiranabe, 2024-01-05:
https://github.com/toppers/hakoniwa-px4sim/commit/b31e212eb8306c3249f39f414d598dea1375136e

前コンパイルが通って，std::tuple への依存を全コードから削除．（third-party除く)

@tmori `tupleToVec3` というコードをそのままの名前で残していますが，うまく glm と 整合が取れるならこのメソッドも捨てていい感じにしてください．現在，tuple という文字列で全検索すると，このメソッドのみが引っかかります．

kenjihiranabe, 2024-01-05:
メモ：
素の構造体にしているので，直値は，VectorType v = {1,1,1} ; とか，関数引数なら，VectorType{1,1,1} などとやります．VectorType(1,1,1) は動かないので注意．

tmori, 2024-01-05:
> [b31e212](https://github.com/toppers/hakoniwa-px4sim/commit/b31e212eb8306c3249f39f414d598dea1375136e)
> 
> 前コンパイルが通って，std::tuple への依存を全コードから削除．（third-party除く)
> 
> @tmori `tupleToVec3` というコードをそのままの名前で残していますが，うまく glm と 整合が取れるならこのメソッドも捨てていい感じにしてください．現在，tuple という文字列で全検索すると，このメソッドのみが引っかかります．

ご対応ありがとうございました！とってもスッキリした感ありますね！
不要メソッドの件、了解です！

closed 2024-01-06 by kenjihiranabe