toppers/hakoniwa-px4sim #119: Unityで、オブジェクトが別のオブジェクトに衝突した時に、受ける外力を計算 or 取得可能かどうか調べる

tmori, 2024-01-06:
# 初期調査結果と見通し

Unityには、オブジェクト衝突時のコールバックイベント関数 OnCollisionEnter がある。

ここで、衝突時の相対速度（collision.relativeVelocity）を取得できる
```
void OnCollisionEnter(Collision collision)
{
    // 衝突時の相対速度を取得できる
    Vector3 relativeVelocity = collision.relativeVelocity;

}
```

relativeVeloctyとは：

```
relativeVelocity は、Unityの物理エンジンにおいて、衝突が発生した際の2つのオブジェクト間の相対速度を表します。具体的には、この値は衝突時の一方のオブジェクトが他方のオブジェクトに対して持つ速度ベクトルです。

たとえば、2つの動くオブジェクトが衝突する場合、relativeVelocityは以下のように計算されます：

relativeVelocity = Va - Vb

この値は、オブジェクト間の速度差を表しています。これにより、衝突の「強さ」や方向を推定でき、衝突の影響をシミュレーションする際に使用できます。
```

UnityのrelativeVelocityにおいて、どちらのオブジェクトから見た相対速度かという点は重要です。Unityのドキュメンテーションによると、relativeVelocityは「この衝突イベントに関連するオブジェクト（スクリプトがアタッチされているオブジェクト）から見た、衝突相手のオブジェクトの速度」と定義されています。

つまり、OnCollisionEnter(Collision collision)関数内でcollision.relativeVelocityを使用する場合、その値はスクリプトがアタッチされているオブジェクトの視点から計算された、衝突相手のオブジェクトの速度です。これは、スクリプトがアタッチされているオブジェクトが静止している場合、衝突相手のオブジェクトの速度そのものを表します。もし両方のオブジェクトが動いている場合は、スクリプトがアタッチされたオブジェクトに対する相手の相対速度を表します。

例えば、オブジェクトA（スクリプトがアタッチされているオブジェクト）が静止していて、オブジェクトBがそれに向かって移動している場合、relativeVelocityはオブジェクトBの速度になります。逆に、オブジェクトAが動いていて、オブジェクトBが静止している場合、relativeVelocityはオブジェクトAの速度の逆方向になります。そして、両方のオブジェクトが動いている場合は、その相対速度が計算されます。

tmori, 2024-01-06:
あー、これ、万博向けのAIアバターにも使えるな。

tmori, 2024-01-06:
relativeVelocityから、反発力を計算する方法（例）：

```
Vector3 reboundForce = relativeVelocity * someFactor; // someFactorは適切な係数
```

kenjihiranabe, 2024-01-06:
droneオブジェクトの速度を計算した値にリセットしてしまうのが楽。
力でやろうとすると、与え続ける時間が必要になるし、たぶん障害物にめり込む。

tmori, 2024-01-06:
Unity側の対応項目を検討

https://github.com/toppers/hakoniwa-unity-drone-model/issues/6

tmori, 2024-01-07:
Unity側の修正が終わった！

https://github.com/toppers/hakoniwa-unity-drone-model/issues/6
https://github.com/toppers/hakoniwa-unity-drone-model/issues/7

tmori, 2024-01-07:
上記で対応できたので、ひとまずクローズします。

closed 2024-01-07 by tmori