toppers/hakoniwa-px4sim #123: 地上座標系と機体座標系の物理モデルの実験をやって比較する

tmori, 2024-01-09:


tmori, 2024-01-09:
# 機体座標系の結果

[alt_result-p6000-p0_11.txt](https://github.com/toppers/hakoniwa-px4sim/files/13870553/alt_result-p6000-p0_11.txt)

[drone_dynamics-p6000-p0.11.csv](https://github.com/toppers/hakoniwa-px4sim/files/13870548/drone_dynamics-p6000-p0.11.csv)

[image]

tmori, 2024-01-09:
# 地上座標家の結果

[alt_result-p6000-p0_11.txt](https://github.com/toppers/hakoniwa-px4sim/files/13870695/alt_result-p6000-p0_11.txt)

[drone_dynamics-p6000-p0.11.csv](https://github.com/toppers/hakoniwa-px4sim/files/13870698/drone_dynamics-p6000-p0.11.csv)

[image]

kenjihiranabe, 2024-01-09:
ぴったり一緒とは行かないみたいだけど，まあ，数値計算の誤差程度に思っておきますかね？

tmori, 2024-01-09:
# 比較結果

ドローンの位置はほぼ同じ動きをしている。
（シミュレーション結果はパソコンの負荷によって変わるので仕方ない）

１点記になるのは、地上座標系のPX4のログでは、以下の警告が発生している点。
（機体座標系の場合は出力されていない）

```
WARN  [health_and_arming_checks] Compass needs calibration - Land now!
```

kenjihiranabe, 2024-01-09:
gyro がうまく同期していない？

tmori, 2024-01-09:
gyroとmagのセンシングデータをもう一度採取してみます。

tmori, 2024-01-09:
# 機体座標系の全データ

[logs_body.zip](https://github.com/toppers/hakoniwa-px4sim/files/13870900/logs_body.zip)

tmori, 2024-01-09:
# 地上座表系の全データ

[Uploading logs_ground.zip…]()

tmori, 2024-01-09:
地上座標系と機体座標系とで Rzの値が全然違う。真逆？

kenjihiranabe, 2024-01-09:
@tmori said ... 初期の方向は東方向に向いていて、ミッションとして、真西に移動させるので、方向転換するのですが、
機体座標系の場合と地上座標系とで、選択した旋回方向が違うようですね。
[image]

kenjihiranabe, 2024-01-09:
選択の違いであれば，分からないですね．．．．でもエラーは気になる．

tmori, 2024-01-09:
> 選択の違いであれば，分からないですね．．．．でもエラーは気になる．

ですよねぇ。明日、もう少し調べてみます。

tmori, 2024-01-11:
# 再チャレンジ

rotor_thruster：nonlinear
パラメータ：Kr=6000, Tr=0.11

コンフィグ：

```
{
  "simulation": {
    "lockstep": true,
    "timeStep": 0.003,
    "logOutputDirectory": "/Users/tmori/project/oss/hakoniwa-px4sim/px4/auto-test/test_resource/logs/",
    "logOutput": {
      "sensors": {
          "acc": true,
          "gyro": true,
          "mag": true,
          "baro": true,
          "gps": true    
      },
      "mavlink": {
          "hil_sensor": true,
          "hil_gps": true,
          "hil_actuator_controls": true
      }
    },
    "mavlink_tx_period_msec": {
      "hil_sensor": 3,
      "hil_gps": 3
    },
    "location": {
      "latitude": 47.641468,
      "longitude": -122.140165,
      "altitude": 121.321,
      "magneticField": {
        "intensity_nT": 53045.1,
        "declination_deg": 15.306,
        "inclination_deg": 68.984
      }  
    }
  },
  "components": {
      "droneDynamics": {
          "physicsEquation": "BodyFrame",
          "collision_detection": false,
          "airFrictionCoefficient": 0.0001,
          "inertia": [ 0.0000625, 0.00003125, 0.00009375 ],
          "mass_kg": 0.1,
          "body_size": [ 0.1, 0.1, 0.01 ],
          "position_meter": [ 0, 0, 0 ],
          "angle_degree": [ 0, 0, 90 ]
        },
        "rotor": {
          "vendor": "None",
          "Tr": 0.11,
          "Kr": 6000,
          "rpmMax": 8000
        },
        "thruster": {
          "vendor": "None",
          "rotorPositions": [
            {"position": [  0.05,  0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [ -0.05, -0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [  0.05, -0.05, 0 ], "rotationDirection": -1.0 },
            {"position": [ -0.05,  0.05, 0 ], "rotationDirection": -1.0 }
            ],
          "HoveringRpm": 4000,
          "parameterB":  1.3e-10,
          "parameterJr": 6.0e-10
        },
        "sensors": {
          "acc": {
            "sampleCount": 1,
            "noise": 0.03
          },
          "gyro": {
            "sampleCount": 1,
            "noise": 0.01
          },
          "mag": {
            "sampleCount": 1,
            "noise": 0.03
          },
          "baro": {
            "sampleCount": 1,
            "noise": 0.01
          },
          "gps": {
            "sampleCount": 1,
            "noise": 0
          }
        }
  }
}

```

tmori, 2024-01-11:
# 機体座標系

[image]

[image]

compass 系のWARNINGが最後の方に出ているけど、まー、たまには出る。

```
tmori@TakashinoMacBook-Pro hakoniwa-px4sim % bash px4/auto-test/auto-test.bash ./px4/auto-test/test_resource ./px4/auto-test/test_scenario/takeoff_and_move.json
INFO: START TEST PARAM=6000-0.11
INFO: Activating hakoniwa
HAKO_CAPTURE_SAVE_FILEPATH : ./capture.bin
HAKO_BYPASS_IPADDR : 127.0.0.1
HAKO_CUSTOM_JSON_PATH : ../config/custom.json
DRONE_CONFIG_PATH : /Users/tmori/project/oss/hakoniwa-px4sim/./px4/auto-test/test_resource/work/drone_config.json
HAKO_BYPASS_PORTNO : 54001
INFO: shmget() key=255 size=1129352 
INFO: hako_master_init() success
Robot: DroneAvator, PduWriter: DroneAvator_drone_motor
channel_id: 0 pdu_size: 88
INFO: DroneAvator create_lchannel: logical_id=0 real_id=0 size=88
Robot: DroneAvator, PduWriter: DroneAvator_drone_pos
channel_id: 1 pdu_size: 48
INFO: DroneAvator create_lchannel: logical_id=1 real_id=1 size=48
WAIT START
INFO: Activating PX4
ps: stdin: not a terminal
[0/1] launching px4 none_iris (SYS_AUTOSTART=10016)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
env SYS_AUTOSTART: 10016
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 971 bytes, decoded 971 bytes (INT32:15, FLOAT:29)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7868392 bytes
INFO  [init] PX4_SIM_HOSTNAME: localhost
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator_mavlink] Simulator connected on TCP port 4560.
INFO: px4 reciver start
INFO: COMMAND_LONG ack sended
ERROR [simulator_mavlink] poll timeout 0, 22
ERROR [simulator_mavlink] poll timeout 0, 22
INFO: start simulation
start
WAIT RUNNING
START CREATE PDU DATA: total_size= 136
INFO: shmget() key=256 size=136 
PDU DATA CREATED
CREATED ADDR=0x100bfc00c
PDU CREATED
INFO: setup start
Rotor vendor: None
Thruster vendor: None
param_A: 1.53281e-08
param_B: 1.3e-10
param_Jr: 6e-10
INFO: setup done
INFO: start simulation
SYNC MODE: true
[... 188 lines cut ...]
upper_score: 98.47268919322288
lower_score: 73.90993666042374
total_score: 87.21663915311585
Score: 87.21663915311585
relative_start_time: 209.281
relative_end_time: 214.281
actual_mean: 40.4612796574685
actual_stdev: 0.35806514298238784
mean_score: 0.0023964281195936682
upper_score: 2.6444981014907484e-13
lower_score: 58.70403042893132
total_score: 19.56880895235039
Score: 19.56880895235039
relative_start_time: 209.281
relative_end_time: 214.281
actual_mean: -0.0017424037192561783
actual_stdev: 0.36738110014568515
mean_score: 99.98482129847709
upper_score: 0.12500361398311918
lower_score: 0.10998207628956566
total_score: 33.40660232958326
Score: 33.40660232958326
INFO: END TEST

```

kenjihiranabe, 2024-01-11:
今までの修正では結果変わらないと思う。
名前直してコンパイル通ったら、ぼくの方でも考えないといけないとろがあるはず。

tmori, 2024-01-11:
やっぱり、そうでしたか。
地上座標系の修正施して、動かしてみましたが、不安定になってしまいましたので。。

tmori, 2024-01-13:
# 再再チャレンジ

コンフィグ：

```json
{
  "simulation": {
    "lockstep": true,
    "timeStep": 0.003,
    "logOutputDirectory": "/Users/tmori/project/oss/hakoniwa-px4sim/px4/auto-test/test_resource/logs/",
    "logOutput": {
      "sensors": {
          "acc": true,
          "gyro": true,
          "mag": true,
          "baro": true,
          "gps": true    
      },
      "mavlink": {
          "hil_sensor": true,
          "hil_gps": true,
          "hil_actuator_controls": true
      }
    },
    "mavlink_tx_period_msec": {
      "hil_sensor": 3,
      "hil_gps": 3
    },
    "location": {
      "latitude": 47.641468,
      "longitude": -122.140165,
      "altitude": 121.321,
      "magneticField": {
        "intensity_nT": 53045.1,
        "declination_deg": 15.306,
        "inclination_deg": 68.984
      }  
    }
  },
  "components": {
      "droneDynamics": {
          "physicsEquation": "BodyFrame",
          "collision_detection": false,
          "airFrictionCoefficient": 0.0001,
          "inertia": [ 0.0000625, 0.00003125, 0.00009375 ],
          "mass_kg": 0.1,
          "body_size": [ 0.1, 0.1, 0.01 ],
          "position_meter": [ 0, 0, 0 ],
          "angle_degree": [ 0, 0, 90 ]
        },
        "rotor": {
          "vendor": "None",
          "Tr": 0.1,
          "Kr": 6000,
          "rpmMax": 8000
        },
        "thruster": {
          "vendor": "None",
          "rotorPositions": [
            {"position": [  0.05,  0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [ -0.05, -0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [  0.05, -0.05, 0 ], "rotationDirection": -1.0 },
            {"position": [ -0.05,  0.05, 0 ], "rotationDirection": -1.0 }
            ],
          "HoveringRpm": 4000,
          "parameterB":  30.0e-10,
          "parameterJr": 3000.0e-10
        },
        "sensors": {
          "acc": {
            "sampleCount": 1,
            "noise": 0.03
          },
          "gyro": {
            "sampleCount": 1,
            "noise": 0.01
          },
          "mag": {
            "sampleCount": 1,
            "noise": 0.03
          },
          "baro": {
            "sampleCount": 1,
            "noise": 0.01
          },
          "gps": {
            "sampleCount": 1,
            "noise": 0
          }
        }
  }
}

```

tmori, 2024-01-13:
# 機体座標系

[image]
[Uploading drone_dynamics.csv…]()

tmori, 2024-01-13:
# 地上座標系
[image]
[drone_dynamics.csv](https://github.com/toppers/hakoniwa-px4sim/files/13927486/drone_dynamics.csv)

tmori, 2024-01-13:
# 結論

同じ結果に見えますね。compassエラーもない。

kenjihiranabe, 2024-01-13:
Yeah! 手計算で微分してみました。

kenjihiranabe, 2024-01-14:
残すはドキュメントの変更のみ。

tmori, 2024-02-17:
地上座標系の対応は以下で実施しているのためクローズします。

https://github.com/toppers/hakoniwa-px4sim/blob/main/hakoniwa/src/assets/drone/physics/ground_frame/drone_dynamics_ground_frame.hpp

closed 2024-02-17 by tmori