toppers/hakoniwa-px4sim #124: ルンゲクッタで試してみる

tmori, 2024-01-09:


tmori, 2024-01-11:
# 前提

以下の２つの関数をそのまま利用する。

* drone_physics::acceleration_in_body_frame()
* drone_physics::angular_acceleration_in_body_frame()

tmori, 2024-01-11:
# コード設計

上記どちらの関数も３次元のベクトルなので、３次元ベクトル変換関数(f_vec)と思うことにする。

## 変数定義

* c_vec：現在の状態
* r_vec：次の状態

tmori, 2024-01-11:
# k1

r_vec = f_vec(c_vec);
k1 = r_vec;

tmori, 2024-01-11:
# k2

k2_vec = {c_vec.x + 0.5 * k1.x * delta_time, c_vec.y + 0.5 * k1.y * delta_time, c_vec.z + 0.5 * k1.z * delta_time};
k2 = f_vec(k2_vec);

tmori, 2024-01-11:
# 次の状態

n_vec = {
    c_vec.x + (delta_time / 6.0) * (k1.x + 2 * k2.x + 2 * k3.x + k4.x),
    c_vec.y + (delta_time / 6.0) * (k1.y + 2 * k2.y + 2 * k3.y + k4.y),
    c_vec.z + (delta_time / 6.0) * (k1.z + 2 * k2.z + 2 * k3.z + k4.z)
};

tmori, 2024-01-11:
というわけで修正してみます。

tmori, 2024-01-11:
めちゃくちゃ冗長だけど、やってみた。

https://github.com/toppers/hakoniwa-px4sim/commit/1e8b264684f282180fd41ecd1f400431a3b66cf7

tmori, 2024-01-19:
ルンゲクッタの実装はこちらです。お暇な時にでもみていただけると嬉しいです。

https://github.com/toppers/hakoniwa-px4sim/blob/25f24e548fb816e6134083f7570cbcbe910b7a95/hakoniwa/src/assets/drone/physics/body_frame_rk4/drone_dynamics_body_frame_rk4.hpp#L218

tmori, 2024-01-20:
# 比較検証結果

PX4を使わずにPID制御実施した結果。

## オイラー法

[image]

[psi_data.csv](https://github.com/toppers/hakoniwa-px4sim/files/13996227/psi_data.csv)
[theta_data.csv](https://github.com/toppers/hakoniwa-px4sim/files/13996228/theta_data.csv)
[phi_data.csv](https://github.com/toppers/hakoniwa-px4sim/files/13996230/phi_data.csv)
[height_data.csv](https://github.com/toppers/hakoniwa-px4sim/files/13996231/height_data.csv)

## ルンゲクッタ法

[image]

[psi_data.csv](https://github.com/toppers/hakoniwa-px4sim/files/13996210/psi_data.csv)
[theta_data.csv](https://github.com/toppers/hakoniwa-px4sim/files/13996211/theta_data.csv)
[phi_data.csv](https://github.com/toppers/hakoniwa-px4sim/files/13996212/phi_data.csv)
[height_data.csv](https://github.com/toppers/hakoniwa-px4sim/files/13996213/height_data.csv)

tmori, 2024-01-20:
csv ファイルの diff を取ってみたところ、差分無しでした。
なので、RK4の実装はあっているように思えます。

kenjihiranabe, 2024-01-20:
よかったね。十分δ時間がカーブに比べて短いので差が出なかった、ということでしょうね。

tmori, 2024-02-17:
以下で対応済みのためクローズします。

https://github.com/toppers/hakoniwa-px4sim/blob/main/hakoniwa/src/assets/drone/physics/body_frame_rk4/drone_dynamics_body_frame_rk4.hpp

closed 2024-02-17 by tmori