toppers/hakoniwa-px4sim #125: ローターとthrusterをjmavsimの実装に合わせてみる

tmori, 2024-01-09:
ローター：

https://github.com/PX4/jMAVSim/blob/66b764ada522893c05224950aa6268c809f8e48a/src/me/drton/jmavsim/Rotor.java#L22

Thruster：
https://github.com/PX4/jMAVSim/blob/66b764ada522893c05224950aa6268c809f8e48a/src/me/drton/jmavsim/vehicle/AbstractMulticopter.java#L142C42-L142C42

kenjihiranabe, 2024-01-09:
微分方程式を除いて，線形に duty rate と回転数をリンクさせるモデルです．

https://github.com/toppers/hakoniwa-px4sim/commit/37d4ef48472c95802894db9255ce0dfdb4d29806

thrust_dynamics.hpp の body_thrust と body_torque の呼び出しをそれぞれ，body_thrust_linear, body_torque_linear に変更してください．また，Kr, Tr はそのまま，A の値を回転数分の１（1/8000） くらいにして呼び出してください．新しくパラメータを A2 とかコンフィグ追加がいいと思います．

tmori, 2024-01-10:
linear用のパラメータ仕様を検討しました。

# rotor

|パラメータ|利用有無|備考|
|---|---|---|
|Tr|O|そのまま利用|
|Kr|x|利用しない（あっても無視する）|
|rpmMax|x|利用しない（あっても無視する）|

# thruster

|パラメータ|利用有無|備考|
|---|---|---|
| rotorPositions |O|そのまま利用|
| HoveringRpm |x|利用しない（あっても無視する）|
| parameterB |x|利用しない（あっても無視する）|
| parameterJr |x|利用しない（あっても無視する）|
|parameterB_linear |O|新規追加|
| parameterA_linear |O|新規追加|

kenjihiranabe, 2024-01-10:
ありがとう。こちらはKr は利用前提で作りました。(ロータの微分方程式は変えていない)。パラメータとして必要です。

tmori, 2024-01-10:
> ありがとう。こちらはKr は利用前提で作りました。(ロータの微分方程式は変えていない)。パラメータとして必要です。

了解です！修正しました。

kenjihiranabe, 2024-01-10:
8000分の1の話は、2乗を1乗-linearにしたのでだいたい、標準回転数分くらいズレるだろう、の予想の数値です。

tmori, 2024-01-10:
> 8000分の1の話は、2乗を1乗-linearにしたのでだいたい、標準回転数分くらいズレるだろう、の予想の数値です。

了解です。設定値も追加しました。

tmori, 2024-01-10:
`rotor_omega_acceleration`は、そのまま利用できることをやっと理解できました。

https://github.com/PX4/jMAVSim/blob/66b764ada522893c05224950aa6268c809f8e48a/src/me/drton/jmavsim/Rotor.java#L22

の指数関数部分をテーラー展開して１次微分の項だけ取れば、こちらの数式と一致するということですね。

kenjihiranabe, 2024-01-10:
そうです！
また、今の定数Krがそれなりに感覚ができてるので崩さない方がいいかと思いそうしてます。jmav
の式にするのは簡単ですので、必要なら一回まわしてから、と思いました。

tmori, 2024-01-10:
# A2 の求め方

[image]

理想状態で、ホバリングするT_0：

dot_w = -T/m + g = 0
→
T_0=g*m

ホバリングするlinearのTは、
T_0=4 * A2 * w_0

なので、A2は、
A2 = T_0 / (4 * w_0)
→
A2 = g*m / (4 * w_0)
となる。

g=9.81, m=0.1kg, w_0=4000rpmとした場合、
A2=0.0000613125

tmori, 2024-01-10:
# コンフィグ

```
        "rotor": {
          "Tr": 0.11,
          "Kr": 6000,
          "rpmMax": 8000
        },
        "thruster": {
          "vendor": "linear",
          "rotorPositions": [
            {"position": [  0.05,  0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [ -0.05, -0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [  0.05, -0.05, 0 ], "rotationDirection": -1.0 },
            {"position": [ -0.05,  0.05, 0 ], "rotationDirection": -1.0 }
            ],
          "HoveringRpm": 4000,
          "parameterB":  30.0e-10,
          "parameterJr": 2500.0e-10,
          "parameterA_linear": 0.0000613125,
          "parameterB_linear": 0.0001666666667
        },
```

tmori, 2024-01-10:
# 調整結果

B2ですが、1/Kr だと、機体の振動が激しく、途中で変な動きになりました。
なので、0.01/Krにしたところ、安定して移動することを確認しました。
B2は、小さい方が安定することがわかってきましたので、どのあたりが良さげなパラメータになりそうな調査してみます。

kenjihiranabe, 2024-01-10:
僕の目算は、前のBの値の1/8000です。

tmori, 2024-01-10:
では、このパラメータセットを試してみます。

Kr=8000
B2=1/8000

kenjihiranabe, 2024-01-10:
あ，ごめんなさい，前動いていた parameterB * (1/8000) => parameterB_linear の意味です．

tmori, 2024-01-10:
実施してみたところ、姿勢角の値が発散しちゃいましたね。。

(Rx, Ry, Rz) = (inf,nan,inf)

kenjihiranabe, 2024-01-10:
Kr = 前動いていた Kr の値で．

tmori, 2024-01-10:
> Kr = 前動いていた Kr の値で．

ということは、
Kr=6000
parameterB_linear=1/8000
ですかね？

tmori, 2024-01-10:
30.0e-10,

tmori, 2024-01-10:
30.0e-10 x 8000  の値を使ってみます。

tmori, 2024-01-10:
# パラメータ設定の目安

* Kr: 8000（固定）
* Aは固定(rpm4000でホバリング）
* **Tr**: 00.5-0.15 (2-20倍あればOK)
* **B_linear**は、前回の値の8000倍くらい

tmori, 2024-01-10:
# ベースパラメータ

```
{
  "simulation": {
    "lockstep": true,
    "timeStep": 0.003,
    "logOutputDirectory": "./",
    "logOutput": {
      "sensors": {
          "acc": true,
          "gyro": true,
          "mag": true,
          "baro": true,
          "gps": true    
      },
      "mavlink": {
          "hil_sensor": true,
          "hil_gps": true,
          "hil_actuator_controls": true
      }
    },
    "mavlink_tx_period_msec": {
      "hil_sensor": 3,
      "hil_gps": 3
    },
    "location": {
      "latitude": 47.641468,
      "longitude": -122.140165,
      "altitude": 121.321,
      "magneticField": {
        "intensity_nT": 53045.1,
        "declination_deg": 15.306,
        "inclination_deg": 68.984
      }  
    }
  },
  "components": {
      "droneDynamics": {
          "physicsEquation": "BodyFrame",
          "collision_detection": false,
          "airFrictionCoefficient": 0.0001,
          "inertia": [ 0.0000625, 0.00003125, 0.00009375 ],
          "mass_kg": 0.1,
          "body_size": [ 0.1, 0.1, 0.01 ],
          "position_meter": [ 0, 0, 0 ],
          "angle_degree": [ 0, 0, 0 ]
        },
        "rotor": {
          "Tr": 0.05,
          "Kr": 8000,
          "rpmMax": 8000
        },
        "thruster": {
          "vendor": "linear",
          "rotorPositions": [
            {"position": [  0.05,  0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [ -0.05, -0.05, 0 ], "rotationDirection":  1.0 },
            {"position": [  0.05, -0.05, 0 ], "rotationDirection": -1.0 },
            {"position": [ -0.05,  0.05, 0 ], "rotationDirection": -1.0 }
            ],
            "parameterA_linear": 0.0000613125,
            "parameterB_linear":  2.40E-06
        },
        "sensors": {
          "acc": {
            "sampleCount": 1,
            "noise": 0.03
          },
          "gyro": {
            "sampleCount": 1,
            "noise": 0.01
          },
          "mag": {
            "sampleCount": 1,
            "noise": 0.03
          },
          "baro": {
            "sampleCount": 1,
            "noise": 0.01
          },
          "gps": {
            "sampleCount": 1,
            "noise": 0
          }
        }
  }
}

```

tmori, 2024-01-10:
# パラメータ探索範囲

Tr = 0.05 0.07 1.0 1.2 1.5
B_linear: 0.01E-06 0.1E-06 1.0E-06  2.0E-06  2.4E-06

tmori, 2024-01-10:
# 探索結果

[image]
[distance_result.csv](https://github.com/toppers/hakoniwa-px4sim/files/13886483/distance_result.csv)

tmori, 2024-01-10:
# 一番良い結果

Tr=0.12
B2=2.0e-06

元々、B2=2.4e-06 だったので、良い線いっているのでは？
と思ったけど、振動するケースだった。

[image]

[image]

tmori, 2024-01-10:
# ２番目に良い結果

Tr=0.05
B2=2.0e-06

[image]

[image]

tmori, 2024-01-10:
# 所感

linearにしても、全体的に高得点は出なかった。
あと、コントローラで操作してみたが、目的の方向への移動に対して、横揺れが大きく、操作しづらい感があった。
これは nonlinearでも同様にあった。

もう少しキビキビ動くパラメータを見つけたい。

ひとまず、以下あたりをさらに探索してみる。

Tr: 0.08 0.09 0.11
B2: 1.1 1.2 1.3 1.4 1.5

kenjihiranabe, 2024-01-10:
今日は手を出せないですが、気になるなら、そちらで例のロータの一次微分式、exp バージョンに変えて、感触試してみてみてもいいよ。
あと、並進のdrag ですが、vでなくてv^2 を使ってるモデルをネットで発見しました。でもこれも良くなる確信がぜんぜんないや。
姿勢角度の積算に、勘違いがあったので、これ直したらもしかしたら、うまく行くかも。

tmori, 2024-01-10:
> 今日は手を出せないですが、気になるなら、そちらで例のロータの一次微分式、exp バージョンに変えて、感触試してみてみてもいいよ。

了解です。考えてみます。

> あと、並進のdrag ですが、vでなくてv^2 を使ってるモデルをネットで発見しました。でもこれも良くなる確信がぜんぜんないや。 姿勢角度の積算に、勘違いがあったので、これ直したらもしかしたら、うまく行くかも。

これは、期待大ですね。

tmori, 2024-01-10:
最新の探索結果。新しい良い結果はありませんでしたね。

[image]

[distance_result.csv](https://github.com/toppers/hakoniwa-px4sim/files/13895564/distance_result.csv)

tmori, 2024-01-10:
ちょっと手詰まり感ありますが、いくつか試してみます。

1. 機能追加：jmavsim実装のローター機能を作って差し替えできるようにする
2. 機能改善：A2のパラメータはHoveringRpmで決まるので、A2のパラメータは削除して自動計算する

tmori, 2024-01-11:
jmavsimの修正しました。試してみましたが、みた感じ、同じ動きに見えましたが、念の為パラメータ探索を昨日と同じ範囲でやってみます。

https://github.com/toppers/hakoniwa-px4sim/commit/0777e75d92ae05a83a3e2cabf99e535a09ae9b60

kenjihiranabe, 2024-01-15:
これ，PID でうまく行っているので，プラント側はクローズでどうでしょう？

tmori, 2024-01-15:
いいと思います！

closed 2024-01-15 by tmori