toppers/hakoniwa-px4sim #128: 空気摩擦の項を二乗でやってみる

tmori, 2024-01-15:


tmori, 2024-01-15:
# 初見

安定飛行するパラメータに対して、空気摩擦の二乗対応を実施しました。

修正コード：acceleration_in_body_frame()

```c
    double dot_u =       - g * s_theta            - (q*w - r*v) - (c/m) * u * u;
    double dot_v =       + g * c_theta * s_phi    - (r*u - p*w) - (c/m) * v * v;
    double dot_w = -T/m  + g * c_theta * c_phi    - (p*v - q*u) - (c/m) * w * w;
```

## 手動でフライトコントロール実行した結果

以下のメッセージが出力されなかった。

```
WARN  [health_and_arming_checks] Preflight Fail: High Accelerometer Bias
```

たまたまかもしれないので、何回かやってみたところ、タイミングによってはやっぱり出力される。
ただ、頻度は低いかもしれないような気がしました。

## 位置ずれの問題が発生するか？

結論：やっぱり発生する

速度の上限を 3.5m/s から 5m/sにしたところ、水平移動時にやはり位置ずれ問題が発生しました。

tmori, 2024-01-16:
色々やってみましたが、劇的にこれで安定化する感じではないですね。

kenjihiranabe, 2024-01-16:
もしかしたら、二次が正しいかもなので、int drag_order = 0,1,2 を引数として加えることを検討したい。

kenjihiranabe, 2024-01-21:
パラメータ追加しました．（デフォルトパラメータ drag2 = 0 として追加してあるので，既存コードは壊れません）
https://github.com/toppers/hakoniwa-px4sim/commit/474e25b121cf17ed718e4b8fc82c635730b94b04

closed 2024-01-21 by kenjihiranabe

tmori, 2024-01-21:
ありがとうございます！
パラメータの方はこちらで対応しておきますね。

kenjihiranabe, 2024-01-21:
探索空間がまた増えた．．．