toppers/hakoniwa-px4sim #133: PX4の拡張カルマンフィルタの調査

tmori, 2024-01-20:
# 目的

箱庭のドローンシミュレータでは、以下の課題がある。

* https://github.com/toppers/hakoniwa-px4sim/issues/114
* https://github.com/toppers/hakoniwa-px4sim/issues/90
* https://github.com/toppers/hakoniwa-px4sim/issues/113#issuecomment-1873540553

これらは、おそらく、PX4内の拡張カルマンフィルタでの推定値とのズレが原因になっていると思われる。
なので、PX4の推定がどのような考え方で設計されているかを把握する必要がある。

# 参考文献

PX4のEKF2（Extended Kalman Filter）に関しては、いくつかの重要な情報源があります。

1. **PX4 User Guide - Using the ECL EKF**: このガイドでは、EKF2の設定と使用方法について詳しく説明しています。これには、IMUや磁力計センサーの数、EKFのインスタンスの実行、様々なセンサー測定の処理方法などが含まれます。また、センサー測定値の遅延補償、IMUデータの使用方法、高度情報のソースとしてGPS、気圧計、レンジファインダーなどをどのように利用するかについても説明しています【6†source】。

2. **ECL/EKF Overview & Tuning - PX4 v1.9.0 User Guide**: こちらのガイドは、EKF2のアルゴリズムがどのようにセンサー測定値を処理して、様々な状態（位置、速度、IMUのバイアス推定値など）を推定するかについて詳しく述べています。EKFの異なる動作モード、必要なセンサーの組み合わせ、静圧位置誤差の補正方法に関しても説明されています【7†source】。

3. **PX4 Developer Guide - Estimators**: このドキュメントでは、PX4の様々な推定器、特にEKF2についての概要が提供されています。さらに、EKF2の使用方法や、他の推定器との比較に関する情報も含まれています【8†source】。

これらのソースは、PX4のEKF2に関する理解を深め、適切な設定やチューニングを行うための有益な情報を提供します。EKF2について詳細を知るには、これらのドキュメントを参照することをお勧めします。

kenjihiranabe, 2024-01-22:
カルマンフィルタ側のチューニングって，すごいたくさんパラメータあるんだね．
https://docs.px4.io/main/en/advanced_config/parameter_reference.html

tmori, 2024-01-22:
カルマンフィルタのパラメータセットと、箱庭側のパラメータセットを合わせる。