toppers/hakoniwa-px4sim #134: QGCでセンサのキャリブレーションできるようにしてはどうか？

tmori, 2024-01-20:
# 背景

たまにコンパスエラーが出るけれど、もしかするとキャリブレーションで解決できたりして？

QGCには、センサキャリブレーション機能がある。

[image]

実機の場合は、機体を手でもって傾けたりしてキャリブレーションするのだが、シミュレータだと難しい。

# 対応案

シミュレータから外部操作できるUIを作って、物理モデルに位置や姿勢を変更できるようにしてみてはどうか。

tmori, 2024-01-20:
こうみると、ピッチ角が９０度のケースがあるのか。。

tmori, 2024-01-21:
物理的に拘束した状態で回転させたい

tmori, 2024-01-21:
ジンバルの中心に機体を拘束して、外力与えて回してみたい

kenjihiranabe, 2024-01-21:
うーん、外力で積分は無理だと思う。実際にやるときは、モータートルクで回さないからなあ。

案は、マグネットを外界とインタラクションさせること。Unity に磁北線張って、角度だすとか？
この絵のキャリブレーションではどのセンサの値とってるのだろ？

tmori, 2024-01-21:
> この絵のキャリブレーションではどのセンサの値とってるのだろ？

ジャイロセンサですね。

kenjihiranabe, 2024-01-21:
ジャイロは外界との関係性確立だから、ドローン単体ではできないと思っている。
あと、位置も積分と、もう一つGPSはUnityから取る手もあるね。

kenjihiranabe, 2024-01-21:
単体やるなら、外力でなく、角度を強制的に変えて、その時のmag値をかえすんかな。

tmori, 2024-01-21:
この構成で機体を制御して、物理モデルから出てくる速度データなどをセンサにそのまま与えて、PX4に返す方法を考えてみましたが、機体が遠くに飛ばないように固定する条件が必要そうですね。。

[image]

kenjihiranabe, 2024-01-21:
いや、ジャイロとmag を勘違いしてた。ジャイロならやりようがある。
外界といったのは、magです。
トルクでやるなら、どこか一点固定して、みたいにコマ回すのだけど、別の力学が必要な予感。

tmori, 2024-01-21:
> トルクでやるなら、どこか一点固定して、みたいにコマ回すのだけど、別の力学が必要な予感。

おー。

kenjihiranabe, 2024-01-21:
トルクでなく、角速度を与えてやれば良さそう。

tmori, 2024-01-22:
キャリブレーション用の関数をbody_frame内部に入れ込む。のが一番近道。

kenjihiranabe, 2024-01-23:
追加情報：
ここに各センサーのキャリブレーションの仕方あり．
https://qiita.com/ezokitsune/items/170ac3cc8b81b382f10a
また，ここに PX4 <-> GrouondControl のキャリブレーションのオフィシャルドキュメントあり．
https://docs.px4.io/main/en/config/compass.html
一回，各キャリブレーションの意味を表にまとめてみます．

tmori, 2024-01-25:
# auto-tune のパラメータ保存先とパラメータ参照方法を調査する

調べ方：
* PX4のドキュメントを読む
* 多分、バイナリで管理されていると思われ、コマンドで取得するような気がしている。
* となると、どのようなパラメータがあるのかを次に調べることになって、
* そのパラメータを参照するコマンドを調べる流れ。

tmori, 2024-01-25:
この辺りが解説部分。

* https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter_basic.html
* https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html

おそらく、以下のパラメータをチューニングしていると思われる。

* Roll rate control ([MC_ROLLRATE_P](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_ROLLRATE_P), [MC_ROLLRATE_I](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_ROLLRATE_I), [MC_ROLLRATE_D](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_ROLLRATE_D), [MC_ROLLRATE_K](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_ROLLRATE_K))
* Pitch rate control ([MC_PITCHRATE_P](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_PITCHRATE_P), [MC_PITCHRATE_I](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_PITCHRATE_I), [MC_PITCHRATE_D](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_PITCHRATE_D), [MC_PITCHRATE_K](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_PITCHRATE_K))
* Yaw rate control ([MC_YAWRATE_P](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_YAWRATE_P), [MC_YAWRATE_I](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_YAWRATE_I), [MC_YAWRATE_D](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_YAWRATE_D), [MC_YAWRATE_K](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MC_YAWRATE_K))

tmori, 2024-01-25:
機体設定で auto tune できるのは、レートコントローラと姿勢コントローラのみ。

[image]

なので、それらの制御プログラムのパラメータが対象と考えるのが自然。

以下が調査結果。

# Multicopter Rate Control

https://docs.px4.io/main/en/advanced_config/parameter_reference.html#multicopter-rate-control

# Multicopter Attitude Control

https://docs.px4.io/main/en/advanced_config/parameter_reference.html#multicopter-attitude-control

tmori, 2024-01-25:
PIDのパラメータチューニングで解決できること

* 機体が目標値に近づかない（発散してしまう）
* 機体が目標値に近づくが振動し続ける

ということは以下の２つのissueは解決できなさそうな気がしてきた。

* https://github.com/toppers/hakoniwa-px4sim/issues/90
* [真後ろに飛行指示を出すと、機体が不安定になる](https://github.com/toppers/hakoniwa-px4sim/issues/113#issuecomment-1873540553)
 
カルマンフィルタの推定値がずれている問題を調査すれば、上記の issueの問題点がより浮き彫りになりそうな予感。

tmori, 2024-01-25:
あと、auto-tuningはセンサのキャリブレーションではなくて、PID制御パラメータのキャリブレーションのように思えてきたので、センサ単独のキャリブレーションもできた方が良さそう。

機体の姿勢を外部から変更するAPIはあるので、ちょっとやってみます。

tmori, 2024-01-25:
コンパス：

[image]

加速度センサ：
[image]

kenjihiranabe, 2024-01-25:
calibration ディレクトリ使ってもらっていいですよ。

tmori, 2024-01-25:
# 入力要件

数値角を少しずつ上限できる：
* ロール
* ピッチ
* ヨー

姿勢角を数値で指定できる：
* ロール
* ピッチ
* ヨー

# 実現方法

姿勢操作用のアプリを作って、キーボード入力を受け付けて、上記を実現する。

tmori, 2024-01-25:
設計できた！

せっかくなので、最近整理した箱庭アセットAPIを使って、第３の箱庭アセットを開発することにしました。

[image]

kenjihiranabe, 2024-01-25:
おー！