toppers/hakoniwa-px4sim #150: v2.1.0 のリリース内容の検討

tmori, 2024-02-12:
# リリース内容案

v2.1.0 はハード側を攻める

1. メーカー毎のセンサモデルを組み込み可能な箱庭側のプラットフォームの整備
2. 箱庭メーカーのセンサモデル（伝達関数の検討＆C実装&MATLAB連携あり）
3. 環境モデル
  3.1. 機体の位置に依存する環境要因の追加
4. 機体の物理特性モデル(伝達関数の検討＆C実装)
  4.1. 温度変動、機械振動

tmori, 2024-02-13:
# フィーチャーモデル案

[image]

tmori, 2024-02-14:
# 機体の物理特性モデル

機械振動については、非常に複雑な関数になると思われ、一旦は、ホワイトノイズとする案もある。

[image]

tmori, 2024-02-14:
# デモの見せ方

デモを実施する際には、視聴者が理解しやすいように、問題の本質を明確に示しつつ、影響がどのように現れるかを具体的に視覚化することが大切です。以下は、温度ドリフトと機械振動が機体の動きに及ぼす影響を示すためのデモの見せ方の例です。

### 温度ドリフトによる影響のデモ方法

1. **シナリオの設定**: 許容温度範囲内での機体の動作をまず示し、その後で温度が許容範囲を超えたときの動作を示す。

2. **ビジュアル化**: 温度が許容範囲を超えると、センサーの出力にオフセットやドリフトが生じることをグラフやアニメーションで視覚化する。

3. **リアルタイムのフィードバック**: 温度が増加するにつれて、機体の応答がどのように変わるかをリアルタイムで示す。

4. **結果の比較**: 許容範囲内外での機体の動きを並べて表示し、違いを強調する。

### 機械振動による影響のデモ方法

1. **フィルタ設定の説明**: ローパスフィルタのカットオフ周波数の異なる設定を用意し、それぞれの影響を比較する。

2. **ノイズの視覚化**: カットオフ周波数が低い場合と高い場合で、機体にどの程度ノイズが乗るかを波形や振動の形で示す。

3. **動きの不安定さの示し方**: 実際の機体の動きをシミュレートし、ノイズが多いほど動きが不安定になることを視聴者に示す。

4. **結果の比較**: フィルタがノイズをどの程度減少させるかを、フィルタなしと比較して示す。

デモでは、視聴者が直感的に理解できるように、テクニカルな詳細を過度に複雑にすることなく、影響を明確にすることが重要です。また、リアルタイムのインタラクティブなデモを用意することで、視聴者の関心を引き、理解を深めることができます。データやシミュレーションの結果を用いて、問題が実際にどのように発生し、それに対してどのような対策が可能かを説明すると良いでしょう。

kenjihiranabe, 2024-02-16:
センサーメーカーが喜びそうなシナリオとして，

1. 外乱なしで正常飛行をデモ（飛び始める）．
2. 外乱環境（例：太陽光が強い場所）を飛行して，熱を持ってきたら，異常状態になる（飛行中）→墜落
3. Aメーカーの，センサーを使うと，，，，うまく行く．

上記過程で，センサー値などをうまくリアルタイム表示して，なるほど，と思わせる．

tmori, 2024-02-16:
ありがとうございます！来週、それやります！

tmori, 2024-02-25:
# 作業リスト

- [ ] センサモデルインタフェース仕様定義
- [ ] センサモデル実装（まずはベタに実装する）
- [ ] センサモデルローダー作成
- [ ] UnityとのI/F仕様定義（外乱）
- [ ] 外乱スクリプト実装（まずはベタに実装する）
- [ ] 外乱オブジェクト作成（太陽光をどう表現するか？）
- [ ] 箱庭PDU定義
- [ ] 箱庭PDU読み込み処理の追加（箱庭側）
- [ ] 箱庭PDU書き込み処理の追加（Unity側）

tmori, 2024-02-25:
# 外乱

温度ドリフトとする

温度ドリフトによる影響が顕著なセンサー：

ドローンに搭載されているセンサーで温度ドリフトの影響を受けやすいものは以下の通りです：

1. **IMU (Inertial Measurement Unit)**: 加速度計とジャイロスコープを含むIMUは、温度変化によりセンサーのバイアスが変動しやすく、これがドローンの安定性や正確な位置決めに影響を与える可能性があります。

2. **バロメトリックプレッシャーセンサー (気圧センサー)**: 温度変化は気圧センサーの読み取りに影響し、結果として高度の測定に誤差が生じることがあります。

3. **GPS**: 温度変化は、GPSレシーバの時計オフセットに影響を与え、位置情報の精度を低下させる可能性があります。

4. **電池のパフォーマンス**: 温度はリポバッテリーの内部抵抗に影響を与え、電池の効率と飛行時間に影響を与えます。

これらのセンサーは、ドローンが正確に飛行し、任務を遂行するためには、正確なデータを提供する必要があるため、温度ドリフトは特に重要な問題となります。補償アルゴリズムやセンサーデータのキャリブレーションを行うことで、温度変化による影響を最小限に抑えることが可能です。

tmori, 2024-02-25:
# アーキテクチャ
[image]

tmori, 2024-02-25:
[image]

tmori, 2024-02-26:
機体の温度表現：

１。色を変える
２。テキストで表示（テキストの色を変える）

tmori, 2024-02-27:
# Unityクラス作成作業リスト

- [ ] パッケージ設計
- [ ] IRobotProperty
- [ ] RobotProperty
- [ ] IEnvironmentObject
- [ ] EnvTemperatureRegion
- [ ] EnvironemntEffector
- [ ] 簡単な動作チェック

tmori, 2024-02-27:
# グラフ化方法

以下のツールを利用してみる。
https://github.com/XCharts-Team/XCharts

* 機能面：リアルタイムにデータをグラフ化してくれる。複数機体の表記も可能に見える
* ライセンス面：MITライセンス
* 組み込み方法：最新リリースをダウンロードしてもらってUnityにインポートしてもらう

まずは、試し見てる。

tmori, 2024-03-11:
最低限のデモはできるようになったのでクローズします。

closed 2024-03-11 by tmori