toppers/hakoniwa-px4sim #151: ドローンを10機、同時にシミュレーションできるようにしてほしい

tmori, 2024-02-14:
# 狙い

* さまざまなドローンの機体評価を同時にやりたい
* ドローンの側制御のシミュレーションできるように

tmori, 2024-02-15:
# アーキテクチャ検討結果

機体を複数にする場合、PX4を何個も起動しないといけないのでオーバーヘッドが高くなりそう。
箱庭側のプロセスの数は同じでいける。

[image]

tmori, 2024-02-15:
hakoniwa-px4sim側の構造を複数機体対応できるようにすればやれそう。

kenjihiranabe, 2024-02-16:
PX4側複数はそうなる気がする．なぜかというと，コンフィグの設定が一機分しかないから，複数のスペックを扱えるようには見えない．群制御を将来やりたいとすると，現在のPID制御ドライバを改良して，複数インスタンス対応の軽い PX4 相当を作るのもありか？（フルスペックは到底無理として．多分このデモはそうやっている気がする）
https://youtu.be/y-C4AId2Za8?si=0uWI2eMBA6-0FnP0&t=2401

kenjihiranabe, 2024-02-16:
この動画作った先生から，返信ありました！
https://x.com/marurur/status/1280491500900507648?s=20

tmori, 2024-02-16:
> PX4側複数はそうなる気がする．なぜかというと，コンフィグの設定が一機分しかないから，複数のスペックを扱えるようには見えない．群制御を将来やりたいとすると，現在のPID制御ドライバを改良して，複数インスタンス対応の軽い PX4 相当を作るのもありか？（フルスペックは到底無理として．多分このデモはそうやっている気がする） https://youtu.be/y-C4AId2Za8?si=0uWI2eMBA6-0FnP0&t=2401

ですよね。１００台とかの群制御を考える場合は、PX4は現実的ではないと思われます。
現状、箱庭内部にPID制御を実行するための機能があるので、そっちを使うほうが良いような気がしています。PX4無しでも箱庭は動くところが良いですね。

tmori, 2024-02-16:
ひとまず、２台動いたので、マージします。

commit 02c7c628 (2024-02-16): fix #151
closed 2024-02-16 by tmori