toppers/hakoniwa-px4sim #164: PID制御の勉強ができる環境を作る

tmori, 2024-02-17:
# 目的

箱庭ドローンシミュレータを制御の勉強として利用できるようにしたい。
そして、参考資料にあるように、制御の競技会を同時実施できるようにする！

参考：
https://www.youtube.com/watch?v=y-C4AId2Za8&t=1840s

tmori, 2024-02-17:
# 構成

[image]

tmori, 2024-02-17:
# 課題など

* PID制御プログラムをモジュールとして組み込めるプラットフォームの検討が必要
* PID制御プログラムは、箱庭とは別の環境でビルドできる必要がある
* なんなら、MATLABで作ってコード生成してもらったものを組み込めるようにしたい

tmori, 2024-02-17:
# PID制御プログラムのモジュール化方法について

* 案1. ダイナミックローディングできるようにする
* 案2. スタティックリンクする？

案1でしょ。

tmori, 2024-02-17:
# 詳細設計

[image]

tmori, 2024-02-17:
# 実装ステップ

1. drone と controllerを接続して実行できるようにする
2. Unityと連携できるようにする
3. ローダーを作る
4. 新しいリポジトリでcontrollerを作成できるようにする

tmori, 2024-02-18:
# 残作業

- [ ] コンフィグファイルの設定がないケースのエラーハンドリング
- [ ] ローダブルモジュールがない場合のエラーハンドリング
- [ ] その他あれば

tmori, 2024-02-19:
# 最後の作業

この環境を利用して、現行の物理モデルのホバリングを理論的に検討したPID制御モデルでコントロールできるようにして、箱庭ラボのブログで紹介する。

- [ ] 平鍋さんの２次伝達関数の理解
- [ ] 伝達関数から理論的にPIDの値を求めるやり方がないか調べる
- [ ] プログラム実装して複数のパラメータで同時に検証してみる。
- [ ] サンプルプログラムを自動作成するツールを作る。
- [ ] 一般公開する
- [ ] 箱庭ラボのブログで紹介する

tmori, 2024-02-19:
# 平鍋さんの２次伝達関数の理解

対象の運動方程式：

$$
\ddot{z} = -u(t)/m + g - d\dot{z}/m
$$

ここで、目標値との誤差の方程式にすると以下となる。

$$
-\ddot{e} = -u(t)/m + g +d\dot{e}/m
$$

$$
e(t) = R - z(t)
$$

$R$ はホバリングする目標値

これをラプラス変換するとこうなって、

$$
-s^{2} E(s) = -U(s)/m + g E(s) + d s E(s) / m
$$

伝達関数にするとこうなる。

$$
Gp(s) = E(s)/U(s) = (1/m) / (s^{2} + (d/m) s + g)
$$

tmori, 2024-02-19:
# 伝達関数から理論的にPIDの値を求めるやり方がないか調べる

これではないか？

https://tajimarobotics.com/pid-stability-pid-control/

kenjihiranabe, 2024-02-19:
>
> これをラプラス変換するとこうなって、
>
> $$ -s^{2} E(s) = -U(s)/m + g E(s) + d s E(s) / m $$
>

gE(s)は、g/s かな。そして、純粋に線形にならないので、E/U が解けない。

>

tmori, 2024-02-19:
なるほど。２次遅れの伝達関数にして、理論的な方向で攻めようかと思いましたが、難しそうですね。

kenjihiranabe, 2024-02-19:
僕の方策は、T/m - g をT' としてT' をコントロール可能な変数と捉えて線形化しました。(スライドの計算メモのページ)

tmori, 2024-02-19:
[image]

なるほど！

tmori, 2024-02-20:
やっと理解できました。P制御と繋げた微分方程式作って、２次の伝達関数にしているのか！

https://speakerdeck.com/hiranabe/math-physics-and-dynamics-of-drone-in-hakoniwa?slide=13

[image]

tmori, 2024-02-20:
まずは、P制御をベースにして、考えていきます。

tmori, 2024-02-20:
# 再検討開始

対象の運動方程式：

$$
\ddot{z} = -\frac{u(t)}{m} + g - d\frac{\dot{z}}{m}
$$

ここで、目標値との誤差の方程式にすると以下となる。

$$
-\ddot{e} = -\frac{u(t)}{m} + g +d\frac{\dot{e}}{m}
$$

$$
e(t) = R - z(t)
$$

$R$ はホバリングする目標値(固定値)

さらに、入力関数を以下とする。

$$
u_g(t) = g -\frac{u(t)}{m}
$$

こうなる

$$
-\ddot{e} = u_g(t) +d\frac{\dot{e}}{m}
$$

これをラプラス変換するとこうなる。

$$
-s^{2}E(s) = U_g(s) +d\frac{sE(s)}{m}
$$

tmori, 2024-02-20:
# P制御側

u(t) = K_p e(t)

これをラプラス変換すると、

U(s) = K_p E(s)

tmori, 2024-02-20:
何やら複雑な感じになったな。。

[image]

tmori, 2024-02-20:
# P制御とプラントモデルを含めた伝達関数の解釈

1/s は単位ステップ関数 $u_s(t)$ のラプラス変換 $U(s)$ なので、単位ステップ信号に対する伝達関数として表現すると、こうなる。

$$
D(s) = \frac{Z(s)}{U(s)} = \frac{(g -\omega_n^{2} R)}{\omega_n^{2}} \frac{\omega_n^{2}}{s^{2} + 2 \zeta \omega_n s + \omega_n^{2}}
$$

tmori, 2024-02-20:
# ２次遅れ要素の特徴

$\omega_n^{2} = \frac{K_p}{m}$

$\zeta = \frac{d}{2\sqrt{mK_p}}$

- [ ] 過制動($\zeta > 1$)
- [ ] 臨界制動($\zeta = 1$)
- [ ] 不足制動($0 < \zeta < 1$)
- [ ] 持続振動($\zeta = 0$)

tmori, 2024-02-20:
# 過制動

条件：$K_p < \frac{d^{2}}{4m}$

tmori, 2024-02-20:
# 臨界制動

条件：
$K_p = \frac{d^{2}}{4m}$

tmori, 2024-02-20:
# 不足制動
条件：
$K_p > \frac{d^{2}}{4m}$

tmori, 2024-02-20:
# P制御側

重力を定常的に与えないと、相当不安定になるのではないか。

$u(t) = K_p ( R - z(t) )+ g$

これをラプラス変換すると、

$U(s) = K_p ( \frac{R}{s} - Z(s) ) + \frac{g}{s}$

tmori, 2024-02-20:
# 再考：プラント側のU(s) にP制御の式をラプラス関数で代入

$s^{2}Z(s) = -\frac{K_p ( \frac{R}{s} - Z(s)) - \frac{mg}{s}  }{m} + \frac{g}{s} - \frac{d}{m} s Z(s)$

左辺に微分項、右辺を積分項にしてみる。

$s^{2}Z(s) + \frac{d}{m} s Z(s) + \frac{K_p}{m} Z(s) = ( -\frac{K_p R}{m}) \frac{1}{s}$

ここからZ(s)を求めるとこうなる。

$Z(s) = \frac{( -\frac{K_p R}{m})}{s(s^{2} + \frac{d}{m} s + \frac{K_p}{m})}$

なので、積分要素と２次の遅れ要素の直列接続とみなせる

$Z(s) = \frac{1}{s} \frac{( -\frac{K_p R}{m})}{s^{2} + \frac{d}{m} s + \frac{K_p}{m}}$

変形すると、

$Z(s) = \frac{m}{K_p}(g -\frac{K_p R}{m}) \frac{1}{s} \frac{\frac{K_p}{m}}{s^{2} + \frac{d}{m} s + \frac{K_p}{m}}$

ここで、

$\omega_n^{2} = \frac{K_p}{m}$

$\zeta = \frac{d}{2\sqrt{mK_p}}$

とおくと、こうなる。

$Z(s) = \frac{( -\omega_n^{2} R)}{\omega_n^{2}} \frac{1}{s} \frac{\omega_n^{2}}{s^{2} + 2 \zeta \omega_n s + \omega_n^{2}}$

kenjihiranabe, 2024-02-20:
この3次式まで僕の手計算と整合しています。
この後逆ラプラスして時間領域で式をだして、グラフ書きました。
[image]

tmori, 2024-02-20:
ありがとうございます！
$\omega_n$ と $\zeta$ の値が違うようですが、同じ感じですね。

ここから、$\zeta$ の範囲から $K_p$ の代表値を決めて、実際にシミュレーションしてみようと思っています。

|特徴|条件|Kp値|
|---|---|---|
|過制動|$K_p < \frac{d^{2}}{4m}$|TODO|
|臨界制動|$K_p = \frac{d^{2}}{4m}$|TODO|
|不足制動|$K_p > \frac{d^{2}}{4m}$|TODO|

kenjihiranabe, 2024-02-20:
ここからは、方針(オーバーシュートx%までとか、整定時間何秒にしたいとか、外乱に強くしたいとか)で決めていくのだと思う。

tmori, 2024-02-20:
# 試してみた

$K_p$ = (1/4.0)*1.0e-05
$\zeta$=0.1

[image]

tmori, 2024-02-20:
# 試してみた(その２)
目標値は -10m
$K_p$ = (1/4.0)*1.0e-04
$\zeta$=0.0316227766

[image]

tmori, 2024-02-20:
# 試してみた(その３)
目標値は -10m
$K_p$ = (1/4.0)*1.0e-00
$\zeta$=0.000316227766
[image]

tmori, 2024-02-20:
# 所感

* $\zeta$ が０に近づくに連れて、持続振動になることが再現された。
* $\zeta$ が１に近づくと、不足振動になり、振動数およびオーバーシュート量が減るが、収束時間は非常に長い。約３時間かかる。

tmori, 2024-02-20:
# PD 制御

z軸はNED座標系。Rは正の値で指定する。

$u(t) = K_p ( z(t) + R  ) + m g + K_d ( \dot{z(t) + R} ) $

これをラプラス変換すると、

$U(s) = K_p ( Z(s) + \frac{R}{s} ) + \frac{m g}{s} + K_d s Z(s) $

tmori, 2024-02-20:
# プラント側のU(s) にPD制御の式をラプラス関数で代入

$s^{2}Z(s) = -\frac{ K_p ( Z(s) + \frac{R}{s}) + \frac{mg}{s} + K_d s Z(s) }{m} + \frac{g}{s} - \frac{d}{m} s Z(s)$

左辺に微分項、右辺を積分項にしてみる。

$s^{2}Z(s) + \frac{d}{m} s Z(s) +  K_d s Z(s) + \frac{K_p}{m} Z(s) = ( -\frac{K_p R}{m}) \frac{1}{s}$

ここからZ(s)を求めるとこうなる。

$Z(s) = \frac{( -\frac{K_p R}{m})}{s(s^{2} + (\frac{d}{m} + K_d ) s + \frac{K_p}{m})}$

なので、積分要素と２次の遅れ要素の直列接続とみなせる

$Z(s) = \frac{1}{s} \frac{( -\frac{K_p R}{m})}{s^{2} + (\frac{d}{m} + K_d) s + \frac{K_p}{m}}$

変形すると、

$Z(s) = \frac{m}{K_p}( -\frac{K_p R}{m}) \frac{1}{s} \frac{\frac{K_p}{m}}{s^{2} + \frac{d}{m} s + \frac{K_p}{m}}$

ここで、

$\omega_n^{2} = \frac{K_p}{m}$

$\zeta = \frac{d + m K_d}{2\sqrt{mK_p}}$

とおくと、こうなる。

$Z(s) = - R \frac{1}{s} \frac{\omega_n^{2}}{s^{2} + 2 \zeta \omega_n s + \omega_n^{2}}$

tmori, 2024-02-20:
# PDパラメータ探索の方針

$\zeta$ が 0.707 が最適な値だと仮定して、 $K_p$ と $K_d$ のバリエーションを探索してみるとどうだろうか。

tmori, 2024-02-20:
# ζ=0.707 のPD制御結果

Kp = 0.25
Kd = 2.234730306

１秒くらいで目標値に収束しました！

[drone_dynamics.csv](https://github.com/toppers/hakoniwa-px4sim/files/14351701/drone_dynamics.csv)

[image]

tmori, 2024-02-20:
# ζ=0.707 のPD制御結果（その２）

Kp = 0.0000025
Kd = 0.00607

こちらはKpが小さいので収束に時間がかかっていますし、収束値は10mになっていません。

[drone_dynamics.csv](https://github.com/toppers/hakoniwa-px4sim/files/14351745/drone_dynamics.csv)

[image]

tmori, 2024-02-20:
> 現在、10m でホバリングしていない事に注意。グラフのvalue は、R-g/wn^2 (→10)の値であり。P制御だけでは定常位置偏差 g/wn^2 がずっと出てしまう。→I制御が必要、という流れ。

コメントありがとうございます！
Kiのほうも同様に数式化できないかやってみます。

あと、オーバーシュート値とオーバーシュートに達する時間も解析的に求められると評価がしやすいですよね。

tmori, 2024-02-20:
# プラント側のU(s) にPID制御の式をラプラス関数で代入

$s^{2}Z(s) = -\frac{ K_p ( Z(s) + \frac{R}{s}) + \frac{mg}{s} + K_d s Z(s) + K_i (  \frac{R}{s^{2}} +  \frac{Z(s)}{s} )}{m} + \frac{g}{s} - \frac{d}{m} s Z(s)$

左辺に $Z(s)$ 項、右辺を $s$ 項にしてみる。

$s^{2}Z(s) + \frac{d}{m} s Z(s) +  K_d s Z(s) + \frac{K_p}{m} Z(s) + K_i \frac{Z(s)}{s} = ( -\frac{K_p R}{m}) \frac{1}{s}  -   \frac{K_i R}{s^{2}}$

tmori, 2024-02-20:
Kiが入ると、２次遅れにならないので、これまでの解き方では難しそう。

tmori, 2024-02-20:
# 現時点の課題まとめと今後の方針

1. PD制御の計算結果が怪しいかもしれない
2. I制御を含めた解析は難しそう

I制御の解析は一旦ペンディングします。
PD制御の計算結果は再度レビューします。

その上で、PD制御のパラメータをいくつか決めて、I制御パラメータを微調整する探索をやって、Unityでビジュアライズするところをゴールにしようと思います。

kenjihiranabe, 2024-02-21:
森さんのやり方だと，一巡伝達関数でなく，閉ループのステップ応答を直接求めるようになっていて，制御の通常コースの，

1. P(s) を与えて，C(S) を設計する課題．
2.そのために，「一巡伝達関数＝C(s)P(s)」が分かると閉ループは「W=CP/(1+CP)」である．
3. W の条件（外乱につよい，すばやく収束する，などなど）がC(S)設計への要求となる．
4. C(S)設計のために，「WでなくCPの特性を見て，要求を満たすCを設計」（CPのボード線図，ナイキスト線図）

という手順が使えなくて，教材としては制御の教科書にうまく乗らないのがちょっと残念に思いました．何かうまくできるといいのだけど．