toppers/hakoniwa-px4sim #169: 外部から機体をコントロールできる制御機能を作る

tmori, 2024-02-29:
# 目的

* AirSimのようにPythonからAPIコールすることで、箱庭内の機体を遠隔操作できるようにする
* サービス向け箱庭環境の土台にする

# 方法

* 最低限の制御とアーキテクチャを検討する（あまり複雑にしない）
* 必要な機能
  * takeoff機能
  * landing機能
  * 目的地移動機能
    * ウェイポイント計画管理機能
    * 位置制御機能
      * X軸のPID制御機能
      * Y軸のPID制御機能
      * Z軸のPID制御機能
    * 姿勢角制御
      * ロール方向のPID制御機能
      * ピッチ方向のPID制御機能
      * ヨー方向のPID制御機能

tmori, 2024-02-29:
# 設計案

制御方法：
* 目的地から現在地までの方向に対して機首を合わせて前進する
* 位置制御では、高さと水平方向の距離のみを制御する
* 姿勢角制御では、基本的には、以下の方針とする。
  * ロール角は常に０にする（水平）
  * ヨー角は、現在地点から目標地点に向かう方向を維持する
  * ピッチ角は、水平方向の距離に比例して上下させる

[image]

tmori, 2024-02-29:
全体的な見直し結果。
[image]

tmori, 2024-03-01:
# 作業リスト

- [ ] 制御状態設計が必要
- [ ] 機体制御インタフェースの修正（速度系の追加が必要）
- [ ] 離陸/着陸モードの処理追加：垂直方向位置制御
- [ ] 離陸/着陸モードの処理追加：垂直方向速度制御
- [ ] 離陸/着陸モードの処理の動作チェック
- [ ] 角速度制御（ヨー）の追加
- [ ] 移動モードの機首回転処理の追加
- [ ] 移動モードの機首回転処理の動作チェック
- [ ] 角速度制御（ロール）の追加
- [ ] 角速度制御（ピッチ）の追加
- [ ] 移動モードの水平移動処理の追加
- [ ] 移動モードの水平移動処理の動作チェック
- [ ] 座標変換ライブラリの追加

kenjihiranabe, 2024-03-01:
PX4側を見ていたら，ある場所に行く，というミッションが来た時に，状態遷移マシンを使って戦略を組み立てているようですね．（詳細はみていない）

tmori, 2024-03-18:
考慮が足りてないような気がしているので再オープン。

closed 2024-04-30 by tmori