toppers/hakoniwa-px4sim #177: Mac向けドキュメントの細かい足りないところ

hisazumi, 2024-03-14:
Macで以下のコンポーネントを追加でインストールしました。どこかスクリプトで自動でインストールしてもらうか、ドキュメントに追加してもらえると良いと思います。

* googletest
* cmake
* jq

また、python scriptで実行する場合には bash drone_control/build.bash の実行が足りてない感じがします（単純に読み込み漏れだったらすみません）。

tmori, 2024-03-14:
> Macで以下のコンポーネントを追加でインストールしました。どこかスクリプトで自動でインストールしてもらうか、ドキュメントに追加してもらえると良いと思います。
> 
> * googletest
> * cmake
> * jq
> 
> また、python scriptで実行する場合には bash drone_control/build.bash の実行が足りてない感じがします（単純に読み込み漏れだったらすみません）。

ご連絡ありがとうございます。
README追加しますねー。あと、インストール・チェックスクリプトも一緒に作ろうと思いました。

tmori, 2024-03-15:
@hisazumi さん

修正してみましたので、ご確認をお願いできますか。

* https://github.com/toppers/hakoniwa-px4sim/commit/d6f37f2c183ce96116a1a997afa3a9f17a987714
* https://github.com/toppers/hakoniwa-px4sim/commit/c8db6baeed436cc832082e498a3f3390c977078f

closed 2024-03-28 by tmori