toppers/hakoniwa-px4sim #178: 現状のPython APIで駆動するフライト制御をもっとキビキビ動くようにしたい

tmori, 2024-03-17:
# 課題
水平移動が極めて遅い。目的地に到着するのに１００秒程度かかってしまう。少なくとも１０倍程度にしたい。

# 目指したいこと
水平移動時のプラントモデルの伝達関数をある程度同定した形で、過渡特性のパラメータをベースにして、理論的にパラメータを求めたい。
# 参考情報

* [制御工学同演習第十回　安定性２，制御系の過渡特性１](https://www.youtube.com/watch?v=NsxcXjliLdU)
* [制御工学同演習第十一回　制御系の過渡特性２，定常特性１](https://www.youtube.com/watch?v=KYVbZQfgQQQ&t=12s)

tmori, 2024-03-17:
# 計測結果

## 高さの推移

[image]

## テイクオフ：
takeoff指示から、１秒程度で目標の高さでホバリングしているので、ここは問題ない。

## 着地：
着地は、１０秒くらいかかっている。若干遅い。
ただ、水平移動に比べたら、そんなに気にならないので、一旦、放置する。

tmori, 2024-03-17:
# 計測結果

## 水平移動の推移
[image]

* 目標地点(X=5、Y＝０)に到達するのに約70秒かかっている。
* オーバーシュートが1mを超えている
* 立ち上がりに10秒かかっている
* オーバーシュートしてから位置補正に60秒もかかっている。。

tmori, 2024-03-17:
# プラントモデル

$$
\begin{array}{llll}
\dot{u_e} = &-\frac{T}{m}(\cos{\phi}\sin{\theta}\cos{\psi} + \sin{\psi}\sin{\phi}) & &- \frac{d}{m}u_e \\
\dot{v_e} = &-\frac{T}{m}(\cos{\phi}\sin{\theta}\sin{\phi} - \sin{\phi}\cos{\psi}) & &-\frac{d}{m}v_e \\
\dot{w_e} = &-\frac{T}{m}(\sin{\phi}\cos{\theta})                  &+g              &-\frac{d}{m}w_e 
\end{array}
$$

tmori, 2024-03-17:
ひとまず、前進の水平移動のみを想定したいので、 $\dot{u_e}$ のみに着目。
$\phi$ = 0°、 $\theta$ = $\theta_m$ ° 、 $\psi$ = 0°
とおく。

tmori, 2024-03-17:
$\dot{u_e} = -\frac{T}{m}\sin{\theta} - \frac{d}{m}u_e$

tmori, 2024-03-17:
# P(s)：ラプラス変換したもの

$s^{2}X(s) = -\frac{U(s)}{m}  \frac{\omega}{s^{2}+\omega^{2}} - \frac{d}{m}sX(s)$

tmori, 2024-03-18:
$U(s)$ は、推力だけど、現状、垂直方向のPID制御でしかコントロールできてない。
水平移動時にも推力をコントロールできるようにしなければならないのではないか。

https://github.com/toppers/hakoniwa-px4sim/issues/169

tmori, 2024-03-27:
* θ=wtではなくて、θのままにしないといけない。
* θについては、オイラー角の変化率から決める。でも、難しそう。（p, q, rの運動方程式から考える必要あり）

kenjihiranabe, 2024-03-28:
ここに、すごいいい記事みつけました。PX4の位置制御の方法です。
https://qiita.com/daisuke-iwakura/items/a9257b3f0f75659c3d09

kenjihiranabe, 2024-03-28:
今の森さん側の作りを変えずに、ざっくり要点を言うと、
- 位置目標だけでなく、速度目標を持ち、速度をフィードバック制御。
- 推力は、上下移動したくないので、上向き成分を重力との釣り合い値T0とすると、T0 = T*cos(θ) となるように Tとθ を制御。残りの成分 T*sin(θ) を水平移動に使う。

tmori, 2024-04-15:
入力のインタフェースをラジコンにする方法もあり。

tmori, 2024-04-20:
アーキテクチャ：
[image]

tmori, 2024-04-21:
[image]

tmori, 2024-04-21:
[image]

tmori, 2024-04-21:
[image]

tmori, 2024-04-21:
作業タスク出し：

- [ ] 機体プロキシのインタフェース設計
- [ ] 機体プロキシの作成
- [ ] 外部制御：姿勢制御作成
- [ ] 外部制御：角速度制御作成
- [ ] 外部制御：スロットルゲイン作成
- [ ] 外部制御：Mixer作成
- [ ] Python側：PS4コントローラ機能作成
- [ ] Python側：ラジコン目標値設定機能作成
- [ ] 結合テスト（Mixerなし）
- [ ] 結合テスト（Mixerあり）

tmori, 2024-04-21:
[image]

この辺りのクラス設計を見直したい。

tmori, 2024-04-21:
# リファクタリングの方向性、必要性を考える

## まずは現行のクラスの役割を一言で言えるか

* DroneControlProxyManager
  * 以下を一括して実行する
    * 箱庭のシミュレーション制御
    * Python側からの要求に応える
    * ドローンのシミュレーションを実行
* DroneControlProxy
  * Python側からの要求に応える
* AircraftSystemTaskManager
  * 箱庭のシミュレーション制御
* AirCraftManager
  * 複数機体の管理
* AircraftSystemContainer
  * １個の機体の管理

tmori, 2024-04-24:
# Mixerの仕様検討

* 野波先生の教科書の 式 2.63 をベースにして、Ωを計算して、そこからDを求める
* ただし、Jrの項の計算は厄介なので、省略する

## 基本式

* U = (T, τφ, τθ, τψ)
* Ω = (Ω1^2, Ω2^2, Ω3^2, Ω4^2)
* Ωi = Kr Di
* Ti = A Ωi
* T = ΣTi
* τψi = BΩi^2
* τψ = Στψi
* (τφ, τθ, 0) = Σ (rxi, ryi, 0) x (0, 0, AΩi^2) = Σ (ryi AΩi^2, rx AΩi^2, 0)

U = A M D

tmori, 2024-04-24:
# 行列での表現

[image]

[image]

[image]

U = A M Ω

tmori, 2024-04-24:
# PWMの計算方法

Ω = A^-1 M^-1 U
から、Ωi^2 を求める。
Ωi = Kr Di　なので、

Di = √Ωi/Kr

tmori, 2024-04-24:
# 上位制御部の設計

３段階に分けて実装する。

1. 高度制御、速度制御を実装して、ラジコン操作で期待した位置、速度で移動することを確認する
2. 位置制御、方位制御を実装して、Python API で、期待した位置へ移動することを確認する
3. ウェイポイントシーケンサを実装して、Python API で、期待した位置へ移動することを確認する
[image]

tmori, 2024-04-30:
Mixerはまだ未完成ではありますが、今後はこちらで管理します。

https://github.com/toppers/hakoniwa-px4sim/issues/205

kenjihiranabe, 2024-09-04:
https://github.com/toppers/hakoniwa-px4sim/issues/178#issuecomment-2073704541

> # PWMの計算方法
> Ω = A^-1 M^-1 U から、Ωi^2 を求める。 Ωi^2の結果からΩiを求める。±√となるが、PWM値は0-1なので、プラスのみ使用する。
> 
> ここで、Ωi = Kr Di　なので、
> 
> Di = Ωi/Kr
> 
> となる

上記は、ラプラス変換での式、Di(s) = (Ktr/(1+s Tr)) * Ωi なので、線形化が必要。
再考中。