toppers/hakoniwa-px4sim #230: 風対応

kenjihiranabe, 2024-06-07:
風には2種類あり、

- 一定の速度の風
- 突風（加速度をもつもの）
 
である。突風に関しては、力として働き、drone の面積なども関係してくるので、まずは、一定速度のものの対応とする。
drone physics が積分器を持たないので、単純に ground_frame での速度を求めた後に、風の速度を足す。
すなわち、現在の速度は、対空速度、とみなし、それに空が動く速度を足して、対地速度とする。

実速度＝対空速度＋空気速度（風）

https://github.com/toppers/hakoniwa-px4sim/commit/e3eaa343270cb7ac615c93cbd7623e038c494a4f

tmori, 2024-10-06:
# アーキテクチャ

## 方針

* Unity のカスタマイズにとらわれない構造を目指す
* 空間情報との親和性を考慮
* 現行のアーキテクチャはそのまま利用できるようにする

## 設計案
現行の箱庭ドローンシミュレータのアーキを変更せずに、箱庭環境アセットを追加すればいけそう。将来的には空間情報との連携も目指します。
[image]

tmori, 2024-10-06:
# 修正範囲

* まずは、箱庭内部での外乱を目指す。PX4連携は要望あれば対応する。
* ひとまず、外乱は、風だけ対応する。

# 作業リスト

- [ ] 外乱データのアップデート(hakonwia-ros2pdu)
- [ ] 外乱データのアクセサ追加
- [ ] 外乱の物理モデル反映
- [ ] 空間情報のデータ構造定義
- [ ] 空間情報のインスタンス作成
- [ ] 箱庭環境アセットの追加（まずはPythonで）
- [ ] 動作確認（まずはラジコン操作で）

tmori, 2024-10-06:
# 空間情報のデータ構造定義

- **空間定義**は`space_areas.json`にて管理し、AABBで範囲を定義します。
- **プロパティ定義**は`area_properties.json`に分けて、空間ごとに風や温度といった属性を`area_id`で関連付けます。

1. **`space_areas.json`**:
   - 各空間の定義を行います。`area_id`で各空間に一意の識別子を与え、`bounds`でAABB（最小点、最大点）による空間の範囲を定義します。

2. **`area_properties.json`**:
   - 空間に対するプロパティ（例：風、温度）を定義します。各プロパティは`area_id`によって`space_areas.json`の空間にリンクします。これにより、空間ごとに異なるプロパティを簡単に設定できます。

### 1. 空間定義（`space_areas.json`）

```json
{
  "space_areas": [
    {
      "area_id": "area_1",
      "bounds": {
        "min": { "x": 0, "y": 0, "z": 0 },   // 最小点
        "max": { "x": 100, "y": 100, "z": 50 } // 最大点
      }
    },
    {
      "area_id": "area_2",
      "bounds": {
        "min": { "x": 101, "y": 0, "z": 0 },
        "max": { "x": 200, "y": 100, "z": 50 }
      }
    }
  ]
}
```

### 2. プロパティ定義（`area_properties.json`）

```json
{
  "area_properties": [
    {
      "area_id": "area_1",
      "properties": {
        "wind": { 
          "velocity": { "x": 5.0, "y": 2.0, "z": -1.0 }  // 風速 (m/s)
        },
        "temperature": 20.0  // 温度 (摂氏)
      }
    },
    {
      "area_id": "area_2",
      "properties": {
        "wind": { 
          "velocity": { "x": -3.0, "y": 1.0, "z": 0.0 }
        },
        "temperature": 25.0
      }
    }
  ]
}
```

tmori, 2024-10-06:
# 空間情報を扱うためのクラス設計

考え方としては、本物の空間情報定義を組み込み可能な構造にすることです。
そのため、abstractクラスで吸収したいと思っています。ひとまずは、箱庭仕様ベースで実装を進めますが、ゆくゆくは差し替えできるようにしたいという意図です。

[image]

commit 2b871ed4 (2024-10-07): Merge pull request #357 from toppers/develop

fix #230
closed 2024-10-07 by tmori