toppers/hakoniwa-px4sim #250: 空気摩擦係数の3次元対応

tmori, 2024-07-05:
# 起点となる修正

https://github.com/toppers/hakoniwa-px4sim/commit/7726c9b2a11962c6fed8e1ed6076b6d2cf1a1862

tmori, 2024-07-05:
# 修正対応方針

1. すでに１次元でのコンフィグ＆処理実装がされているので、既存のものはそのまま動かせるようにする
2. 3次元用のインタフェースを用意することで対応する

tmori, 2024-07-05:
# 修正範囲

1. コンフィグ周り
  * https://github.com/toppers/hakoniwa-px4sim/blob/main/hakoniwa/src/config/drone_config.hpp#L161 
2. コンフィグ設定まわり
  * https://github.com/toppers/hakoniwa-px4sim/blob/7726c9b2a11962c6fed8e1ed6076b6d2cf1a1862/hakoniwa/src/assets/drone/aircraft/aircraft_factory.cpp#L88-L89
3. 物理ダイナミクスのインタフェース
  * https://github.com/toppers/hakoniwa-px4sim/blob/7726c9b2a11962c6fed8e1ed6076b6d2cf1a1862/hakoniwa/src/assets/drone/include/idrone_dynamics.hpp#L50-L51
4. 物理ダイナミクスの処理呼び出し
  * ３次元の場合は、そちらを優先する

kenjihiranabe, 2024-07-05:
matlab i/f も影響あるので、そちらは後回しにしています。

tmori, 2024-07-05:
# 内部処理実装方針

* 内部データとしては、3次元を前提としたデータで管理する。
* １次元の設定の場合は、すべて同じ値を設定することで対応する。
* このようにすることで、内部処理での分岐がなくなる。

kenjihiranabe, 2024-07-05:
いいと思います！

tmori, 2024-08-06:
あー、これ忘れてました。
PID制御問題の対応時に合わせて修正しようと思います。