toppers/hakoniwa-px4sim #253: 東尋坊でドローン飛ばす時は高速で飛ばしたい。

tmori, 2024-07-12:
東尋坊でドローンを飛ばそうとすると、現状のロール/ピッチ角をの上限20度にすると、移動の遅さが気になる。

なので、角度上限を50度にして動かしたところ、高速移動できるようになったけど、停止させようとする時の反動か何かで、数値計算結果がNANになってUnityが落ちる現象が発生。

上限を40度にしたら、問題は発生しなくなった。

kenjihiranabe, 2024-07-12:
ぼくもデモ動画を見ていて、もっとギューンと動くといいな、と思った。（録画のタイムラプスでもいいかも）
数値計算のNaNですが、ジンバルロック（quaternionにしていない）の話ではなさそうですね？（Unitiy側なので）
一度現象を見てみたいです。

tmori, 2024-07-17:
# 再現結果

## Unity側のログ

```
transform.position assign attempt for 'DroneTransporter' is not valid. Input position is { NaN, NaN, NaN }.
```

## drone dynamics
[drone_dynamics.csv](https://github.com/user-attachments/files/16258680/drone_dynamics.csv)

tmori, 2024-07-17:
値が発散していますね。

[image]

tmori, 2024-07-17:
線形モデルのプラント使って発生した場合は、クォータニオンの問題ではなさそうですね。

https://github.com/toppers/hakoniwa-px4sim/blob/main/matlab-if/drone_acceleration_by_linear_at_hover.c

tmori, 2024-07-17:
あ、いや違う。

変換側が怪しいのか？

https://github.com/toppers/hakoniwa-px4sim/blob/511858e3be99be7c289a97c5fe9b298a3d5fcefc/hakoniwa/src/assets/drone/physics/body_frame/drone_dynamics_body_frame.hpp#L55

https://github.com/toppers/hakoniwa-px4sim/blob/511858e3be99be7c289a97c5fe9b298a3d5fcefc/hakoniwa/src/assets/drone/physics/body_frame/drone_dynamics_body_frame.hpp#L60

tmori, 2024-07-17:
* ground_vector_from_body
  * 割り算なし、tan未使用
* euler_rate_from_body_angular_velocity
  * [割り算あり](https://github.com/toppers/hakoniwa-px4sim/blob/511858e3be99be7c289a97c5fe9b298a3d5fcefc/drone_physics/body_physics.cpp#L175)、[tan使用](https://github.com/toppers/hakoniwa-px4sim/blob/main/drone_physics/body_physics.cpp#L173)

kenjihiranabe, 2024-07-17:
θ=90付近では0割と同じになります。
sec が発散(1/cos)し、tan も発散。
θを追跡して、どこで90に近くなっているかみたいな。

[image]

tmori, 2024-07-17:
再現してグラフ化してみました。

Ryがθですね。90度はπ/2 なので、約1.57 ですね。
グラフから、θが90度を横切っていることを読み取れますね。

[image]

tmori, 2024-07-17:
ただし、値がビンゴになっている感じではないですね。

ここで発散したのではないようです。

制御側が急激に角度変更に対応しきれず、発散してしまったように見えました。

tmori, 2024-07-17:
このようにパラメータ調整したらNANが発生しなくなりました。

```
--- a/drone_control/common/include/radio_controller.hpp
+++ b/drone_control/common/include/radio_controller.hpp
@@ -95,8 +95,8 @@ private:
     {
         PidRateControlOutputType out = prev_rate_out;
         if (angular_rate_simulation_time >= angular_rate_cycle) {
-            out.p = get_target_rate_value(this->angular_rate_roll, 1, M_PI/10.0, roll.target, roll.current);
-            out.q = get_target_rate_value(this->angular_rate_pitch, 1, M_PI/10.0, pitch.target, pitch.current);
+            out.p = get_target_rate_value(this->angular_rate_roll, 1, M_PI/1000.0, roll.target, roll.current);
+            out.q = get_target_rate_value(this->angular_rate_pitch, 1, M_PI/1000.0, pitch.target, pitch.current);
             out.r = get_target_rate_value(this->angular_rate_yaw, 20, M_PI/4.0, yaw.target, yaw.current);
             angular_rate_simulation_time = 0;
             prev_rate_out = out;
```

tmori, 2024-07-17:
急激な角度変化に対応できる制御を考えるんですかね。。
ひとまず原因がわかってよかったです。

kenjihiranabe, 2024-07-18:
おお、ありがとう。止まる時とかですかね？
直感的な解釈としては「急激に角度を変更を行うようなトルクをかけて制御しようとしている」ということだと思います。

``　if (angular_rate_simulation_time >= angular_rate_cycle) {
``
というのはどういう意味でしたか？

tmori, 2024-07-18:
> というのはどういう意味でしたか？

これは、姿勢角速度制御の制御周期用のコードです。

tmori, 2024-08-06:
PID制御の問題として扱うので、クローズします。

closed 2024-08-06 by tmori