toppers/hakoniwa-px4sim #284: 機体の想定速度をあらかじめ決める必要があった

tmori, 2024-08-04:


tmori, 2024-08-04:
# 現状の構成

- 機体の質量：0.1kg
- 空気摩擦係数：0.05
- 機体・姿勢角度制御の上限値：20度

上記条件下では、物理的に、機体の速度はどうやっても7m/sec弱までが上限となる。

 $u = \frac{mg \sin \theta}{d}$
 $v = \frac{mg \sin \phi}{d}$
 
 物理式：速度は推力では変更できない

[image]

tmori, 2024-08-04:
# 今後の課題

- 機体に求める速度上限を決まる必要がある
- 想定する機体の水平移動時の空気摩擦係数を決める必要がある（多分、機体の形状で決まると思われる）
- 上記から以下を決める。これらは、想定される運用や制御の限界などから求めるのでは？
  - 機体の質量
  - 機体の姿勢角度の上限値

tmori, 2024-08-04:
空気摩擦係数を下げて、速度を出せるようにしたけど、そうなると制動距離が伸びる。
そのため、姿勢角の上限を増やしてみたが、そうすると制御が不安定になる。

kenjihiranabe, 2024-08-05:
空気摩擦ですが、１次を0にして、２次でやってみては？よりきびきび動くようになると思う。
速度上限付近で同じ抵抗にするようにすると、それより低速では抵抗減るので、上限までの応答が上がる。

tmori, 2024-08-05:
> 空気摩擦ですが、１次を0にして、２次でやってみては？よりきびきび動くようになると思う。 速度上限付近で同じ抵抗にするようにすると、それより低速では抵抗減るので、上限までの応答が上がる。

流石です！
２次だけでやってみたら、上限速度40m/secも余裕で行けました。

***before:***
```json
"airFrictionCoefficient": [ 0.05, 0 ],
```

***after:***
```json
"airFrictionCoefficient": [ 0, 0.0001 ],
```

２次だけにすると、すごく加速するので、制御できなくなってしまう感じです。
上記のパラメータだと、ラジコン操作では発散しなくなりました。
ただし、速度が40m/secまで増えると、制動距離がめちゃくちゃ伸びますね。（体感で１００m近く）

kenjihiranabe, 2024-08-05:
100m !

tmori, 2024-08-05:
25m/secまでなら今のパラメータで制御が安定しました。

これは、３秒後に、25m/secを目標値とした制御実験結果です。
20秒後に、速度を０にして停止させる実験も併せて行なっています。

[image]

加速：
```
OK c(Steady state value)  : 24.943   (Target: 25±0.250 m)
OK T_r(Rise time)         : 5.889 s (Target: ≤ 10.000 s)
OK T_d(Delay time)        : 3.768 s (Target: ≤ 5.000 s)
NG O_s(Maximum overshoot) : 1.152   (Target: ≤ 1.000 m)
OK T_s(5% settling time)  : 7.218 s (Target: ≤ 20.000 s)
```

停止：
```
OK c(Steady state value)  : 0.007   (Target: 0±0.250 m)
OK T_r(Rise time)         : 5.328 s (Target: ≤ 10.000 s)
OK T_d(Delay time)        : 3.456 s (Target: ≤ 5.000 s)
OK O_s(Maximum overshoot) : 0.234   (Target: ≤ 1.000 m)
OK T_s(5% settling time)  : 6.549 s (Target: ≤ 20.000 s)
```

tmori, 2024-08-05:
ブレーキ踏んで、停止するまでの距離はやっぱり100mくらいですね。
ブレーキかけた時間は20秒で320m付近。停止した時の距離は420m付近。

[image]

tmori, 2024-08-05:
ちなみに、速度目標値を40m/secにすると、こんな感じで振動してしまいます。

[image]

kenjihiranabe, 2024-08-05:
なるほどー。モグラ叩きになってしまうとよくないですが、PID のそれぞれが特性にどう影響与えるのか、訂正的な記述がここに。

https://controlabo.com/pid-velocity-control/#:~:text=%E3%81%8C%E3%83%9D%E3%82%A4%E3%83%B3%E3%83%88%E3%81%A7%E3%81%99%E3%80%82-,PID,-%E3%82%B2%E3%82%A4%E3%83%B3%E3%81%AE

tmori, 2024-08-05:
# 一般的な式（ChatGPTに聞いたもの）

[image]

* 空気密度 𝜌
  * ρ は 1.225 kg/m³（標準大気圧下）
* 空気抵抗係数 𝐶𝑑
  * Cd は、マルチロータードローンで一般的に 0.6 から 1.0
* 投影面積 𝐴
  * A は、ドローンの形状とサイズ
  
 時速90km/sec(20m/sec)出せる様にするための dの値は、0.000837855 。

この場合、 Cdを0.6と仮定すると、Aの値は、 0.00228 m²である必要がある。

kenjihiranabe, 2024-08-05:
ぼくも聞いてみました。
ドローンの空気抵抗については、一般的に速度の2乗に比例すると考えられています。これは、流体力学の基本原理に基づいています。
具体的には、以下のように説明できます：

低速域では、空気抵抗は速度にほぼ比例します（層流領域）。
しかし、ドローンの通常の飛行速度範囲では、空気抵抗は速度の2乗にほぼ比例します（乱流領域）。

この関係は以下の式で表されます：
F = 1/2 * ρ * v^2 * Cd * A
ここで、
F: 空気抵抗力
ρ: 空気密度
v: 速度
Cd: 抵抗係数
A: 前面投影面積
この式から、速度(v)が2倍になると、空気抵抗力(F)は4倍になることがわかります。
ただし、実際のドローンの飛行では、姿勢や風の影響など、他の要因も空気抵抗に影響を与えることに注意が必要です。
この情報についてさらに詳しく知りたい場合や、ドローンの制御に関する他の側面について質問がありましたら、お知らせください。

tmori, 2024-08-06:
ひとまず、二乗の方が良さそうなので、これでコミットします。
今後は理論ベースで再評価を実施していく予定です。

commit 6ab0d5f5 (2024-08-06): fix #284
closed 2024-08-06 by tmori