toppers/hakoniwa-px4sim #296: Ubuntuマニュアルのつまづきポイント

tmori, 2024-08-24:
# Ubuntuマニュアルでつまづいたところ

https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/ubuntu22.04_install%26operation_pythonAPI%E7%B7%A8.md

- [ ] [manual-pre-1.2.3. gitの設定] 使うだけなら、ユーザ登録なくても良いのでは？と思いました。
- [ ] [箱庭コマンドおよびライブラリのインストール手順](https://github.com/toppers/hakoniwa-px4sim/blob/main/hakoniwa/README-ja.md#%E7%AE%B1%E5%BA%AD%E3%82%B3%E3%83%9E%E3%83%B3%E3%83%89%E3%81%8A%E3%82%88%E3%81%B3%E3%83%A9%E3%82%A4%E3%83%96%E3%83%A9%E3%83%AA%E3%81%AE%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB%E6%89%8B%E9%A0%86wsl2macubuntu)が漏れておりました(こちらを1.3.2 の前に実行しておく必要があったりします...)。
  - [ ] 1.3.1 の install.bash は、1.3.2 の後ですね。。（オリジナルが間違えてますが。。）
  - [ ] 1.3.3 の実行ディレクトリがわからなかったので、hakoniwa-px4simの下でという一言があるとわかりやすくなると思いました。
  - [ ] 1.4.3 `cd ~/work/hakoniwa-px4sim/hakoniwa` は、`cd ~/work/hakoniwa-px4sim/`ですね。

cubicrootk, 2024-08-24:
箱庭ドローンシミュレータ pythonAPI利用編 (Typo)
P4. 12.1
 bash→bashへの切り替え ->  bash→dashへの切り替え

buildko89, 2024-08-25:
修正しました。コミットしておきます。

buildko89, 2024-08-25:
https://github.com/toppers/hakoniwa-px4sim/blob/develop/docs/manual/ubuntu22.04_install%26operation_PX4%E7%B7%A8.md
https://github.com/toppers/hakoniwa-px4sim/blob/develop/docs/manual/ubuntu22.04_install%26operation_pythonAPI%E7%B7%A8.md
↑
アップしました。確認お願いします。

cubicrootk, 2024-08-25:
px4sim の8/25版 (pdf版)について、Typoの報告です。
p.13 1.4.1
$ cd ~/work/hakoniwa-px4sim/hakoniwa/px4/PX4-Autopilot
$ cd ~/work/hakoniwa-px4sim/px4/PX4-Autopilot

p.15 1.4.3
$~/work/hakoniwa-unity-drone-model
$cd ~/work/hakoniwa-unity-drone-model

p.18 1.5.2
アプリケーション設定の通信リンクをクリックして、設定内容を設定します。
アプリケーション設定の通信リンクをクリックして、初めて使う場合下の方の”＋”ボタンを押して、設定ウインドウに以下の内容を設定します。因みにポート18570はUDPの事です。

cubicrootk, 2024-08-26:
px4sim の8/25版の手順で動作しました、ありがとうございます。
Typoとコメントを追記します。

1.3.1 
・ インストール手順
$ bash build.bash
$ bash install.bash

1.3.4
各インストールした結果を反映させるため、環境変数の設定を実施します。
$ vi .bashrc

$ cd ~/
$ vi .bashrc

1.4.2.
無事に箱庭コア機能が起動したら、PX4を起動したコンソールの方に
　ERROR [simulator-mavlink] poll timeout 0, 111
の様なエラー表示がでるかもしれません。次のUnity起動で止まるはずです。

1.5.3. 
PX4起動→箱庭コア機能起動→Unityアプリ起動の各手順に従って、各機能を起動します。
起動後に、QGCを起動します。
（立ち上げていたPX4/箱庭コア/Unityを終了せずにQGCのみを起動し直すと
　takeoffした状態でQGCにUnityが接続されて、QGc画面左のメニューではLandを実行できるはずです）

tmori, 2024-08-27:
@cubicrootk 

> ログリプレイ機能についてはエラーが出て未解決状態です。

エラーログを教えていただけますか。

buildko89, 2024-08-27:
@cubicrootk 
https://github.com/toppers/hakoniwa-px4sim/blob/develop/docs/manual/ubuntu22.04_install%26operation_PX4%E7%B7%A8.md
https://github.com/toppers/hakoniwa-px4sim/blob/develop/docs/manual/ubuntu22.04_install%26operation_pythonAPI%E7%B7%A8.md
ご指摘事項を修正。pdf版は、Slackにて展開します。

commit 27faede5 (2024-08-27): Re-Fix #296

Fixed a suggestion from @cubicrootk in #296