toppers/hakoniwa-px4sim #299: ミキサーに関する調査（参考文献のメモ）

kenjihiranabe, 2024-08-28:
https://www.mathworks.com/content/dam/mathworks/mathworks-dot-com/solutions/automotive/files/jp-expo-2015/industrial-revolution-of-the-skies-and-near-future-due-to-drones.pdf
https://www.docswell.com/s/Kouhei_Ito/KDVNVK-2022-06-15-193343#p6

tmori, 2024-09-05:
現状の実装情報の共有：

* 現行ソース：
  * ファクトリー部分：[aircraft_factory.cpp](https://github.com/toppers/hakoniwa-px4sim/blob/514cdcd3f2dbd24c473c682ee9cfe6268e836199/hakoniwa/src/assets/drone/aircraft/aircraft_factory.cpp#L185-L194)
  * コール部分：[drone_control_proxy.hpp](https://github.com/toppers/hakoniwa-px4sim/blob/main/hakoniwa/src/assets/drone/controller/drone_control_proxy.hpp#L347-L357)
  * 本体：[drone_mixer.hpp](https://github.com/toppers/hakoniwa-px4sim/blob/main/hakoniwa/src/assets/drone/controller/drone_mixer.hpp)
* ビルド方法
* DRONE_ENABLE_MIXERを有効にする
* 動作確認方法
  * Python APIでドローン飛行させてみて、エラーが出なければOK?

kenjihiranabe, 2024-09-06:
計算方法の戦略を、ここに修正記述。
https://github.com/toppers/hakoniwa-px4sim/issues/178#issuecomment-2073695433

tmori, 2024-09-14:
ひとまず、従来のミキサーでも動作することを確認しました。

tmori, 2024-09-14:
ミキサーを機体パラメータでオンオフできるようにしました。

https://github.com/toppers/hakoniwa-px4sim/issues/334

closed 2024-09-18 by kenjihiranabe