toppers/hakoniwa-px4sim #328: 高度制御の理論的導出

tmori, 2024-09-12:


tmori, 2024-09-12:
# 補足情報

### $\( \Delta T(s) / \Delta U(s) \)$ の導出:

線形化では、**動作点 $\( U_0 \)$** に対する入力の小さな変動 $\( \Delta U(s) \)$ に対するトルクの変動 $\( \Delta T(s) \)$ を考えます。具体的には、次のように展開されます：

1. **元の関係式**:
$T(s) = N A \Omega(s)^2 = N A \left( \frac{K_r}{T_r s + 1} U(s) \right)^2$

2. **動作点周りでの摂動**：
   入力 $\( U(s) \)$ を定常値 $\( U_0 \)$ とその変化 $\( \Delta U(s) \)$ に分けます：
$   U(s) = U_0 + \Delta U(s)$
   このとき、 $\( U(s)^2 \)$ は次のように展開されます：
 $  U(s)^2 = (U_0 + \Delta U(s))^2 = U_0^2 + 2 U_0 \Delta U(s) + \Delta U(s)^2$
   ここで、**高次の小さな変動項** $\( \Delta U(s)^2 \)$ は無視できるため、次のように近似できます：
$   U(s)^2 \approx U_0^2 + 2 U_0 \Delta U(s)$

3. **トルクの変動 $\( \Delta T(s) \)$ を求める**：
   これにより、トルク $\( T(s) \)$ の変動は次のように表されます：
   $T(s) = N A \left( \frac{K_r}{T_r s + 1} \right)^2 (U_0^2 + 2 U_0 \Delta U(s))$
   したがって、トルクの変動 $\( \Delta T(s) \) $ は次の形になります：
$   \Delta T(s) \approx 2 N A U_0 \cdot \left( \frac{K_r}{T_r s + 1} \right)^2 \Delta U(s)$

4. **線形化された伝達関数**  $\( \frac{\Delta T(s)}{\Delta U(s)} \)$ ：
   動作点周りの小さな摂動に対するトルクの変動に対して、伝達関数は次のように表されます：
$   \frac{\Delta T(s)}{\Delta U(s)} = 2 N A U_0 \cdot \frac{K_r^2}{(T_r s + 1)^2}$

### 結論：
線形化された摂動系の伝達関数 $\( \frac{\Delta T(s)}{\Delta U(s)} \)$ は次のようになります：

$\frac{\Delta T(s)}{\Delta U(s)} = 2 N A U_0 \cdot \frac{K_r^2}{(T_r s + 1)^2}$

tmori, 2024-09-12:
# 制御側

$C(s) = (K_p + mg) + K_i/s + sK_d$

tmori, 2024-09-13:
# 各種パラメータ

以下を前提とする
https://github.com/toppers/hakoniwa-px4sim/issues/326

* m: 0.71
* N: 4
* Kr: 2896
* Tr: 1.93e-02

tmori, 2024-09-13:
# プラント全体

$\frac{\Delta T(s)}{\Delta U(s)} = 2 N A U_0 \cdot \frac{K_r^2}{(T_r s + 1)^2}$

$\frac{Z(s)}{\Delta T(s)} = -\frac{1}{s(ms + d)}$

$\frac{Z(s)}{\Delta U(s)} = -\frac{2 N A U_0 K_r^2}{s(ms + d)(T_r s + 1)^2}$

tmori, 2024-09-13:
# 制御側

$C(s) = (K_p + U_0) + K_i/s + sK_d$

tmori, 2024-09-13:
# 各種パラメータ

以下を前提とする
https://github.com/toppers/hakoniwa-px4sim/issues/326

* m: 0.71
* N: 4
* Kr: 2896(最大回転数)
* Tr: 1.93e-02
* $U_0$ : $\sqrt{mg/NA Kr^{2}}$
  * ホバリング状態であり、 $T_0$ が、重力と一致する
  *  $T_0 = NA K_r^{2} U_0^2$ から、 $U_0$ を求めた

tmori, 2024-09-13:
# L(s) を求める

$C(s) = K_p  + K_i/s + sK_d = \frac{K_d s^{2} + Kp s + Ki}{s}$

$P(s) = -\frac{2 N A U_0 K_r^2}{s(ms + d)(T_r s + 1)^2}$

$L(s) = C(s) P(s) = -\frac{2 N A U_0 K_r^2(K_d s^{2} + Kp s + Ki)}{s^{2}(ms + d)(T_r s + 1)^2}$

tmori, 2024-09-13:
# 閉ループ特性方程式 1+𝐿(𝑠)を求める

$1 + L(s) =  \frac{s^{2}(ms + d)(T_r s + 1)^2 - 2 N A U_0 K_r^2(K_d s^{2} + Kp s + Ki)}{s^{2}(ms + d)(T_r s + 1)^2}$

tmori, 2024-09-13:
うーん、そもそも不安定であることが判明してしまった。まいった。

tmori, 2024-09-13:
わかった！NED座標系だから、PID制御パラメータは全部マイナスにしないといけないということか！

tmori, 2024-09-13:
# 調整後のパラメータ

```json
{
    "plants": [
        {
            "comment": "Rotor",
            "num": [
                "2 * N * A * U0 * Kr**2"
            ],
            "den": [
                "Tr**2",
                "2*Tr",
                "1"
            ]
        },
        {
            "comment": "Altitude",
            "num": [
                "-1"
            ],
            "den": [
                "m",
                "d",
                "0"
            ]
        }
    ],
    "controller": {
        "num": [
            "Kd",
            "Kp",
            "Ki"
        ],
        "den": [
            "1",
            "0"
        ]
    },
    "constants": {
        "Kp": -3.6182108001838924,
        "Ki": -0.1,
        "Kd": -0.5599136193021486,
        "m": 0.71,
        "g": 9.81,
        "N": 4,
        "d": 0.05,
        "A": 8.3e-7,
        "Tr": 1.93e-02,
        "Kr": 2896,
        "U0": "math.sqrt( (m * g) / (N * A * (Kr**2)) )",
        "Wc": 20,
        "PM": 30
    }    
}
```

tmori, 2024-09-13:
# 閉ループ伝達関数の極

左反平面！

[image]

```
システムの極: [-7.81364819e+01 +0.j         -8.31995767e+00+21.83536742j
 -8.31995767e+00-21.83536742j -8.89321092e+00 +0.j
```

tmori, 2024-09-13:
# ボード線図と位相線図

[image]

```
ゲイン余裕 (Gain Margin): 11.013343356723455 dB
位相余裕 (Phase Margin): 30.009422184890894 degrees
ゲイン余裕発生周波数: 44.97084634065474 rad/s
位相余裕発生周波数: 20.059957375420947 rad/s
```

tmori, 2024-09-13:
# 箱庭での実験結果

このパラメータを使ってもうまくいかなかった。
振動してしまう。

以下にしたら良い感じだった。

```
PID_PARM_ALT_Kp 1.0
PID_PARM_ALT_Ki 0.10
PID_PARM_ALT_Kd 50.0
```

箱庭の制御は高度を正負反転しているので、正の値が正しい

tmori, 2024-09-13:
# このギャップがどこからくるものなのか？

- ミキサーの不連続性による影響もありそう。結構、推力をマイナスにしようとしている
- 離散時間での誤差もありそう
- 制御周期の問題もあるかも

tmori, 2024-09-13:
# 検証のためにミキサーなしでやってみよう

## プラント

$Z(s) = -\frac{ \Delta T(s)}{s(ms + d)}$

## 制御

$C(s) = K_p  + K_i/s + sK_d$

kenjihiranabe, 2024-09-13:
あれ？

1. **元の関係式**:
$T(s) = N A \Omega(s)^2 = N A \left( \frac{K_r}{T_r s + 1} U(s) \right)^2$

ですが、これ違ってませんか？

$T(t) = N A \Omega(t)^2$

なので、 $f(t)^2$ をラプラス変換する必要があって、これは不能です。ですが、 $t$ の領域で $U_0$ と  $T_0$ のまわりで線形化することはできて、そこからラプラス変換ですね。カンファレンス聴きながら、もう一回式を追ってみます。

kenjihiranabe, 2024-09-14:
https://github.com/toppers/hakoniwa-px4sim/issues/328#issuecomment-2347422951

$\Delta T(s)/\Delta U(s)$ は通常の一時遅れでした。(U->Ω->T)

$\Delta T(s)/\Delta U(s) = 2 \Omega_0 N A  \frac{K_r}{Tr s + 1}$ 

[image]
[image]

kenjihiranabe, 2024-09-18:
とりあえず、全体の流れを整理して、式をもう一度作ってみた。

- u でなくΔu を使う（そうしないと、ΔΩが作れない）-> U0 を足さない
- Lの先頭のマイナスが気になる。z座標系をにしたときに、L(s)+1 = 0 の極が異なってしまう（まだどこかミスしている）。
[image]

kenjihiranabe, 2024-09-18:
野波先生の 式(2.92) と上記 P(s) は一致するので、よしとする。

kenjihiranabe, 2024-09-18:
大好物です❗️線型代数版と合わせて理解が進みます。
シルエットは「軌道」と言い換えられるかと思ったのですがちょっと違いました。「<τ>の軌道」ならあってますか？

kenjihiranabe, 2024-09-18:
zが下向きの時、u=duty や Ωを上げる方向なので、Kp はマイナス値になるんですね。

kenjihiranabe, 2024-09-18:
手計算した結果、この式と全く一緒になりました！
https://colab.research.google.com/drive/14hSlp1h4ffHOOTq3G3upaA6x1kknOPqh

kenjihiranabe, 2024-09-18:
いろいろ Kp, Ki, Kd を触ってみて、再度、理論的に分かったこと。

- Ki は不要。高度の運動方程式に 1/s があるので、「内部モデル原理」により、誤差は必ず 0 になる。
- Kd は安定化のために必ず必要。ないと位相が -180 を超えてしまう。

Kp = -1, Ki=0, Kd = -10 で位相余裕を確保してみた。

[image]

tmori, 2024-09-19:
# 箱庭で実験した結果です。

## 測定条件

- 目標高度：-3m
- Kp: -1
- Kd: -1, -10

## 測定結果

### Kp=-1, Kd=-1

目標高度に到達しましたが、整定時間が長いのと、結構振動しますね。。

```
NG c(Steady state value)  : -2.999   (Target: 0±-0.030 m)
OK T_r(Rise time)         : 1.550 s (Target: ≤ 10.000 s)
OK T_d(Delay time)        : 1.659 s (Target: ≤ 5.000 s)
NG O_s(Maximum overshoot) : 2.216   (Target: ≤ 1.000 m)
NG T_s(5% settling time)  : 79.721 s (Target: ≤ 20.000 s)
```
[image]

### Kp=-1, Kd=-10

目標高度に到達しました。整定時間も短くなりましたが、まだ振動しますね。。

```
NG c(Steady state value)  : -3.000   (Target: 0±-0.030 m)
OK T_r(Rise time)         : 1.614 s (Target: ≤ 10.000 s)
OK T_d(Delay time)        : 1.687 s (Target: ≤ 5.000 s)
NG O_s(Maximum overshoot) : 1.646   (Target: ≤ 1.000 m)
NG T_s(5% settling time)  : 27.410 s (Target: ≤ 20.000 s)
```
[image]

tmori, 2024-09-19:
ちなみに、箱庭のパラメータで実施した結果はこんな感じです。

- Kp = -10
- Ki = -0.1
- Kd = -500

```
NG c(Steady state value)  : -3.001   (Target: 0±-0.030 m)
OK T_r(Rise time)         : 1.474 s (Target: ≤ 10.000 s)
OK T_d(Delay time)        : 0.981 s (Target: ≤ 5.000 s)
OK O_s(Maximum overshoot) : 0.129   (Target: ≤ 1.000 m)
OK T_s(5% settling time)  : 2.031 s (Target: ≤ 20.000 s)
```
[image]

kenjihiranabe, 2024-09-19:
ありがとう！ Kp=-1, Ki=0, Kd=-1 で線形と箱庭で比較。
線形
[image]
箱庭
[image]

結構線形モデルと挙動が違うなぁ。線形だとダンパが効くけど、箱庭では効いていない。
ダンパになるのは、Kd と空気抵抗係数、だけど、空気抵抗係数はそろえている？

tmori, 2024-09-19:
> 結構線形モデルと挙動が違うなぁ。線形だとダンパが効くけど、箱庭では効いていない。
ダンパになるのは、Kd と空気抵抗係数、だけど、空気抵抗係数はそろえている？

空気抵抗係数は 0.05 で、一致させていますね。

箱庭のパラメータ：

https://github.com/toppers/hakoniwa-px4sim/blob/267c15cdd1ae9b2e83df08474d9c00bd8f791e91/hakoniwa/config/mixer-api/drone_config_0.json#L41

kenjihiranabe, 2024-09-19:
ありがとう。なるほど。そうなると、そもそも線形化したモデルは、箱庭のモデルから乖離している気がするなぁ。。。どこか計算間違っているのかなぁ。二人とも同じ答えになったしなぁ。。。どう詰めていくのかが今は不明です。
箱庭側のBode線図って実測（システム同定）でかけないものか、、、control にそんな機能あるといいのにな、Chirp 信号を入れて測るらしいが、、、
https://blog.control-theory.com/entry/2024/03/28/153529

kenjihiranabe, 2024-09-19:
ここで、箱庭のパラメータと合わせている、ということでしたが、
https://github.com/toppers/hakoniwa-px4sim/blob/267c15cdd1ae9b2e83df08474d9c00bd8f791e91/hakoniwa/config/mixer-api/drone_config_0.json#L41

たしか、機体側にもこのパラメータなかったでしたっけ？（確か、x, y, z の３方向で別々に設定していたはず）

tmori, 2024-09-19:
> ここで、箱庭のパラメータと合わせている、ということでしたが、
> 
> https://github.com/toppers/hakoniwa-px4sim/blob/267c15cdd1ae9b2e83df08474d9c00bd8f791e91/hakoniwa/config/mixer-api/drone_config_0.json#L41
> 
> たしか、機体側にもこのパラメータなかったでしたっけ？（確か、x, y, z の３方向で別々に設定していたはず）

こちらは、機体側のパラメータになりますね。x, y, z の3方向で別々の空気摩擦抵抗設定できるようにする話はまだ未対応ですね。。

kenjihiranabe, 2024-09-19:
https://github.com/toppers/hakoniwa-px4sim/blob/267c15cdd1ae9b2e83df08474d9c00bd8f791e91/hakoniwa/src/config/drone_config.hpp#L163C1-L164C1

ここは、このコンフィグパラメータを読んでいるということであってますか？

tmori, 2024-09-19:
> https://github.com/toppers/hakoniwa-px4sim/blob/267c15cdd1ae9b2e83df08474d9c00bd8f791e91/hakoniwa/src/config/drone_config.hpp#L163C1-L164C1
> 
> ここは、このコンフィグパラメータを読んでいるということであってますか？

はい、あってます。

tmori, 2024-09-21:
箱庭のPID制御プログラム、よくみてみたら誤りがありました。

```C
    double calculate(double sp, double input) {
        target = sp;
        double error = target - input;
        integral += error * delta_time;

        double derivative = (first_time) ? 0.0 : error - prev_error;
        first_time = false;
        prev_error = error;
        return Kp * error + Ki * integral + Kd * derivative;
    }
```

PID制御式(連続時間)：
[image]

PID制御式(離散時間)：
[image]

Kdのところ、Δtで割らないとダメですね。。

```diff
- double derivative = (first_time) ? 0.0 : error - prev_error;
+ double derivative = (first_time) ? 0.0 : (error - prev_error) / delta_time;
```

tmori, 2024-09-21:
# さらに制御周期は連続系と同じようにシミュレーション周期と一致させてみたら

```
NG c(Steady state value)  : -3.000   (Target: 0±-0.030 m)
OK T_r(Rise time)         : 2.044 s (Target: ≤ 10.000 s)
OK T_d(Delay time)        : 0.646 s (Target: ≤ 5.000 s)
OK O_s(Maximum overshoot) : 0.000   (Target: ≤ 1.000 m)
OK T_s(5% settling time)  : 2.788 s (Target: ≤ 20.000 s)
```

振動が止まりました。

[image]

tmori, 2024-09-21:
# Kp=1.0, Kd=0.3の結果
[image]

tmori, 2024-09-22:
# throttle_gainのパラメータが悪さしていました。

PID制御で決めた値は以下のように推力に変換されます。

```cpp
    double run_thrust_control(double power) {
        return (this->mass * this->gravity) + (this->throttle_gain * power);
    }
```
- ` (this->mass * this->gravity)`は、 $T_0$ に対応します。
- powerは、PID制御で決まる値です。
  - 決めた値は、`±PID_PARAM_MAX_POWER` でリミット入れてます。
  - 今回採用した値：PID_PARAM_MAX_POWER 9.81

ここで、`throttle_gain` は10.0でしたが、これは連続系での解析では1.0に相当するので、これを連続系に合わせて、1.0に変更しました。

結果：
[image]

```
NG c(Steady state value)  : 3.000   (Target: 0±-0.030 m)
OK T_r(Rise time)         : 1.582 s (Target: ≤ 10.000 s)
OK T_d(Delay time)        : 1.098 s (Target: ≤ 5.000 s)
OK O_s(Maximum overshoot) : 0.022   (Target: ≤ 1.000 m)
OK T_s(5% settling time)  : 4.253 s (Target: ≤ 20.000 s)
```

PIDパラメータ：
```
PID_PARM_ALT_Kp 1.0
PID_PARM_ALT_Ki 0.0
PID_PARM_ALT_Kd 1.0
```

グラフとしては同じようにオーバーシュートしてきました。ただ、応答時間は約10倍くらい遅いのが気になるところ。。

tmori, 2024-09-22:
# ミキサーの種類

ミキサーは、`linear` でないと、同じ結果になりませんね。
`None`の場合は、ホバリングしませんでした。。

tmori, 2024-09-22:
# 伝達関数とシミュレーションの条件を変えてみる

伝達関数側　：ローターを外して、直接、推力を与える
シミュレータ：ミキサーとロータを外して、直接、推力を与える

結果、ほぼ同じ振る舞いで、同じ時間スケールに見える。

## 伝達関数側のステップ応答

[image]

## シミュレータの応答

[image]

tmori, 2024-09-22:
# 考察

伝達関数側のモデル化が、シミュレータの側のものと一致していない可能性があるのでは？

tmori, 2024-09-22:
# シミュレータの処理の流れ（高度制御）

1. [高度]→PID制御→[推力]
2. [推力]→ミキサ→[Duty]
3. [Duty]→ロータ→[RotorSpeed]
4. [RotorSpeed]→スラスタ→[推力]

tmori, 2024-09-22:
# ローターの処理

```C
    void run(double control) override
    {
        //w += (control - w) * (1.0 - Math.exp(-dt / tau));
        this->w += (control - this->w) * (1.0 - exp(-this->delta_time_sec/ this->param_tr));
        this->total_time_sec += this->delta_time_sec;
    }
```

tmori, 2024-09-22:
# スラスタの処理

```C
double rotor_thrust(
    double A, /* the A parameter in Trust = A*(Omega)^2 */
    double omega /* in rpm */ )
{
    /**
     * Nonami's book (2.50)
     * T = A * (Omega)^2
     */
    return A * (omega * omega);
}
```

tmori, 2024-09-22:
# ミキサー処理

```C

        // linearized version
        PwmDuty run_linear(double mass, double thrust, double torque_x, double torque_y, double torque_z)
        {
            const double m = mass; // pass through mass from caller
            const int N = ROTOR_NUM;
            const double g = GRAVITY;
            const double A = param_A;
            const double T0 = m*g; // equibilium
            const double omega0 = std::sqrt(T0/(N*A)); // equibilium
            const double delta_T = thrust - T0;
            const double Kr = param_Kr;
            
            glm::dvec4 delta_U = glm::dvec4(delta_T, torque_x, torque_y, torque_z);
            glm::dvec4 delta_omega_times_omega0 = 0.5 * MA_inv * delta_U;
            glm::dvec4 delta_omega = delta_omega_times_omega0/omega0;

            glm::dvec4 omega = omega0 + delta_omega;

            PwmDuty duty = {};
            for (int i = 0; i < ROTOR_NUM; i++) {
                if (omega[i] < 0) {
                    if (mixer_info.enableErrorLog) {
                        std::cout << "ERROR: thrust:" << thrust << " tx: " << torque_x << " ty: " << torque_y << " tz: " << torque_z << std::endl;
                        std::cout << "ERROR: Invalid caluculation of Omega to duty because of omega("<< i << ") is minus value(" << omega[i] << ")..." << std::endl;
                    }
                    omega[i] = 0.0;
                }
                else {
                    duty.d[i] = omega[i] / Kr;
                }
                if (mixer_info.enableDebugLog) {
                    std::cout << "Motor " << i << ": Omega = " << omega[i] << ", PWM Duty = " << duty.d[i] << std::endl;
                }                
            }
            return duty;
        }
```

tmori, 2024-09-22:
# ブロック線図をもう一度

[image]

kenjihiranabe, 2024-09-22:
抜けてないと思う。位置だけなら、ミックス行列かけても単にN倍なので。

tmori, 2024-09-22:
# プログラムからMixerの伝達関数を求めてみる

```
delta_T = thrust - T0
delta_U = glm::dvec4(delta_T, torque_x, torque_y, torque_z)
delta_omega_times_omega0 = 0.5 * MA_inv * delta_U
delta_omega = delta_omega_times_omega0/omega0
omega = omega0 + delta_omega;
duty.d[i] = omega[i] / Kr;
```

トルクは０と考えてよくて、`MA_inv` は、おそらく `1/A` だけが残る。

そうすると、こうなるのでは？

$\frac{\Delta U}{\Delta T} = \frac{1}{2A\Omega_0 K_r}$

tmori, 2024-09-22:
# レッツとらい！

プログラムの都合上、Controller側にMixerかけないので、Plant側に書いた。

```json

# plantsにプラントモデルの伝達関数を列挙する
# controllerに制御側の伝達関数を定義する
# constantsに定数を定義する
data = {
    "plants": [
        {
            "comment": "Mixer",
            "num": [
                "1"
            ],
            "den": [
                "2 * N * A * W0 * Kr"
            ]
        },
        {
            "comment": "Rotor",
            "num": [
                "2 * N * A * W0 * Kr"
            ],
            "den": [
                "Tr",
                "1"
            ]
        },
        {
            "comment": "Altitude",
            "num": [
                "-1"
            ],
            "den": [
                "m",
                "d",
                "0"
            ]
        }
    ],
    "controller": {
        "num": [
            "Kd",
            "Kp",
            "Ki"
        ],
        "den": [
            "1",
            "0"
        ]
    },
    "constants": {
        "Kp": -1,
        "Ki": 0.0,
        "Kd": -1,
        "m": 0.71,
        "g": 9.81,
        "N": 4,
        "d": 0.05,
        "A": 8.3e-7,
        "Tr": 1.93e-02,
        "Kr": 2896,
        "W0": 1448,
        "U0": "math.sqrt( (m * g) / (N * A * (Kr**2)) )",
        "Wc": 1002,
        "PM": 90
    }
}

```

tmori, 2024-09-22:
# 伝達関数モデルのステップ応答結果

ボード線図：
[image]

```
ゲイン余裕 (Gain Margin): inf dB
位相余裕 (Phase Margin): 59.34687263836736 degrees
ゲイン余裕発生周波数: nan rad/s
位相余裕発生周波数: 1.6457394067612157 rad/s
```

ステップ応答：
[image]

tmori, 2024-09-22:
# 箱庭ドローンシミュレータの結果

```
NG c(Steady state value)  : 1.000   (Target: -1±-0.010 m)
OK T_r(Rise time)         : 1.576 s (Target: ≤ 10.000 s)
OK T_d(Delay time)        : 1.198 s (Target: ≤ 5.000 s)
OK O_s(Maximum overshoot) : 0.085   (Target: ≤ 1.000 m)
OK T_s(5% settling time)  : 4.373 s (Target: ≤ 20.000 s)
```

[image]

tmori, 2024-09-22:
おー、ほぼ理論通りの結果になった！！
やったー。

tmori, 2024-09-22:
ログリプレイしてみた。

https://github.com/user-attachments/assets/06a92ba1-25ef-40db-a244-07cdf4143cfb

tmori, 2024-09-22:
# パラメータ調整してみる。

- Kp = 1.0
- Ki = 0.0
- Kd = 1.0

## 考察

理論ベースでの考察として、
- 位相余裕は60度くらいあって安定感はある。
- カットオフ周波数は、1.6rad/secくらいしかないので、即応性が気になる。

実際、箱庭ドローンシミュレータでシミュレーションしてみると、
-  整定時間が4.373 sなので、もう少し早くしたい。

## トライ

理論ベースでパラメータ調整してみた。

- Kp = 10
- Ki = 0
- Kd = -10

```
ゲイン余裕 (Gain Margin): inf dB
位相余裕 (Phase Margin): 71.34235069314849 degrees
ゲイン余裕発生周波数: nan rad/s
位相余裕発生周波数: 13.655721668281638 rad/s
```

位相余裕も増えたし、カットオフ周波数は10倍近くなった！

ステップ応答

[image]

## 箱庭ドローンシミュレータで実験してみる。

オーバーシュートがなくなって、整定時間が１秒近く短縮した。

```
NG c(Steady state value)  : 3.000   (Target: -3±-0.030 m)
OK T_r(Rise time)         : 2.043 s (Target: ≤ 10.000 s)
OK T_d(Delay time)        : 0.768 s (Target: ≤ 5.000 s)
OK O_s(Maximum overshoot) : 0.000   (Target: ≤ 1.000 m)
OK T_s(5% settling time)  : 2.908 s (Target: ≤ 20.000 s)
```
[image]

kenjihiranabe, 2024-09-23:
いい感じですね。理論とシミュレーションが合うのが気持ちいい。

tmori, 2024-09-23:
理解しました！

あとは、この辺りの線形モデル化のドキュメントで説明ですね。

tmori, 2024-11-06:
こちらで管理するので、一旦、クローズ

https://github.com/toppers/hakoniwa-drone-education/issues/25

closed 2024-11-06 by tmori