toppers/hakoniwa-px4sim #351: プラントのシステム同定可能なモジュールを作る

tmori, 2024-10-03:
# 追加対象

- ルンゲクッタモデルの境界条件を取り払ったものの追加

tmori, 2024-10-04:
# 実験結果

## 測定条件

- 入力
  - sineカーブ
  - 周波数：1Hz
  - Thrust：6.585

- 出力
  - 高度の速度：Vz
 
- データの範囲
  - 計測時間：100秒

## 実行コマンド

```
bash -x ../src/tools/eval-plant.bash 100 1 T:6.585 Tx:0 Ty:0 Tz:0 | tee log.txt
```

## 実験結果

[image]

tmori, 2024-10-04:
# 実験結果(0.1Hz)

## 測定条件

- 入力
  - sineカーブ
  - 周波数：0.1Hz
  - Thrust：6.585

- 出力
  - 高度の速度：Vz
 
- データの範囲
  - 計測時間：1000秒

## 実行コマンド

```
bash -x ../src/tools/eval-plant.bash 1000 0.1 T:6.585 Tx:0 Ty:0 Tz:0 | tee log.txt
```

```
bash ../src/tools/show-inout.bash 5 955 in.csv drone_log0/drone_dynamics.csv Vz
```

## 実験結果

```
python ../src/tools/hako_TimelineAnalyzer.py  input.csv output.csv  --diff --columns input.value output.Vz  --diff --start_time 500 --duration 40
```

[image]

ローター回転数：
[image]

tmori, 2024-10-04:
# 実験結果(0.01Hz)

## 測定条件

- 入力
  - sineカーブ
  - 周波数：0.01Hz
  - Thrust：6.585

- 出力
  - 高度の速度：Vz
 
- データの範囲
  - 計測時間：1000秒

## 実行コマンド

```
bash -x ../src/tools/eval-plant.bash 1000 0.01 T:6.585 Tx:0 Ty:0 Tz:0 | tee log.txt
```

```
bash ../src/tools/show-inout.bash 5 955 in.csv drone_log0/drone_dynamics.csv Vz
```

## 実験結果

```
python ../src/tools/hako_TimelineAnalyzer.py  input.csv output.csv  --diff --columns input.value output.Vz  --diff --start_time 500 --duration 400
```

[image]

ローター回転数：
[image]

tmori, 2024-10-04:
# 実験結果(10Hz)

## 測定条件

- 入力
  - sineカーブ
  - 周波数：10Hz
  - Thrust：6.585

- 出力
  - 高度の速度：Vz
 
- データの範囲
  - 計測時間：100秒

## 実行コマンド

```
bash -x ../src/tools/eval-plant.bash 100 10 T:6.585 Tx:0 Ty:0 Tz:0 | tee log.txt
```

```
bash ../src/tools/show-inout.bash 5 95 in.csv drone_log0/drone_dynamics.csv Vz
```

## 実験結果

```
python ../src/tools/hako_TimelineAnalyzer.py  input.csv output.csv  --diff --columns input.value output.Vz  --diff --start_time 80 --duration 0.4
```

[image]

ローター回転数：
[image]

tmori, 2024-10-04:
# 実験結果(100Hz)

## 測定条件

- 入力
  - sineカーブ
  - 周波数：100Hz
  - Thrust：6.585

- 出力
  - 高度の速度：Vz
 
- データの範囲
  - 計測時間：10秒

## 実行コマンド

```
bash -x ../src/tools/eval-plant.bash 10 10 T:6.585 Tx:0 Ty:0 Tz:0 | tee log.txt
```

```
bash ../src/tools/show-inout.bash 5 5 in.csv drone_log0/drone_dynamics.csv Vz
```

## 実験結果

```
python ../src/tools/hako_TimelineAnalyzer.py  input.csv output.csv  --diff --columns input.value output.Vz  --diff --start_time 3 --duration 0.5
```

[image]

ローター回転数：
[image]

kenjihiranabe, 2024-10-04:
メモ：座標の正負の関係で、出力はもともと 180° ずれていることに注意。

tmori, 2024-10-04:
# 実験結果(3Hz)

## 測定条件

- 入力
  - sineカーブ
  - 周波数：3Hz
  - Thrust：6.585

- 出力
  - 高度の速度：Vz
 
- データの範囲
  - 計測時間：300秒

## 実行コマンド

```
bash -x ../src/tools/eval-plant.bash 300 3 T:6.585 Tx:0 Ty:0 Tz:0 | tee log.txt 
```

```
bash ../src/tools/show-inout.bash 50 250 in.csv drone_log0/drone_dynamics.csv Vz
```

## 実験結果

```
python ../src/tools/hako_TimelineAnalyzer.py  input.csv output.csv  --diff --columns input.value output.Vz  --diff --duration 1
```

[image]

ローター回転数：
[image]

tmori, 2024-10-04:
# 実験結果(30Hz)

## 測定条件

- 入力
  - sineカーブ
  - 周波数：30Hz
  - Thrust：6.585

- 出力
  - 高度の速度：Vz
 
- データの範囲
  - 計測時間：300秒

## 実行コマンド

```
bash -x ../src/tools/eval-plant.bash 300 30 T:6.585 Tx:0 Ty:0 Tz:0 | tee log.txt 
```

```
 bash ../src/tools/show-inout.bash 100 200 in.csv drone_log0/drone_dynamics.csv Vz
```

## 実験結果

```
python ../src/tools/hako_TimelineAnalyzer.py  input.csv output.csv  --diff --columns input.value output.Vz  --diff --duration 0.1
```

[image]

ローター回転数：

[image]

tmori, 2024-10-04:
# 実験結果(0.3Hz)

## 測定条件

- 入力
  - sineカーブ
  - 周波数：0.3Hz
  - Thrust：6.585

- 出力
  - 高度の速度：Vz
 
- データの範囲
  - 計測時間：1000秒

## 実行コマンド

```
bash -x ../src/tools/eval-plant.bash 1000 0.3 T:6.585 Tx:0 Ty:0 Tz:0 | tee log.txt
```

```
bash ../src/tools/show-inout.bash 5 955 in.csv drone_log0/drone_dynamics.csv Vz
```

## 実験結果

```
python ../src/tools/hako_TimelineAnalyzer.py  input.csv output.csv  --diff --columns input.value output.Vz  --diff --start_time 500 --duration 10
```

[image]

ローター回転数：
[image]

tmori, 2024-10-04:
# 実験結果(0.03Hz)

## 測定条件

- 入力
  - sineカーブ
  - 周波数：0.03Hz
  - Thrust：6.585

- 出力
  - 高度の速度：Vz
 
- データの範囲
  - 計測時間：1000秒

## 実行コマンド

```
bash -x ../src/tools/eval-plant.bash 1000 0.03 T:6.585 Tx:0 Ty:0 Tz:0 | tee log.txt
```

```
bash ../src/tools/show-inout.bash 5 955 in.csv drone_log0/drone_dynamics.csv Vz
```

## 実験結果

```
 python ../src/tools/hako_TimelineAnalyzer.py  input.csv output.csv  --diff --columns input.value output.Vz  --diff --start_time 500 --duration 100
```

[image]

ローター回転数：
[image]

kenjihiranabe, 2024-10-05:
とりあえず、グラフにしてみました。後で考察書きます。
https://docs.google.com/spreadsheets/d/1JACdNpwAYr-OUya5DmPa8r_1QfsI-4M4G-kJ_zLCvQ0/edit?usp=sharing

[image]

commit 4ed5000b (2024-10-03): fix #351
closed 2024-10-06 by tmori