toppers/hakoniwa-px4sim #358: version 2.7.0 のリリース

tmori, 2024-10-07:
# 機能拡張内容

- [ ] ミキサー対応
- [ ] 風対応
- [ ] マニュアル対応
- [ ] Windows版のインストーラ対応
- [ ] 制御機能の大幅な改修
- [ ] Python APIの機能追加（Unity座標系のAPI対応、キャンセル機能）
- [ ] ラジコン操作のプロポ対応
- [ ] ドローン制御工学教育向けの機能アップデート

buildko89, 2024-10-08:
@yamati-kz さん
[windows_hakowin_installer.md](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/windows_hakowin_installer.md)の追記の件は、了解です。

ちょっと別件で必要になり、突貫で作ってしまったところがあります。

ご提案の章立てのイメージは良いかと思います。
ゲームパッドのデバッグについては、追記した方が、無難かと思います。

今後、なるべくドキュメントを一本化していく方向で整理予定ではあります。

yamati-kz, 2024-10-08:
@buildko89 さん　ご返事ありがとうございます。
https://github.com/toppers/hakoniwa-px4sim/pull/361 にて追記してみました。
記載して記述があるといいかなと思った内容です。

> ゲームパッドのデバッグについては、追記した方が、無難かと思います。

私自身Windowsで試したことがないので、今後、手順書を試しながらご提案させていただければと思います。

buildko89, 2024-10-09:
issue ♯362は対応します。少々お待ちください。

tmori, 2024-10-09:
自分は、以下のissueをやっていきますね。

 https://github.com/toppers/hakoniwa-px4sim/issues/368
 https://github.com/toppers/hakoniwa-px4sim/issues/369
 https://github.com/toppers/hakoniwa-px4sim/issues/370
 https://github.com/toppers/hakoniwa-px4sim/issues/371
 https://github.com/toppers/hakoniwa-px4sim/issues/359
 https://github.com/toppers/hakoniwa-px4sim/issues/360

buildko89, 2024-10-09:
@yamati-kz さん @tmori さん
issue #362対応しました。PR投げました。

yamati-kz, 2024-10-09:
@buildko89 さん @tmori さん
https://github.com/toppers/hakoniwa-px4sim/issues/362 ご対応ありがとうございました。

commit 0fbe9bc3 (2024-10-19): fix #358
closed 2024-10-28 by tmori