toppers/hakoniwa-px4sim #362: docs/manual/drone/drone1.pngにヨーを追記

yamati-kz, 2024-10-08:
@tmori さん
Issues : https://github.com/toppers/hakoniwa-px4sim/issues/358 にてマニュアル整備のことがうたわれているかと思いますが、
https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/drone/drone1.png にてヨーの記述があるといいのではないでしょうか？
是非、初学者のためご検討いただけたらと思います。

tmori, 2024-10-08:
> @tmori さん Issues : #358 にてマニュアル整備のことがうたわれているかと思いますが、 https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/drone/drone1.png にてヨーの記述があるといいのではないでしょうか？ 是非、初学者のためご検討いただけたらと思います。

おっしゃる通りですね。
ドローンの中心に対して、回転を表す矢印に「ヨー」の表記を追加するとよさそうですね。

buildko89, 2024-10-09:
@yamati-kz さん
了解です。記述しましょう。

yamati-kz, 2024-10-09:
@buildko89 さん
お手数お掛けしみません。私がやるとフォントが合わないとか、せっかくの図を汚してしまいそうで。

buildko89, 2024-10-09:
@yamati-kz さん
修正しました。PR投げましたので、少々お待ちを。

yamati-kz, 2024-10-09:
@buildko89 さん
早々のご対応ありがとうございます。
PRの内容を拝見いたしました。

yamati-kz, 2024-10-09:
@tmori さん　@buildko89 さん
Closeとさせていただきます。mainリポジトリへのマージ済みを確認できましたので。
お忙しい中、ご対応ありがとうございました。

closed 2024-10-09 by yamati-kz