toppers/hakoniwa-px4sim #363: ラジコン操作のプロポ対応マニュアル（Windows？）

tmori, 2024-10-09:


yamati-kz, 2024-10-10:
@buildko89 さん
カメラ向きの上下に関しての記述があるといいかなと思うのですが、いかがでしょうか？
[windows_hakowin_installer.md](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/windows_hakowin_installer.md)内の[5.3.1.1. PS4コントローラの操作定義](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/windows_hakowin_installer.md#5311-ps4%E3%82%B3%E3%83%B3%E3%83%88%E3%83%AD%E3%83%BC%E3%83%A9%E3%81%AE%E6%93%8D%E4%BD%9C%E5%AE%9A%E7%BE%A9) に　追記するでありますとか。

例えばですが、表とゲームパッド図にそれぞれ
|No|ファイル名|種類|用途|
|:---|:---|:---|:---|
|6|十字キー/上下|カメラ向きを上下に調整します。|

ゲームハッドの画像にもカメラ向きを上、カメラ向きを下、と文字を入れる。

[image]

ゲームパッドに機能割当する際に、Windowsとその他OSで割当が違うことから、カメラ向き上下を上記画像に入れてしまうのか、別途別の図にするのか迷われるかもしれません。

以下、アドバイス頂いたときのリンクです。
https://github.com/toppers/hakoniwa/discussions/184#discussioncomment-10550892
https://github.com/toppers/hakoniwa/discussions/184#discussioncomment-10575931

お忙しいとは思いますがご検討いただけますと幸いです。

buildko89, 2024-10-11:
@yamati-kz さん
了解です。追記するようにします。

yamati-kz, 2024-10-26:
@buildko89 さん

https://github.com/toppers/hakoniwa-px4sim/issues/363#issuecomment-2404465798 を追記いただきありがとうございました。

tmori, 2024-10-26:
めっちゃわかりやす！
https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/rcdebug.md

closed 2024-10-26 by tmori