toppers/hakoniwa-px4sim #367: Windows版のインストーラ対応のドキュメント

tmori, 2024-10-09:


buildko89, 2024-10-09:
やっておきます！

tmori, 2024-10-09:
ありがとございます！

tmori, 2024-10-09:
@buildko89 実は、[こちら](https://github.com/toppers/hakoniwa-px4sim/issues/359)の対応にともって、ドローン制御のパラメータが外部ファイル化されましたので、同梱いただくファイルが増えることになりました。詳細は、READMEに記載予定ですので、若干のインストーラの変更対応が必要になるかもです。。

tmori, 2024-10-10:
@buildko89 README追記しておきましたー。

https://github.com/toppers/hakoniwa-px4sim/blob/main/drone_control/README-ja.md#%E5%88%B6%E5%BE%A1%E3%83%91%E3%83%A9%E3%83%A1%E3%83%BC%E3%82%BF%E3%83%95%E3%82%A1%E3%82%A4%E3%83%AB

buildko89, 2024-10-13:
とりあえず、ビルド確認はOK！
明日、動作確認してみます。

そして、Windowsのリリースドキュメントが古いことに気が付く…修正しておきます。

yamati-kz, 2024-10-22:
@buildko89 さん　お忙しいところすみません。

Windows版インストーラマニュアルで日本語版と英語版で目次に差分があります。
両マニュアルとも章立てに差分は見当たりませんでした。
英語版を作成されたときに日本語版では章番号が振ってあっても目次になかったものを追記なさったようです。
今後日本語版に目次の追記をされますでしょうか？もし私で良ければ追記してPRさせていただきますがいかがでしょうか？

日本語版 [windows_hakowin_installer.md](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/windows_hakowin_installer.md)　では、下図のような目次
[image]
そして、
英語版 [windows_hakowin_installer_en.md](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/windows_hakowin_installer_en.md)　では、下図のような目次
[image]

になっております。ご検討いただけたらと思います。

buildko89, 2024-10-22:
@yamati-kz さん
ありがとうございます。現状、
v2.7.0のリリースに向けて、ドキュメント含めて、Windowsインストーラなどを改版することをしています。

ドキュメントの改版は、目次含めて対応しますので、少々お待ちください。

yamati-kz, 2024-10-26:
@buildko89 さん

> ドキュメントの改版は、目次含めて対応しますので、少々お待ちください。

マニュアル改版のご対応ありがとうございました。
ご検討いただきました目次ですが、英語版と日本語版ともに、すっきりしてわたしのようなの素人にはとてもありがたいです。
十字キーによるカメラ上下ですが表および画像に追記いただけたことを確認いたしました。

今後マニュアルに沿ってインストールしてWindowsで試してみます。

tmori, 2024-10-26:
RadioControlとFlightControlのところスッキリわかりやすい！

https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/windows_hakowin_installer.md

クローズしますねー。

closed 2024-10-26 by tmori