toppers/hakoniwa-px4sim #386: ローターの物理モデルをより現実的にする（かどうかの調査）

kenjihiranabe, 2024-10-15:
打ち合わせ後に、ちょっとペンディングで悩みができたのでここにメモ。

現状：physics のローターのモデル化が、線形近似による Kr, Kt のみの線形モデルになってる。
理想：より物理的な定数による非線形モデル。
弊害：ホバー回転数より大きくズレる場合（テイクオフ時）に動きが悪い（のか良すぎるのかどちらか）。

参照：
https://www.docswell.com/s/Kouhei_Ito/K38V1P-2024-02-10-094123#p128

書き換えるなら、ここ。Kr, Tr に丸めてしまっているものを、モーターの物理定数で書き換えるか。。。。
あるいは、いったん、時定数をそのままにして、内部実装を逆算して、実装して動きを観察してから考えるか。

```
/* The rotor dynamics modeled as a first-order lag system. eq.(2.48) */
double rotor_omega_acceleration(
    double Kr, /* gain constant */
    double Tr, /* time constant */
    double omega, /* in rpm */
    double duty_rate /* 0-1 of PWM */    ) {
    /*
     * See Nonami's book (2.48)
     * G(s) = Kr/(Tr s + 1). 
     */
    return (Kr * duty_rate - omega) / Tr;
}
```
https://github.com/toppers/hakoniwa-px4sim/blob/6b92a5d71907a7c01224ead4a4733a6d432f7fcc/drone_physics/rotor_physics.cpp#L43

kenjihiranabe, 2024-10-15:
https://www.docswell.com/s/Kouhei_Ito/K38V1P-2024-02-10-094123#p128
より、参照ページを抜き出す。

[image]
[image]
[image]
[image]
[image]
[image]
[image]

これは、線形化された後。 $\Omega は実は、\Delta \Omega$ であり、ホバー回転数中心。
[image]
[image]

kenjihiranabe, 2024-10-15:
検討結果。
やるなら、モーターにまつわる定数

$R$ : 電気抵抗
$C_Q$ : トルク係数（現在の $B$ パラメータで、取得可能）
$\omega_0$: ホバー回転数
$J$ : モーターイナーシャ
$K$ : 逆起電力定数
$V_{BAT}$ : バッテリー電圧

をモーターパラメータとして与えないといけない。これらを使って、

$T_r = \frac{JR}{K^2+2RC_Q \omega_0}$

$K_r = \frac{KV_{BAT}}{K^2+2R C_Q \omega_0}$

と計算したものが、今与えている定数、 $T_r, K_r$ である。

以下、手計算で納得。

[image]

kenjihiranabe, 2024-10-15:
@tmori 上記新パラメータにすると、相当変更がでると思うので、今はペンディングでよいか？勢いがあれば、より物理をリアルに近づけるためにやってしまう手はある。。。。（パラメータ増える）

tmori, 2024-10-15:
> @tmori 上記新パラメータにすると、相当変更がでると思うので、今はペンディングでよいか？勢いがあれば、より物理をリアルに近づけるためにやってしまう手はある。。。。（パラメータ増える） （drone_physics 側の修正は、前のを残して新しい関数に変えて、式を変えるだけで（1時間で）できそう。）

良いですね。
置き換えではなくて、機能追加にしたいと思いました。

箱庭ドローンシミュレータの機体パラメータで切り替えできる作りにすれば、既存機能に影響は与えないかと。

kenjihiranabe, 2024-10-15:
前のものは残す前提で行きます。
さらに調査していますが、こうへい、さんの資料以外にも、モーターのモデル化はいろいろありそうで、
上記のこうへいさんモデルは、w^2 を抵抗力として含むものでしたが、他にも、2次遅れとしてモデル化するものありました。
https://www.isc.meiji.ac.jp/~mcelab/kikai_jikken1/20141210_jikken1_ppt.pdf

kenjihiranabe, 2024-10-17:
手計算の結果、をこうへいさんに見せたら、Tr はこうへいさんの計算ミスということが判明。
新たに、4.60E-02 を使用することにする。

ロータ時定数 | τ | Tr | 4.60E-02 | [s] | JR/(DR+K^2+2RCqω0)
-- | -- | -- | -- | -- | --
ロータ時定数（間違い） | τ | Tr | 1.93E-02 | [s] | こうへいさん資料の値

https://docs.google.com/spreadsheets/d/1JACdNpwAYr-OUya5DmPa8r_1QfsI-4M4G-kJ_zLCvQ0/edit?gid=156552135#gid=156552135

kenjihiranabe, 2024-10-18:
一旦できたのでコミットしました。新しい関数（同じ名前でたくさんの物理・電気定数引数でオーバーライド）作って、テストしました。
関数自体は、「たった1行」なんですが、調査とテストが。。。
これが、その新関数。
https://github.com/toppers/hakoniwa-px4sim/blob/a12948b0640026bde4c262f680a291455fa80ecd/drone_physics/rotor_physics.cpp#L63

コミット全体はこちら。（コメントとかテストコードとか）
https://github.com/toppers/hakoniwa-px4sim/commit/43a0e5803516230c31f301f842653614fafb7016

## ローターのみの比較テスト
omega は新旧(a,b)二つの各速度。acc は新旧(a,b)二つの角加速度（関数の出力値）。
[image]

## 挙動の差分のポイント。
- バッテリ電圧の duty_rate 比率分だけの電圧をかけることにしている。よって、バッテリ低下で出力下がることも再現できる。
- 上記グラフでは、昔(a)よりも新しい方(b)は最大の回転数が低い。ただし、ホバーする回転角速度(1448)は確保した。
- 最大回転数が足らなければバッテリ電圧を現在の11.1[V] から 14.8 [V]にあげないといけないかもしれない。

## グラフの見方。
omega は新旧(a,b)二つの各速度変化。acc は新旧(a,b)二つの角度変化率。両方とも頭打ちになるが、
- 
- aはあらかじめ求めておいたホバー回転数の2倍に収束（そのように Kr を決めたから当たり前）
- b はバッテリーの最大電圧（と回転による抵抗、2乗を含む）の限界で頭打ち。

## テスト時の定数。
旧定数： Kr = 2896, Tr = 4.5e-2
新定数： Vbat = 11.1, R = 0.12, Cq = 3.0e-8, K = 3.28e-3, J = 8.12e-6

定数の仕様は、こちら。（こうへいさんのノミナルモデルのロータ部分を使った）
https://docs.google.com/spreadsheets/d/1JACdNpwAYr-OUya5DmPa8r_1QfsI-4M4G-kJ_zLCvQ0/edit?usp=sharing

kenjihiranabe, 2024-10-18:
残作業

- [ ] README ドキュメントに追加記述
- [ ] 森さんのコードから呼んでもらって感触を得る

あとは、将来に、

- [ ] 電流*電圧モデル実装して、電池残量を管理するモジュール。（drone_physics 内は状態を持たないようにしたいので、森さんのどこかに Battery class を）

tmori, 2024-10-30:
ビジュアライズ部分を入れて、風と連携させると、より効果的なデモになりそうです。

kenjihiranabe, 2024-11-06:
#### 1 ローターの回転角速度（バッテリー電圧を使った非線形モデル）

もう一つのモデルは、ローターの角速度 $\Omega(t)$ がバッテリー電圧 $V_{bat}$ のデューティー比 $d(t)$ で制御されるというものです。
ローターはモーター(電圧によるトルク発生)とプロペラ(トルクによる推力発生)で構成されています。

参照： https://www.docswell.com/s/Kouhei_Ito/KDVNVK-2022-06-15-193343#p2 eq (3)

$$
\begin{array}{l}
J \dot{\Omega}(t) + D \Omega(t) + C_q \Omega(t)^2 = K i(t) \\
L \dot{i}(t) + R i(t) + K \Omega(t) = e(t) \\
e(t) = V_{bat} d(t) 
\end{array}
$$

ここで、
- $J$ - プロペラの慣性モーメント。[ $kg m^2$ ]
- $D$ - プロペラの減衰係数。[ $N m s/rad$ ]
- $L$ - モーターのインダクタンス。[ $H$ ]
- $R$ - モーターの抵抗。[ $\Omega$ ]
- $K$ - ローターのトルク定数。[ $N m/A$ ]
- $d(t)$ - モーターのデューティー比。( $0.0 \le d(t) \le 1.0$ )
- $V_{bat}$ - バッテリー電圧。[ $V$ ]
- $e(t)$ - モーターに印加される電圧。$V_{bat}d(t)$ に等しい。[ $V$ ]
- $i(t)$ - モーターの電流。[ $A$ ]
- $\Omega(t)$ - 得られるプロペラの角速度。[ $rad/s$ ]

インダクタンス $L$ は非常に小さいため無視しすると、

$ i(t) = (e(t) - K \Omega(t))/R $

さらに、電流 $i(t)$ を消去すると、 $\Omega(t)$ は以下のような非線形微分方程式になります。

$$
J \dot{\Omega}(t) + (D + K^2/R) \Omega(t) + C_q \Omega(t)^2 = (K/R) V_{bat} d(t)
$$

$$
\dot{\Omega}(t) = (K V_{bat} d(t) - (K^2+DR) \Omega(t) - C_q R \Omega(t)^2) /JR
$$

関数名は，`rotor_omega_acceleration` の別バージョンです．

https://github.com/toppers/hakoniwa-px4sim/blob/develop/drone_physics/README-ja.md
参照

kenjihiranabe, 2024-11-14:
再度計算してみた。森さんの計算とも合っていて、やっぱり2分でバッテリー尽きる感じ。
https://docs.google.com/spreadsheets/d/1JACdNpwAYr-OUya5DmPa8r_1QfsI-4M4G-kJ_zLCvQ0/edit?gid=156552135#gid=156552135

Qf=0.0 としている点が気になったので、式にも追加してみたが、結果、さらにバッテリーを食う計算。
[image]
[image]

tmori, 2024-11-14:
ちなみに、ホバリング可能な電圧は 9Vくらいなので、実際に飛行できる時間はもっと短くなりますよね。。

kenjihiranabe, 2024-11-14:
現在のぼくの計算だと、上記表リンクにある、

ホバー電圧 | e_0 |   | 7.05 | [V] | K ω0 + Cq R mg/(4KRCt) | = Kω0+(CqR/K)ω0^2
-- | -- | -- | -- | -- | -- | --

で、それに、CPUその他の電圧足すと、ということですよね。
現実的だますには、
1. 容量を増やす
2. 抵抗あげる（森さん案）
3. 翼をイナーシャ変えずに大きくする（Ct 上げる）、などですかね。

tmori, 2024-11-14:
* 1が一番簡単ですね。。
* 2は、ちょっとやってみましたが、PIDパラメータ選定のやり直しなりそうでしたので厳しそうです。10倍したらそもそも浮上しませんでした。。

kenjihiranabe, 2024-11-14:
容量をしれっと、増やしておいて、一旦は、終結しましょう。
こうへいさんモデルにも間違いはありますしね。

tmori, 2024-11-14:
はい、そうしましょう。

kenjihiranabe, 2024-11-14:
↑問題を残してクローズ。

closed 2024-11-14 by kenjihiranabe