toppers/hakoniwa-px4sim #390: Omega の単位（rpm -> rad)

kenjihiranabe, 2024-10-18:
$\omega_0$ - 釣り合いでの回転数の単位ですが、physics 側は、rad/s ですべて対応します。関数仕様や omega_acceleration が返す値の計算はすべて同じです（コメントのみの修正）。

こうへいさんの資料でも 
- $\omega_0 = \sqrt{mg/4Ct}$

となっていて、これを現在の config で計算すると、 
- $\omega_0 = 1448 \quad rad/s$
 
になります。RPM 単位にすると、 
- $HoveringRpm = \omega_0/2 \pi \quad rpm$ 
- $HoveringRpm = 230.46 \quad rpm$
 
となります。

@tmori  そちらで修正しやすいように直してください。シミュレーションに影響しないと思っていますが。。。。

tmori, 2024-10-30:
今日、電車でやります。

tmori, 2024-10-30:
# 修正案

1. 機体のコンフィグ仕様は変更なし
2. 機体パラメータをランタイム側でrad/sに変換する

変換が必要なパラメータ：
* rotor.Kr
* rotor.rpmMax
* thruster.HoveringRpm

tmori, 2024-10-30:
# HoveringRPMの変更ポイント

https://github.com/toppers/hakoniwa-px4sim/blob/b6a0bae8c15d5031c156bb529e7f4b9d896053e0/hakoniwa/src/assets/drone/aircraft/aircraft_factory.cpp#L144-L177

tmori, 2024-10-30:
# rotor.Kr

マクロ値を書き換える。

https://github.com/toppers/hakoniwa-px4sim/blob/b6a0bae8c15d5031c156bb529e7f4b9d896053e0/hakoniwa/src/assets/drone/aircraft/aircraft_factory.cpp#L53

tmori, 2024-10-30:
# rpmMaxの変更ポイント

マクロ値とマクロ名を書き換える。

https://github.com/toppers/hakoniwa-px4sim/blob/b6a0bae8c15d5031c156bb529e7f4b9d896053e0/hakoniwa/src/assets/drone/aircraft/aircraft_factory.cpp#L51C9-L51C16

kenjihiranabe, 2024-10-30:
ここ、注意です。rpm でセッティングならば、2 $pi$ で割ってください。
https://github.com/search?q=repo%3Atoppers%2Fhakoniwa-px4sim%201448&type=code

tmori, 2024-10-30:
> ここ、注意です。rpm でセッティングならば、2 π で割ってください。 https://github.com/search?q=repo%3Atoppers%2Fhakoniwa-px4sim%201448&type=code

了解です！
KrとRpmMaxも同じですよね。

tmori, 2024-10-30:
@kenjihiranabe 

[image]

ここの変換ですが、
RPMは回転数/分なので、
rad/secからRPMに変換するには

```
ω0 x 60/2π
```

ではないでしょうか？

tmori, 2024-10-30:
一旦、コード側の修正終わりました。パラメータの方は、平鍋さんの回答あった後に対応予定です。

https://github.com/toppers/hakoniwa-px4sim/commit/297bc6a8a31eae7a6c1c2066f28f01a8a5a0e302

kenjihiranabe, 2024-10-30:
>ω0 x 60/2π
>ではないでしょうか？

ですです！

tmori, 2024-10-30:
@kenjihiranabe 修正しました。

ご確認ください。

https://github.com/toppers/hakoniwa-px4sim/commit/2b985cde0590e7bc7d559be7802fd14348d10d07

tmori, 2024-10-31:
@kenjihiranabe 
対応完了しましたので、ご確認ください！

https://github.com/toppers/hakoniwa-px4sim/pull/408

kenjihiranabe, 2024-10-31:
動作確認できたので、クローズします。

closed 2024-10-31 by kenjihiranabe