toppers/hakoniwa-px4sim #395: docs/manual/drone/drone1.pngの英語化とwindows_hakowin_installer_en.mdへの挿入

yamati-kz, 2024-10-21:
@buildko89 さん　お忙しいところすみません。すでに取り組まれていらっしゃるかもしれません
[docs/manual/drone/drone1.png](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/drone/drone1.png) を英語化して
[windows_hakowin_installer_en.md](https://github.com/yamati-kz/hakoniwa-px4sim/blob/develop/docs/manual/windows_hakowin_installer_en.md)に挿入して、日本語版と同様にするといいのではないでしょうか、ご検討いただけたらと思います。

buildko89, 2024-10-22:
@yamati-kz さん
ご指摘ありがとうございます。

英語化にあたっては、私の方で対応していないため、別途調整とさせてください。

一応、以下のように英語版はできてます。

https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/windows_hakowin_installer_en.md

yamati-kz, 2024-10-22:
@buildko89 さん　
コメントありがとうございます。承知しました。

もし英語化する際にお手間が省けるようでしたら私が試しに作成した下図drone1-en.pngをベースしていただいても構いません。
PRしたものに黒い点がありましたので消したものです。

[image]

buildko89, 2024-10-22:
@yamati-kz さん
作図ありがとうございます。別途、英語版対応している方と相談して、対応考えます。

yamati-kz, 2024-10-28:
題名を変えさせていだだきました。日本語版の構成が少々変わっておりましたので。
ご対応を検討されていらっしゃると思いますが、英語版ではこれらの部分が不要というご方針ならCloseしていただいてかまいません。

buildko89, 2024-10-29:
@yamati-kz さん
英語版は、日本語版に合わせる方向で作業中です。少々お待ちください。