toppers/hakoniwa-px4sim #398: cmake の version

kenjihiranabe, 2024-10-24:
MacOS の方を最新 cmake 3.30にしたら動かなくなった、、、3.10 以上で上方互換方針が変わったらしい（warning）。
また、Errorも出ている。ひとまず、drone_physics だけ動くので、ぼくの作業は問題ない。
@tmori どこかのタイミングで一度バージョン上げてくれると嬉しい。

```
hakoniwa-px4sim % cmake --version
cmake version 3.31.0-rc1

CMake suite maintained and supported by Kitware (kitwa
hakoniwa-px4sim % cmake .
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  **Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at drone_physics/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.**

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at matlab-if/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- HAKONIWA_MATLAB_BUILD=
CMake Deprecation Warning at hakoniwa/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at hakoniwa/third-party/hakoniwa-core-cpp-client/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- CMAKE_SYSTEM_PROCESSOR: arm64
-- CMAKE_OSX_ARCHITECTURES: arm64
-- BUILD_TYPErelease
-- GCOVdisabled
CMake Deprecation Warning at hakoniwa/third-party/hakoniwa-core-cpp-client/core/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- BUILD_TYPErelease
-- GCOVdisabled
-- OS_TYPE=posix
CMake Warning (dev) at /Applications/CMake.app/Contents/share/cmake-3.31/Modules/FetchContent.cmake:1373 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  hakoniwa/third-party/hakoniwa-core-cpp-client/core/src/CMakeLists.txt:16 (FetchContent_Declare)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Using the multi-header code from /Users/hiranabe/Desktop/Code/hakoniwa-px4sim/_deps/json-src/include/
-- CMAKE_VERSION:3.31.0-rc1
-- OS_TYPE=posix
-- CMAKE_VERSION:3.31.0-rc1
-- OS_TYPE=posix
-- CMAKE_VERSION:3.31.0-rc1
-- OS_TYPE=posix
-- CMAKE_VERSION:3.31.0-rc1
-- PYTHON_INCLUDE_DIRS:
-- Python_INCLUDE_DIRS:/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/include/python3.11
-- PYTHON_LIBRARIES:
-- Python_LIBRARIES:/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/libpython3.11.dylib
-- Python_LIBRARY_DIRS:/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib
CMake Deprecation Warning at hakoniwa/third-party/hakoniwa-core-cpp-client/examples/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- CMAKE_VERSION:3.31.0-rc1
-- OS_TYPE=posix
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /Applications/CMake.app/Contents/share/cmake-3.31/Modules/FindThreads.cmake:226 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  hakoniwa/third-party/hakoniwa-core-cpp-client/examples/hello_world/CMakeLists.txt:32 (find_package)

-- Configuring incomplete, errors occurred!
```

tmori, 2024-10-24:
了解です！version2.7.0のリリースが終わったあとに対応しますね。

tmori, 2024-10-30:
cmake version 3.30.5

tmori, 2024-10-30:
brew upgrade cmake

kenjihiranabe, 2024-10-31:
やっと、cmake / make / pthread の関連のローカルで動きました（ただし drone_physics 内のみ）。homebrew でインストールしたものと、x-code でインストールしたものがコンフリクトして、cmake が選んだコンパイラと、環境がずれたんですね。
でも、hakoniwa 全体の compile は動いていません。。。。調整中。

kenjihiranabe, 2024-10-31:
全体の cmake での引っ掛かりです。なんか新しい error で、drone_workspace の ChatGPT に関するものでした。
もし簡単に推測つくならみてください。

hakoniwa-px4sim % pwd
/Users/hiranabe/Desktop/Code/hakoniwa-px4sim
hakoniwa-px4sim % cmake .
-- HAKONIWA_MATLAB_BUILD=
-- CMAKE_SYSTEM_PROCESSOR: arm64
-- CMAKE_OSX_ARCHITECTURES: arm64
-- BUILD_TYPErelease
-- GCOVdisabled
-- BUILD_TYPErelease
-- GCOVdisabled
-- OS_TYPE=posix
CMake Warning (dev) at /opt/homebrew/Cellar/cmake/3.30.5/share/cmake/Modules/FetchContent.cmake:1373 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  hakoniwa/third-party/hakoniwa-core-cpp-client/core/src/CMakeLists.txt:16 (FetchContent_Declare)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Using the multi-header code from /Users/hiranabe/Desktop/Code/hakoniwa-px4sim/_deps/json-src/include/
-- CMAKE_VERSION:3.30.5
-- OS_TYPE=posix
-- CMAKE_VERSION:3.30.5
-- OS_TYPE=posix
-- CMAKE_VERSION:3.30.5
-- OS_TYPE=posix
-- CMAKE_VERSION:3.30.5
-- PYTHON_INCLUDE_DIRS:
-- Python_INCLUDE_DIRS:/opt/homebrew/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/include/python3.13
-- PYTHON_LIBRARIES:
-- Python_LIBRARIES:/opt/homebrew/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/lib/libpython3.13.dylib
-- Python_LIBRARY_DIRS:/opt/homebrew/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/lib
-- CMAKE_VERSION:3.30.5
-- OS_TYPE=posix
-- CMAKE_VERSION:3.30.5
-- OS_TYPE=posix
-- CMAKE_VERSION:3.30.5
-- OS_TYPE=posix
-- OS_TYPE=posix
CMake Warning (dev) at /opt/homebrew/Cellar/cmake/3.30.5/share/cmake/Modules/FetchContent.cmake:1373 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  hakoniwa/src/CMakeLists.txt:13 (FetchContent_Declare)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at drone_control/CMakeLists.txt:23 (add_subdirectory):
  The source directory

    /Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/workspace/ChatGPT3_5

  does not contain a CMakeLists.txt file.

CMake Error at drone_control/CMakeLists.txt:23 (add_subdirectory):
  The source directory

    /Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/workspace/ChatGPT4

  does not contain a CMakeLists.txt file.

CMake Error at drone_control/CMakeLists.txt:23 (add_subdirectory):
  The source directory

    /Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/workspace/DroneAvator

  does not contain a CMakeLists.txt file.

CMake Error at drone_control/CMakeLists.txt:23 (add_subdirectory):
  The source directory

    /Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/workspace/DroneFlightController

  does not contain a CMakeLists.txt file.

CMake Error at drone_control/CMakeLists.txt:23 (add_subdirectory):
  The source directory

    /Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/workspace/Gemini

  does not contain a CMakeLists.txt file.

tmori, 2024-10-31:
古いディレクトリが残っているのかも。

以下が、最新の状態です。

```
% ls drone_control/workspace 
FlightController                        hako_module_drone_controller.c
RadioController                         hako_module_drone_controller_impl.h
```

kenjihiranabe, 2024-10-31:
通りました！ありがとう。
小さな Warning が出ています。簡単なので、直してしまいませんか？
（class のメンバ宣言と初期化の順番だと思われる）
In file included from /Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/common/include/drone_alt_controller.hpp:5:
/Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/common/include/flight_controller_types.hpp:14:11: warning: field 'current' will be initialized after field 'target' [-Wreorder-ctor]
   14 |         : current(current_val), target(target_val) {}
      |           ^~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~~~~~~
      |           target(target_val)    current(current_val)
/Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/workspace/FlightController/hako_module_drone_controller_impl.cpp:17:64: warning: unused parameter 'context' [-Wunused-parameter]
   17 | int hako_module_drone_controller_impl_is_operation_doing(void *context)
      |                                                                ^
In file included from /Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/workspace/FlightController/hako_module_drone_controller_impl.cpp:2:
In file included from /Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/common/include/drone_controller.hpp:7:
/Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/common/include/drone_angle_controller.hpp:57:12: warning: private field 'pos_simulation_time' is not used [-Wunused-private-field]
   57 |     double pos_simulation_time = 0;
      |            ^
/Users/hiranabe/Desktop/Code/hakoniwa-px4sim/drone_control/common/include/drone_angle_controller.hpp:71:12: warning: private field 'spd_simulation_time' is not used [-Wunused-private-field]
   71 |     double spd_simulation_time = 0;
      |            ^
4 warnings generated.