toppers/hakoniwa-px4sim #399: 加速度レベルでの風対応

kenjihiranabe, 2024-10-24:
風への対応は現在、drone_physics を呼び出す側で、

- 現在速度 += 風速度

としている。これは対空速度を表現していると見れば妥当だが、

- 加速度センサーが働かない

という問題がある。加速度を通して速度を上げることを考える。

方法は、現在の「風の抵抗係数」にあたる部分を、 $-d1(u - air.u) -d2(u - air.u)^2$ のように変更する。

## 変更場所
`acceleration_in_body_frame` およびその呼び出し

- [ ]  x, y, z 方向抵抗のインターフェイスに揃える(1方向のものは廃止し、呼び出し側を変える）
- [ ]  その時に、座標系は機体で入れること（まずは現在と同じ等方からはじめる）
- [ ]   air 引数を追加し、 $-d_1(u - air.u) -d_2 sign(u - air.u) (u - air.u)^2$ とする。向きに注意

## 参考文献

[image]

tmori, 2024-10-31:
@kenjihiranabe 
対応完了しましたので、ご確認ください！

https://github.com/toppers/hakoniwa-px4sim/pull/408

kenjihiranabe, 2024-10-31:
風の2乗の影響を入れ、コードを完成させた。ただし、森さんの側からは今は2乗は使われていない認識。
一旦完了。
https://github.com/toppers/hakoniwa-px4sim/commit/f86f7f451abb1208ae7b0e2bd2d4a7e6948e43e4

closed 2024-10-31 by kenjihiranabe