toppers/hakoniwa-px4sim #405: リンク切れ「Linux環境でのカメラ操作」/ ubuntu22.04_install&operation_pythonAPI編.md

yamati-kz, 2024-10-28:
@buildko89 さん @tmori さん　

v2.7リリース作業ありがとうございました。大変お疲れのところすみません。

[ubuntu22.04_install&operation_pythonAPI編.md]() ,  [1.4.5. カメラ操作について](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/ubuntu22.04_install%26operation_pythonAPI%E7%B7%A8.md#145-%E3%82%AB%E3%83%A1%E3%83%A9%E6%93%8D%E4%BD%9C%E3%81%AB%E3%81%A4%E3%81%84%E3%81%A6) の最後の[Linux環境でのカメラ操作](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/ubuntu/hako81.png) の部分でリンクが切れております。

ご報告まで、失礼いたします。

buildko89, 2024-10-29:
@yamati-kz さん

了解です。明日直します。本日は、外出中…

buildko89, 2024-10-29:
@yamati-kz さん
対応しました。確認ください。

yamati-kz, 2024-10-29:
@buildko89 さん
Closeさせていただきます、画像が表示されることを確認いたしましたので。
お疲れのところありがとございました。

closed 2024-10-29 by yamati-kz