toppers/hakoniwa-px4sim #409: 離陸するには？(ubuntu22.04_install&operation_pythonAPI編.md)

yamati-kz, 2024-11-02:
@tmori さん　こんにちは。
すみません、アドバイスいただけたらと思います。

PS4コントローラのXボタンを押してプロペラが回った後に左側JoyStickを操作しても離陸できません。
以前の手順を思い出しながら試したのですがうまく行かない状態です。2台のPCで試しましたがうまくいかないのは同じです。
ubuntu22.04.5LTSをクリーンインストール後、ubuntu22.04_install&operation_pythonAPI編.mdに沿ってインストールしています。
[1.4.1. 箱庭コア機能の起動](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/ubuntu22.04_install%26operation_pythonAPI%E7%B7%A8.md#141-%E7%AE%B1%E5%BA%AD%E3%82%B3%E3%82%A2%E6%A9%9F%E8%83%BD%E3%81%AE%E8%B5%B7%E5%8B%95) , [1.4.2. Unityアプリの起動](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/ubuntu22.04_install%26operation_pythonAPI%E7%B7%A8.md#142-unity%E3%82%A2%E3%83%97%E3%83%AA%E3%81%AE%E8%B5%B7%E5%8B%95) , [1.4.4. PS4のコントローラでのドローン操作](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/ubuntu22.04_install%26operation_pythonAPI%E7%B7%A8.md#144-ps4%E3%81%AE%E3%82%B3%E3%83%B3%E3%83%88%E3%83%AD%E3%83%BC%E3%83%A9%E3%81%A7%E3%81%AE%E3%83%89%E3%83%AD%E3%83%BC%E3%83%B3%E6%93%8D%E4%BD%9C) の順に各アプリを起動しています。
[1.4.3. sampleアプリの起動](https://github.com/toppers/hakoniwa-px4sim/blob/main/docs/manual/ubuntu22.04_install%26operation_pythonAPI%E7%B7%A8.md#143-sample%E3%82%A2%E3%83%97%E3%83%AA%E3%81%AE%E8%B5%B7%E5%8B%95) では飛行して荷物も運べています。

以下に各Terminalのログを添付します。
```
#1.4.1. 箱庭コア機能の起動
ububtu@Lenovo:~$ cd ~/work/hakoniwa-px4sim/hakoniwa
ububtu@Lenovo:~/work/hakoniwa-px4sim/hakoniwa$ bash drone-app.bash ../../hakoniwa-unity-drone-model/DroneAppLinux ./config/api_sample
hakoniwa master is enabled
HAKO_CAPTURE_SAVE_FILEPATH : ./capture.bin
HAKO_BYPASS_IPADDR : 127.0.0.1
HAKO_CUSTOM_JSON_PATH : ../../hakoniwa-unity-drone-model/DroneAppLinux/custom.json
DRONE_CONFIG_PATH : ./config/api_sample
HAKO_BYPASS_PORTNO : 54001
INFO: LOADED drone config file: ./config/api_sample/drone_config_0.json
INFO: hako_master_init() success
INFO: setup start
DroneDynamicType: BodyFrame
INFO: logpath: ./drone_log0/drone_dynamics.csv
Rotor vendor: None
Thruster vendor: None
param_A: 8.30482e-07
param_B: 3e-08
param_Jr: 0
INFO: mixer is not enabled
INFO: loading drone & controller: 0
controller param path: ../drone_control/config/param-api.txt
SIMULATION_DELTA_TIME: 0.003
PID_ALT_CONTROL_CYCLE: 0
PID_ALT_MAX_POWER: 9.81
PID_ALT_MAX_SPD: 10
PID_ALT_THROTTLE_GAIN: 1
MASS: 0.71
GRAVITY: 9.81
PID_ALT_Kd: 1.5
PID_ALT_Ki: 0.1
PID_ALT_Kp: 10
PID_ALT_SPD_Kd: 0.1
PID_ALT_SPD_Ki: 0.1
PID_ALT_SPD_Kp: 5
SIMULATION_DELTA_TIME: 0.003
POS_CONTROL_CYCLE: 0
PID_POS_MAX_SPD: 20
PID_POS_X_Kd: 0.6
PID_POS_X_Ki: 0
PID_POS_X_Kp: 1
PID_POS_Y_Kd: 0.6
PID_POS_Y_Ki: 0
PID_POS_Y_Kp: 1
SPD_CONTROL_CYCLE: 0
PID_POS_MAX_ROLL: 10
PID_POS_MAX_PITCH: 10
PID_POS_VX_Kd: 0.1
PID_POS_VX_Ki: 0
PID_POS_VX_Kp: 20
PID_POS_VY_Kd: 0.1
PID_POS_VY_Ki: 0
PID_POS_VY_Kp: 20
SIMULATION_DELTA_TIME: 0.003
HEAD_CONTROL_CYCLE: 0
PID_YAW_RPM_MAX: 10
PID_YAW_Kd: 0
PID_YAW_Ki: 0
PID_YAW_Kp: 1.1
[... 105 lines cut ...]
Controller:  rc_config/ps4-control-lnx.json
Mode:  2
Number of joysticks: 1
ジョイスティックの名前: Sony Interactive Entertainment Wireless Controller
ボタン数 : 13
INFO: Success for external initialization.
button event: switch_index=0 event_op_index=0 down: True event_triggered=True
button event: switch_index=0 event_op_index=0 down: False event_triggered=False
stick event: stick_index=1 op_index=1 event.value=-0.6078676717429121 stick_value=-0.10862393788948718
stick event: stick_index=1 op_index=1 event.value=-0.631397442548906 stick_value=-0.13196938051834645
stick event: stick_index=1 op_index=1 event.value=-0.6706137272255623 stick_value=-0.1584208023221179
stick event: stick_index=1 op_index=1 event.value=-0.7098300119022187 stick_value=-0.19118369922901654
stick event: stick_index=1 op_index=1 event.value=-0.7333597827082126 stick_value=-0.226458960282856
stick event: stick_index=1 op_index=1 event.value=-0.7647328104495377 stick_value=-0.262057403663311
stick event: stick_index=1 op_index=1 event.value=-0.7882625812555315 stick_value=-0.30119008212143855
stick event: stick_index=1 op_index=1 event.value=-0.8117923520615253 stick_value=-0.3395690817172594
stick event: stick_index=1 op_index=1 event.value=-0.8196356089968566 stick_value=-0.37156978194966356
stick event: stick_index=1 op_index=1 event.value=-0.8510086367381817 stick_value=-0.4080152392723312
stick event: stick_index=1 op_index=1 event.value=-0.8666951506088443 stick_value=-0.4414610172794147
stick event: stick_index=1 op_index=1 event.value=-0.8745384075441756 stick_value=-0.4711458629050161
stick event: stick_index=1 op_index=1 event.value=-0.8823816644795068 stick_value=-0.49639812536579064
stick event: stick_index=1 op_index=1 event.value=-0.8980681783501694 stick_value=-0.5254942500019136
stick event: stick_index=1 op_index=1 event.value=-0.9137546922208319 stick_value=-0.5495713929736409
```
お手数をおかけしてすみません。#407をチェックする前に、まずは修正前で動かそうとしています。

tmori, 2024-11-02:
多分、ここだと思われます。

```
bash drone-app.bash ../../hakoniwa-unity-drone-model/DroneAppLinux ./config/api_sample
```

ラジコン操作は、rcですね。
```
bash drone-app.bash ../../hakoniwa-unity-drone-model/DroneAppLinux ./config/rc
```

yamati-kz, 2024-11-02:
@tmori さん @buildko89 さん

ありがとうございました、離陸できました。rcでコマンド叩いてます。
以前にもご教示いただいた気もしますが、少々あやふやで。以前に動いたのは何だったのか思い出せません。
そして、思ったのがubuntu22.04_install&operation_pythonAPI編.mdに反映させた方がいいということです。
windows版の章立を後ほど確認してみます。

yamati-kz, 2024-11-03:
原本の流れを壊さないように検討してみましたがいかがでしょうか？
ご検討いただけたらと思います。

変更前
[image]

変更後
[image]

buildko89, 2024-11-04:
@yamati-kz さん
私の方で、Ubuntu対応がまだできてないのですが、確認頂いた内容で問題ないですので、修正願います。

私も、Ununtu24.04.1 Ubuntu22.04.5の環境作って試してみます。

yamati-kz, 2024-11-04:
@buildko89 さん
PR #411 のご承認をお願いいたします。https://github.com/toppers/hakoniwa-px4sim/issues/409#issuecomment-2453456933 を反映させました。

Ununtu 24.04.1LTSを初めて使い、そして試してみましたが問題なく動作しました。ただpip3の使い方がUnuntu 24.04LTSと22.04LTSと違うので戸惑いました。ChatGPTとこちらの記事が参考になりました。https://qiita.com/AzukiImo/items/5c6dd7e870c9965e41f3

buildko89, 2024-11-05:
@yamati-kz さん
#411 マージしました。ご対応ありがとうございました。

そうなんですよね…24.04だと、若干22.04と違いがあって…もう少し見てからドキュメント整理します。

yamati-kz, 2024-11-05:
@buildko89 さん
Closeいたします。ご対応ありがとうございました。

24.04でのその後ですがPC再起動後に動かなかったり、OSのクリーンインストールしては悪戦苦闘中です。素人なりに試してみます。

closed 2024-11-05 by yamati-kz