toppers/hakoniwa-px4sim #412: Ubuntu24.04LTS対応(ubuntu22.04_install&operation_pythonAPI編.md)

yamati-kz, 2024-11-05:
@tmori さん @buildko89 さん　こんにちは。いつもありがとうございます。

インストールマニュアル等のUbuntu24.04LTS対応をご検討される際のご参考になれば幸いです。
このプロジェクトとして、OSやpythonに関してどこまで記述するのか難しいかとは思いますが。

現状のubuntu22.04_install&operation_pythonAPI編.mdでもISOファイルからインストールメディアを作成して、PCへクリーンインストールしますと、インストールマニュアルの手順の他に、pip、pygame、numpyを自力でインストールする必要があります。

[Ubuntu 22.04.5 LTS (Jammy Jellyfish)](https://releases.ubuntu.com/jammy/) / ubuntu-22.04.5-desktop-amd64.iso の例.
```
$ sudo apt install python3-pip
$ sudo pip3 install pygame numpy
```
[Ubuntu 24.04.1 (Noble Numbat)](https://releases.ubuntu.com/noble/) / ubuntu-22.04.5-desktop-amd64.iso の例.
```
$ sudo apt install python3-pip
$ sudo apt install python3-pygame python3-numpy
```
上記24.04.1の場合は、PCを再起動しても問題なく動作しております。数回OSクリーンインストールから試した結果です。
ただ下のような方法ですとPS4コントローラ操作用Pythonを起動する際等々うまくいっておりません。Terminalを閉じたり、PC再起動した場合もです。
```
$ python3 -m venv myenv
$ source myenv/bin/activate
$ pip install pygame numpy
```

tmori, 2024-11-05:
@yamati-kz さん、ご連絡ありがとうございます！ 何回もクリーンインストールを試していただいて大感謝です！！

不足するPythonライブラリの件ですが、箱庭ドローンシミュレータとしての` requirements.txt` を定義しておくと良さそうですね。

これがあれば、以下のコマンドで必要なライブラリを一括でインストールできるようになります：

```
pip install -r requirements.txt
```

なお、自分用に作ったものは[ここ](https://github.com/toppers/hakoniwa-px4sim/blob/main/hakoniwa/python/requirements.txt)にありますが、どこかで整理して、トップディレクトリにおきたいと思っておりました...。

tmori, 2024-11-05:
> Terminalを閉じたり、PC再起動した場合もです。

なんと。。myenvの中にインストールされると思っていました。。

source はされてますよね？？