toppers/hakoniwa-px4sim #433: クオータニオン対応

kenjihiranabe, 2024-12-04:
いつかやる
https://interface.cqpub.co.jp/wp-content/uploads/IF03_074-080_E04-1.pdf

kenjihiranabe, 2024-12-04:
Quaternion 、Euler 角相互変換についてより詳しい記事（上記記事と同一著者）
https://interface.cqpub.co.jp/wp-content/uploads/IF03_054-069_A104-3.pdf

[image]
[image]
[image]

kenjihiranabe, 2024-12-05:
オイラー角とクオータニオン対応メモ。

## 分かったこと

- クオータニオンは「姿勢」を表し、同じ役割のオイラー角と１：１対応する（時間微分でなくこの両者が相互変換）。
- 現在の姿勢は、クオータニオンを正として維持、そこから、必要時に上記によってオイラー角に変換するべき。
- よって、aircraft 内に姿勢クオータニオンを保持する必要がある。そして、クオータニオンの時間微分を積分してアップデートしていく。
- クオータニオンの時間微分は、 $\dot{q} = \frac{1}{2} q \otimes w$ である。（野波先生の本には1/2がヌケ）

## 対応方針

###  drone_physics

現在の関数 euler_rate_from_body_angular_velocity（ピッチ90°付近で内部0割破綻する） に対応する新たな関数、quaternion_velocity_from_body_angulear_velocity を作成する。これは上記時間微分を担当。結果 $\dot{q}=\frac{dq}{dt}$ を返す。
- 
- 
### - aircraft

DroneDynamicsBodyFrame クラス（３つある）のなかに、EulerType(initial, angle), EulerRateType(angularVelocity), それぞれ対応するクオータニオンを保持する。
  - drone_dynamics_body_frame.hpp
  - drone_dynamics_body_frame_rk4.hpp
  - drone_dynamics_body_frame_matlab.hpp

上記 physics 側関数を呼び出して、eular でなく、quaternion を計算、積算する。

## 注意

- クオータニオン要素の並び順には二つの流儀があり、ベクトルとしての順番に注意。 (q0,q1,q2,q3) として、 xyzw か wxyz か。
- 特に glm を使う場合、順序は xyzw となっている（0.997以降。0.997以前は wxyz らしい）https://qiita.com/syoyo/items/0177fee6778746207ed7
- 最新版は、1.0系になっているので、hakoniwaもアップグレードした方がよい。（本家 https://github.com/g-truc/glm ）
- Quaternionとオイラー角の変換詳説。このページの書き方だと、先頭の回転順xyz （z-y-xの順）に対応する。
https://qiita.com/aa_debdeb/items/3d02e28fb9ebfa357eaf

kenjihiranabe, 2024-12-07:
まずは、オイラーとクオータニオンの相互変換

- euler_from_quaternion
- quaternion_from_euler

作成。テストを回すが、特殊ケースで $\theta = \pm 90$ のときに特殊扱い必要。テストが失敗する例は、 $q=[-1/2, -1/2, -1/2, 1/2]^T$ で、これを図示してみた。$cos \theta = 0$ すなわち、ピッチ角が垂直のときにジンバルロックが起こるので、Q->E の解は一意に決まらなくなる。その特殊ケースを入れて、再度テスト。ピッチが、90°の近傍 0.001 範囲にあるときは、特殊なロジック（ $\phi=0$ ）を入れて走らせたら、全テストケースが通るようになった。なので、とりあえず、これで変換部分は終わって、本丸の Quaternion の時間微分に入る。
もしかしたら、ground_vector_from_body_frame など、EulerAngleType を姿勢としてパラメータに渡しているところは、Quaternion 姿勢に変えるかも。（Quaternion -> Euler でも姿勢は 0割などなく、安定）。

[image]

kenjihiranabe, 2024-12-10:
本丸の Quaternion の時間微分できました。これで、euler_rate_from_body_angular_velocity がクオータニオン化できます。
physics 側一旦終了です。aircraft 側での変更方針について、明日、説明しますね。
https://github.com/toppers/hakoniwa-px4sim/commit/1a6564fda027912bc3e648654da9d11e697bf422

kenjihiranabe, 2024-12-11:
クオータニオンバージョンで飛んだので、残りは、

- [ ] drone_dynamics_body_frame.hpp の整理

です。本日の議論です。

[image]

tmori, 2024-12-11:
あー、あと、センサ系に影響ないかはPX4で実験してみる必要があると思われます。

空いた時間でトライしてみます。

kenjihiranabe, 2024-12-26:
動いたので、クローズ。

closed 2024-12-26 by kenjihiranabe