toppers/hakoniwa-px4sim #443: workflow のビルドが失敗する

kenjihiranabe, 2024-12-17:
drone_physics 最新をコミット
https://github.com/toppers/hakoniwa-px4sim/issues/434#issuecomment-2547561522

した流れのビルド結果が失敗。

https://github.com/toppers/hakoniwa-px4sim/actions/runs/12366402155

考察：

1. win/ubuntu/macos の順に走る。先頭がこけたら、あとはキャンセル ----> 二人がまずビルドするであろう macos を先頭にしては？
2. cmake の警告が必ず出るのが気になる。直してしまいたい。
3. win で warning が多くでるが、できれば、クリーンに潰してしまうか、理由があって出るものは warning 番号で殺してしまい、ログをクリーンにしたい。
4. 本質的に、なぜビルドが失敗しているのか、僕にはわからなかった。。。。 HELP!! @tmori 

よく出る cmake の警告（自分でビルドしても必ず出る）：
```
  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

CMake Deprecation Warning at third-party/hakoniwa-core-cpp-client/core/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.
```

tmori, 2024-12-17:
Macでビルドしたらエラー出ました。

ビルド方法：
```bash
cd drone_contrl
bash build.bash
```

エラー：
```
[ 37%] Linking CXX shared module libFlightController.so
ld: Undefined symbols:
  hako::drone_physics::body_vector_from_ground(hako::drone_physics::VectorType const&, hako::drone_physics::EulerType const&), referenced from:
      DronePosController::run_pos(DronePosInputType&) in hako_module_drone_controller_impl.cpp.o
  hako::drone_physics::ground_vector_from_body(hako::drone_physics::VectorType const&, hako::drone_physics::EulerType const&), referenced from:
      DroneAltController::run_spd(DroneAltSpdInputType&) in hako_module_drone_controller_impl.cpp.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [workspace/FlightController/libFlightController.so] Error 1
make[1]: *** [workspace/FlightController/CMakeFiles/FlightController.dir/all] Error 2
```

body_vector_from_ground()が見つからないのが原因のようですね。。

tmori, 2024-12-17:
直してみました。

kenjihiranabe, 2024-12-17:
ありがとう、他のことは喫緊ではないので、安心しました。

kenjihiranabe, 2024-12-22:
cmake の警告を直してみました。ダウンロードのタイムスタンプ抽出を明示的に ON しました。
推奨らしい。
https://github.com/toppers/hakoniwa-px4sim/commit/506fb37b24a86224f79d8820069cff9853ba2657

```
CMake Warning (dev) at /opt/hostedtoolcache/cmake/3.31.3/x64/cmake-3.31.3-linux-x86_64/share/cmake-3.31/Modules/FetchContent.cmake:1373 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  third-party/hakoniwa-core-cpp-client/core/src/CMakeLists.txt:16 (FetchContent_Declare)
This warning is for project developers.  Use -Wno-dev to suppress it.
```

kenjihiranabe, 2024-12-23:
合わせて、こちらも対応(HAKO_CMAKE_VERSION -> 3.20)

https://github.com/toppers/hakoniwa-px4sim/actions/runs/12450438239/job/34757250754
```
Run cd drone_physics && cmake -D HAKO_CLIENT_OPTION_FILEPATH=../cmake-options/linux-cmake-options.cmake . && make  && make test
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.
```

kenjihiranabe, 2024-12-26:
森さんに修正お願いして、新環境にて対応。

closed 2024-12-26 by kenjihiranabe