toppers/hakoniwa-px4sim #445: ubuntu22.04_install&operation_PX4編.mdの章番号の修正。

yamati-kz, 2024-12-24:
@tmori さん @buildko89 さん　お疲れ様です。

ubuntu22.04_install&operation_PX4編.md をみながらインストールしていく中で思ったのですが
文脈からFig.2のような章立てとするのはどうでしょうか？

|          |
|:-------:|
|[image]|
|Fig 1. 変更前|

|          |
|:-------:|
|[image]|
|Fig 2. 変更後|

以上。

buildko89, 2024-12-25:
ありがとうございます！

buildko89, 2024-12-25:
マージさせて頂きました！

yamati-kz, 2024-12-25:
ありがとうございました。お忙しい中、すみません。

closed 2024-12-25 by yamati-kz

tmori, 2024-12-25:
@yamati-kz さん、いえいえ、ご提案ありがとうございます！

（いま、大リファクタリング大会中で、手が回らず、すみません。。目は回ってますが。。

yamati-kz, 2024-12-26:
@tmori さん
こちらこそすみません、重箱の隅のそのまた隅で。
大リファクタリング大会、本当にお疲れ様です。その大変さはよくわかるつもりです。
これからもド素人目線でご提案できればと思います。